pachi81/GlucoDataHandler #32: Support ESP32

lostboy86, 2023-02-06:
M5stack and M5stick ESP32 was an arduino all in one (display, battery , wifi, ble etc) , used to show blood glucose with nightscout value.

On xdrip4ios, value sents directly over bluetooth.
On Android, with Juggluco, there is no support for nightscout directly, and expecially for one-minute value update for m5stack.

Many people use m5stack/stick in car, motorbike, sport activity, home clock/alarm at bed, easy to buy, cheapest, and programmable

Support for this features would be great.

Thanks in advance
[image]
[image]

pachi81, 2023-02-06:
Hi,
I haven't heard about these devices, yet.
But it looks very interesting. 
At the moment I'm thinking about nightscout upload...
For direct connection I will need to buy one, first.
As I only have limited time, it will takes some time.

lostboy86, 2023-02-06:
If you would to test, i’m ready for you, no problem.I have m5stack and m5stickC Plus (i also have a cystic fibrosis plus Diabethes T1, for this i use m5stack/stick when i drive, at my bed, etc)For what i read online, the value of glucodatahandler can write directly over bluetooth with serial bluetooth console, but i don’t understand how to take the value of glucodatahandler on tasker, i use Automate, but i’m very n00b for this   Da: pachi81Inviato: lunedì 6 febbraio 2023 20:23A: pachi81/GlucoDataHandlerCc: lostboy86; AuthorOggetto: Re: [pachi81/GlucoDataHandler] Support ESP32 (Issue #32) Hi,I haven't heard about these devices, yet.But it looks very interesting.At the moment I'm thinking about nightscout upload...For direct connection I will need to buy one, first.As I only have limited time, it will takes some time.

pachi81, 2023-02-06:
Hi,
GlucoDataHandler sends its data to the Wear OS App and to Tasker, but does not write direct over bluetooth to any other device.
Maybe Tasker can do this, but I have no idea, how this can be realized.
At the moment I also don't know, how the data has to be formated to send to M5Stack.

lostboy86, 2023-02-06:
GitHub - Mollayo/BluetoothSerialForTasker: A plugin for Tasker to
communicate with a Bluetooth device using the Serial Port Profile (SPP).
<https://github.com/Mollayo/BluetoothSerialForTasker>

There are many plugin from tasker to arduino esp32

I try on Automate to retrieve your data, the plugin of glucodahandler work
correctly but i don't know how work, for example i set a variable "gluco"
for direct value of sensor, and display on a notify on phone, but when i
test the macro only notification appear with "glicemia : " text but nothing
value

pachi81, 2023-02-06:
Not sure how familiar you are with Tasker.
You have to add a new profile, select event, plugin, GlucoDataHandler. Then create a task, add notfication and for text or title use %glucose.
But %glucose only contains data, if the task is triggered by my app, otherwise it is empty.
If you want to store my value in a Tasker Variable, this variable must contains at least one big letter, for example %ReceivedGlucose. Such a variable can be used in other profiles and tasks.

pachi81, 2023-02-06:
I have created an example Tasker profile:
https://1drv.ms/u/s!AvqtWFwQmBjwv9tKDfpFd-NCCfYYYg
Download it, go to Tasker, long press on Profile Tab and select import.

lostboy86, 2023-02-07:
Dont' have tasker, i try to test today, thank you very much for the guide.

On my phone i have automate and macrodroid instead paid Tasker, only for
test tour Glucodatahandler reception of value.

Today i also try to connect with m5stack, i'm so sad i cant't help more.

pachi81, 2023-02-07:
Good luck!

lostboy86, 2023-02-07:
Sorry for another question, what watchfaces do you use on wear?

I want to delete juggluco app on wear, and have a similar watchfaces normal
with you complications.

Do you know or use a watchfaces with central bigger glucose and trend, or
something similar?

Thanks for the suggest.

If  you want, i can pay for you an m5stick/stack, for development, the
community of diabetic for this is bigger.

pachi81, 2023-02-07:
At the moment I´m using the default ones from my TicWatch. But I haven´t found anything, where the complication is big and in central position. 
If you are connected to you phone, you can checkout my [watchfaces ](https://github.com/pachi81/GlucoseWatchfaces/blob/master/images/Pujie_Black_Watchfaces.jpg) for Pujie Black. The xDrip big is maybe what you are looking for...
But then you need to buy tasker and pujie black, but you don´t need GlucoDataHandler on wear, the version for the phone is enough.
[Here ](https://github.com/pachi81/GlucoseWatchfaces/blob/master/README.md) you can find an installation instruction.

lostboy86, 2023-02-07:
I try to stop (not unistall, initially, i'm outside and i dont' want to
loose glucose data if anything not work) glucodata on wear, but data not
arrive in wear after

I unistall juggluco wear, i only have glucodata wear app on wear, and
glucodata and juggluco on phone (value show on glucodata phone app), i can
unistall the wear glucodata?

lostboy86, 2023-02-07:
Another problem,

In ambient mode, with Juggluco watchfaces value update , with watchfaces
and glucodata complications, if i dont' touch the wear, value not updates

pachi81, 2023-02-07:
In ambient mode, the values only updates on change of the minute. This is also a problem of juggluco. If the values are not updating, you can try to activate the foreground option.

lostboy86, 2023-02-07:
Yes i have activated, juggluco watchfaces work every minute, glucodata
complications on juggluco watchfaces also work every minute, glucodata
complications on other watchfaces not update in ambient mode .

I try 10 watchfaces

pachi81, 2023-02-07:
Have you tried to add the glucodata complication to the juggluco watchfaces and check, if the value are updated?

pachi81, 2023-02-07:
And you can see the notification of GlucoDataHandler on wear?

lostboy86, 2023-02-07:
Yes, notification glucodata on wear, and complications glucodata are work
correctly on juggluco watchfaces ( small value obiusvly)

Update value Is consistent.

If i unistall juggluco wear, and put complications glucodata on another
watchfaces, not update in ambient mode

pachi81, 2023-02-07:
Strange can you try to create Logcat for that?

lostboy86, 2023-02-07:
Umh i dont'k know how, i'm a n00b, only program some website code or arduino

pachi81, 2023-02-07:
Ok, but you have installed my app using adb?
Then it is the same, connect to wear:
`adb connect [IP of the wear]`
and then execute the logcat statement
`adb logcat | findstr GlucoDataHandler. > glucodatahandler.log`

Wait some minutes in ambient and then press "CTRL + C" to stop logcat.
In the folder you will find the file glucodatahandler.log and send it to me.

lostboy86, 2023-02-07:
Thanks for the suggest,  at now i tr build watchfaces with Watchmaker, and all work.I try to misure the power consumption and update you Inviato da Posta per Windows Da: pachi81Inviato: martedì 7 febbraio 2023 11:43A: pachi81/GlucoDataHandlerCc: lostboy86; AuthorOggetto: Re: [pachi81/GlucoDataHandler] Support ESP32 (Issue #32) Ok, but you have installed my app using adb?Then it is the same, connect to wear:adb connect [IP of the wear]and then execute the logcat statementadb logcat | findstr GlucoDataHandler. > glucodatahandler.logWait some minutes in ambient and then press "CTRL + C" to stop logcat.In the folder you will find the file glucodatahandler.log and send it to me.

pachi81, 2023-02-07:
Just another question, what kind of watch do you have? Have tried an watchface from play store?
If yes, can you send me a link?
Thank you!

lostboy86, 2023-02-07:
Huawei Watch 2 Sport LTE, with WearOS 2.

I try the watchfaces build with wear, then i try juggluco and then
watchmaker.

For m5stick/stack ble support, this sound very Easy and simple to integrate
on Glucodatahandler,

https://m5stack.hackster.io/happy83cz/m5stickc-watch-c51464

Trough serial bluetooth console, phone can send data (20bytes long)

pachi81, 2023-02-26:
I will close this issue, as it can be done with Tasker at the moment.

closed 2023-02-26 by pachi81