pachi81/GlucoDataHandler #33: Be kind with non-tech people

psonnera, 2023-02-06:
This is a great app, can you put more screenshots of the app, various configuration menus etc. for non tech people?
Thanks!

pachi81, 2023-02-06:
Hi,
I know the docu is not my main topic 😉
This App does not have a lot of configurations, it still provides Wear OS Complications, which are part of many watchfaces.
The Tasker integration is not needed for that. Tasker can be used for automations, but this is not such easy.

Or what especcialy you are meaning?

Regards, Michael

psonnera, 2023-02-07:
Let me try to find time to do something and show you what I mean. The product is good, just that following the instructions can be conplex for some. Will PR when done. Thanks!

pachi81, 2023-02-22:
Hi, 

I have checked your docu and it looks really good. I know, I have to add some more docu, but starting with your changes is great:
https://github.com/pachi81/GlucoDataHandler/pull/39

Can you have a look?

psonnera, 2023-02-23:
I will try to take better screenshots this weekend. Thanks!

pachi81, 2023-02-23:
Thank you! But you don't have to. It should be my task writing the docu 😀

pachi81, 2023-02-26:
Thank you for your input. I close this issue.

closed 2023-02-26 by pachi81