pachi81/GlucoDataHandler #80: Polish translation of your app

froster82, 2023-06-28:
Hello!

I find your app very helpful and really appreciate your work, and would like to thank you for that.
I prepared Polish versions of strings and would like to ask you if you can implement them to make the app available in Polish?
[pl-PL.zip](https://github.com/pachi81/GlucoDataHandler/files/11893710/pl-PL.zip)
If anything else is required for that, just let me know. I will try to deal with Readme and Installation files next.
Best regards,
Arek Deliś

pachi81, 2023-06-28:
Hi,
I have added your values. Can you verify it:
[GDH_0.9.1.5.zip](https://github.com/pachi81/GlucoDataHandler/files/11895136/GDH_0.9.1.5.zip)

The changes can be found here:
https://github.com/pachi81/GlucoDataHandler/pull/81/commits/d3c061d729b6ebba9b3438bc1ecb1c13c99579ef

For "Colored AOD" I´m using "Kolorowy AOD". Correct?

Regards, Michael

froster82, 2023-06-28:
Hi!Thanks for reply! Great! Kolorowy AOD will be great. I am now away from my computer, so I will take a look at it when I can at home. BestArek
-------- Oryginalna wiadomość --------Od: pachi81 ***@***.***> Data: 28.06.2023  15:24  (GMT+01:00) Do: pachi81/GlucoDataHandler ***@***.***> DW: froster82 ***@***.***>, Author ***@***.***> Temat: Re: [pachi81/GlucoDataHandler] Polish translation of your app (Issue #80) 
Hi,
I have added your values. Can you verify it:
GDH_0.9.1.5.zip
The changes can be found here:
d3c061d
For "Colored AOD" I´m using "Kolorowy AOD". Correct?
Regards, Michael

froster82, 2023-06-29:
Hi!

I have checked the apps on both phone and watch and they look all right.
Thanks again for great work!

I have one more question, though :)
Can you somehow make the phone version show delta and trend in the status bar? I use Juggluco and I really miss this feaure on the phone. I am really glad you made it possible to see reliably the trend arrow and values on the watch, but it would be great to have those options also on the phone.

Best,
Arek

pachi81, 2023-06-29:
Hi Arek,
thank you for the feedback.
If you did not find any problems, I will create a pre-release 9.2 on monday.

A permanent notification will be not a big thing, as I already use it for Android Auto.

I also thought about a small widget for the phone.

But at the moment with the beautiful weather I only have less time 😉
So it will take a while...

Regards, Michael

froster82, 2023-06-29:
Great, much appreciated!
And I totally understand and agree with time issues :)
Best,
Arek

pachi81, 2023-07-04:
Fixed with version [0.9.2](https://github.com/pachi81/GlucoDataHandler/releases/tag/0.9.2)

pachi81, 2023-07-14:
I re-open it, as there is more translation todo.
I also gave you access to this repo, so can add change for translation directly to the files.
At the moment, I´m working on the notification, which has some new settings.

froster82, 2023-07-14:
Hi! When you have your final English version, I will look at it and see what has changed. I haven't had time to get a hold of Github functionality yet, so if I cannot upload my updated versions, I will just drop them here.

pachi81, 2023-07-14:
I created a new PR: https://github.com/pachi81/GlucoDataHandler/pull/83
I think, you can edit the pl-strings.xml file direct in the browser, so you do not have to upload anything. Maybe this is the easiest way for you.

Here is the current version, so can see the settings in the app.
[GlucoDataHandler_0.9.2.4.zip](https://github.com/pachi81/GlucoDataHandler/files/12049504/GlucoDataHandler_0.9.2.4.zip)

froster82, 2023-07-14:
I usually uplod the current version of the English file into my CAT tool, and can easily trace all changes. I will try to deal with this as soon as possible, but my day-to-day work orders come first ;)

pachi81, 2023-07-14:
Yes no problem, it is not time critical ;-)
I also have limited time, so I know this situation very well :-D

froster82, 2023-07-16:
Hi! I have dealt with the strings file from mobile folder. The easiest way for me to do that is to upload it to my CAT tool, so I will just attach it here. Can you paste it where you need to, or do I have to do it as part of review proces? I have never used Github before, that's why I am not sure of the procedure :)
[strings_pl.zip](https://github.com/pachi81/GlucoDataHandler/files/12063881/strings_pl.zip)

pachi81, 2023-07-16:
Thank you! No problem, I can put it in the code as well, if you do not have experience using github.

froster82, 2023-07-16:
Can I just paste the whole code into strings in PL folder? Or maybe can I copy the file? How to do it best? For future reference.

froster82, 2023-07-17:
Great, thanks for that. I appreciate it, because in the future I will not have to take your time to do that.

pachi81, 2023-07-17:
No problem, the replacement of the strings is not a big thing, so I can also do it. Only with the translation to polish, I have some problems ;-)
Thank you for supporting my app.

froster82, 2023-07-19:
I just installed the version you provided above and I think I found 2 minor bugs. In the notification, between the glucose value and the trend arrow, I think there is a space missing. In the app itself, those are separated ok with a space and it looks better. Also, the trend arrow in the status bar (when it is up- or downwards) gets cut on the right-hand side. Both of these can be seen in the scrrns.
[image]
[image]
[image]

froster82, 2023-07-19:
I am on Samsung galaxy s9 with Android 10. Maybe this is the reason for this difference. Also - would it be possible in some future version to get in the status bar a trend arrow with delta value at the same time? Just as one of the options on the wear app. (I know, I know - I can be a pain the a## :) ).

froster82, 2023-07-19:
Oh, I just like throwing some crazy ideas around, because sometimes they are found to be quite good. And on other occassions they are not so good. I am just trying to achieve the same level of functionality on the phone, as I now have on my watch. With your "trend arrow + delta value" complication I am now able to determine better the rate of glucose increase or decrease. People without the watch do not have this really helpful feature now.

pachi81, 2023-07-19:
At the moment I'm implementing a widget:
[image]

Maybe then you do not need an additional notification.

Would you use a second notification, for having an additional status bar icon?

froster82, 2023-07-19:
Now this is a wonderful news and takes care of all the things I wrote above. With a widget we have all the information needed. Great! Thanks!

pachi81, 2023-07-19:
If you like, you can try it:
[GlucoDataHandler_0.9.2.6.zip](https://github.com/pachi81/GlucoDataHandler/files/12094614/GlucoDataHandler_0.9.2.6.zip)

It would be interesting for me, if you have also some problems with space between value and arrow.

froster82, 2023-07-19:
Everything looks fine on my device. 
[image]

pachi81, 2023-07-19:
But it looks different as on my device... The arrow is more in the middle and not on the right border as on your device. 
Maybe I can improve it...

pachi81, 2023-07-19:
Hi,

here is a new version, where I have adjust the notification:
[GlucoDataHandler_0.9.2.7.apk.zip](https://github.com/pachi81/GlucoDataHandler/files/12099818/GlucoDataHandler_0.9.2.7.apk.zip)
Can you check, how it looks, now.

For the widget, I do not have so much influence in Android 10. In higher version I can change the width of the images dynamically.
Maybe I will find another solution.

froster82, 2023-07-20:
Hi!
I attached screens from yeasterday (99) and from today (109) so you can see how the things changed.
[Screens.zip](https://github.com/pachi81/GlucoDataHandler/files/12102556/Screens.zip)
Wider than 1x1 widget in Android 10 is still alligned to the right, but that is not something which is troublesome.

pachi81, 2023-07-20:
Great! Now it looks much better!
I found some other solution for setting the layout.
I will play around with that.

pachi81, 2023-07-20:
Hi,
I have changed the layout-implementation of the widgets to be more dynamically independent of the android version. 
Would be great, if you can test it and send me a screenshot, how it looks on your device. 
[GlucoDataHandler_0.9.2.8.apk.zip](https://github.com/pachi81/GlucoDataHandler/files/12103946/GlucoDataHandler_0.9.2.8.apk.zip)

Thank you! 
Regards, Michael

froster82, 2023-07-20:
1x1 looked better in the previous version with bigger value. 
1x2 is I think more in the centre. 

[image]

pachi81, 2023-07-20:
I think both looks better, now. If you resize the widget to the right, the delta and time will be move the right.
But I recognized, that your status bar icon is still cut off.
I have to change it again...

pachi81, 2023-07-20:
So, now the status bar icon should work...
I also did some smaller changes to the widget layouts.
[GlucoDataHandler_0.9.2.9.apk.zip](https://github.com/pachi81/GlucoDataHandler/files/12105708/GlucoDataHandler_0.9.2.9.apk.zip)

pachi81, 2023-07-20:
Yes, this would be no big deal.

pachi81, 2023-07-20:
Here are 2 additional widgets: 
[GlucoDataHandler_0.9.2.10.apk.zip](https://github.com/pachi81/GlucoDataHandler/files/12110169/GlucoDataHandler_0.9.2.10.apk.zip)

pachi81, 2023-07-20:
Yes, I know. I haven't yet time to create new previews.

pachi81, 2023-07-20:
Hi,
there are 2 new text lines in the strings. xml:
https://github.com/pachi81/GlucoDataHandler/pull/84/files#diff-bd6311cc0b6875d4350bc94577a3598d016c2287731d2a5d9036cde4c9640ac6R71

-> line 71 and 72 
I added an option for a bigger statusbar icon, so you do not have the cut one and I can have a bigger one :-)

Now, you can try to edit it direct in the commit ;-)

froster82, 2023-07-20:
I edited those on my phone, but it seems it worked. You should now have them ready for approval.

pachi81, 2023-07-21:
Hi, 
I added some more strings for the second notification - config.
Now you can have a second notification with delta value:
[GlucoDataHandler_0.9.2.12.apk.zip](https://github.com/pachi81/GlucoDataHandler/files/12122598/GlucoDataHandler_0.9.2.12.apk.zip)

pachi81, 2023-07-21:
Thank you! Here is the version with your translation to verify:
[GlucoDataHandler_0.9.2.13.apk.zip](https://github.com/pachi81/GlucoDataHandler/files/12122933/GlucoDataHandler_0.9.2.13.apk.zip)

froster82, 2023-07-21:
Hi! I spotted and corrected one more mistake and commited the change. Widgets with previews look great. Secong and bigger icons also work fine on my device.

pachi81, 2023-07-21:
Also the bigger trend icon? It isn´t cut off? I´m surprised :-)

froster82, 2023-07-21:
Ok, wrote too soon :D The tip is a bit cut off when the arrow goes up
[image]

froster82, 2023-07-21:
OK, one more thing translated - as promissed Readme and Install instruction in Polish :) I had some time, and could finish it. But without Polish screens yet. 
[Readme and Installation - PL.zip](https://github.com/pachi81/GlucoDataHandler/files/12129434/Readme.and.Installation.-.PL.zip)

pachi81, 2023-07-21:
Wow, great! 
I'm not at home this weekend. I will have a look next week.

JodokusQ, 2023-07-23:
Hallo pachi81,
Der Trendpfeil zeigt in die falsche Richtung. Ist auch auf deinen Screenshots zu sehen. Ansonsten funktioniert das und sieht bei mir genauso aus, wie bei dir. Habe ein S23+

JodokusQ, 2023-07-23:
Korrektur:
Jetzt ist es Richtig. Anscheinend nur ein Initialisierungsproblem

pachi81, 2023-07-24:
> Hallo pachi81, Der Trendpfeil zeigt in die falsche Richtung. Ist auch auf deinen Screenshots zu sehen. Ansonsten funktioniert das und sieht bei mir genauso aus, wie bei dir. Habe ein S23+

Was genau meintest du mit "falsche Richtung"? Das eine ist der Trendpfeil und das andere der Delta Wert, die können schonmal unterschiedlich sein, während der Trend noch steigend ist, kann das Delta schon negativ sein. Das kann entweder an einer kurzen Schwankung liegen oder an einer Trendwende...

froster82, 2023-07-24:
I think the trend direction and delta values might be a bit misleading in the beginning, but probably it is just as you described above.
Coming back to Polish version - I have one question about one sentence regarding "rate (trend)".
"arrow rotate dynamically between +2.0 (↑) and -2.0 (↓)" - do you refer here to mmol/l or just the mathematical rate? Because in the Polish version I wrote "+36 (+2,0) (↑) a -36 (-2,0)(↓) " and now I am not sure if it was necessary to convert this value to mg/dl.

pachi81, 2023-07-24:
No, this is the mathematical rate. I don´t know, how this is calculated, because I got this value from Juggluco or xDrip+ and do not calculate anything on my side.
I will change it in your polish version.

froster82, 2023-07-24:
OK, thanks. I saw values wtih decimals, so I automatically converted them. Only later I thought that these may be mathematical.
I corrected this and pasted Polish screens to Readme.
[pl-PL.zip](https://github.com/pachi81/GlucoDataHandler/files/12142743/pl-PL.zip)

pachi81, 2023-07-24:
Hi Arek,

can you check the result: 
https://github.com/pachi81/GlucoDataHandler/blob/develop/README_PL.md
https://github.com/pachi81/GlucoDataHandler/blob/develop/INSTALLATION_PL.md

Thank you!

froster82, 2023-07-24:
Hi! Looks fine! I only forgot to attach screenshot from Wear in the ZIP file, and pasted it inside docx.
Here it is: 
[image]

pachi81, 2023-07-24:
Done

froster82, 2023-07-24:
The only thing that can still be changed is the first capital letter in "wersja polska", Otherwise everything looks ready to go.

pachi81, 2023-07-24:
Done

pachi81, 2023-08-27:
Hi,
I have two new strings for translation:
https://github.com/pachi81/GlucoDataHandler/pull/90/files#diff-bd6311cc0b6875d4350bc94577a3598d016c2287731d2a5d9036cde4c9640ac6

After the last Android Auto update, the dummy media player does not work. So I have to simulate selecting a "song" to still have the value in the overview. Now it looks like here:

[image]

Because I´m not sure, if this have any influence to other media player apps, one have to activate it in the settings...

pachi81, 2023-09-18:
New strings are available:
https://github.com/pachi81/GlucoDataHandler/pull/92/files#diff-bd6311cc0b6875d4350bc94577a3598d016c2287731d2a5d9036cde4c9640ac6
and
https://github.com/pachi81/GlucoDataHandler/pull/92/files#diff-53aead5c1105150732f126121569424fd9ac595640a45fc4ea67942b08ed5f6d

I created a floating widget and added some tasker state events.

froster82, 2023-09-19:
Hi! When I have a moment, I will have a look at them.

pachi81, 2023-09-19:
Thank you! You have time, I´m still testing and maybe I will change some more things.

pachi81, 2023-10-18:
I close this ticket, as I contact you direct in the merge. So this ticket is obsolete.

Thank you very much for your support!!!

closed 2023-10-18 by pachi81