pachi81/GlucoDataHandler #85: Add floating widget

lostboy86, 2023-08-04:
I'm a strong enthusiast user of your work.

Can you implement, like Juggluco, the floating widget with colors in rage, out, ecc, value, etc?
I use this widget on my android radio (not android auto), over Agama and user radio interfaces (like google maps, etc)

Thanks for your hard work

pachi81, 2023-08-04:
Hi,
as this is part of Juggluco, I´m not sure, if it is really needed. 
But you are right, the colors and delta you do not have in the Juggluco version.
At the moment, there is not much time and I´m on holiday for the next time. 
So it will take some time...

lostboy86, 2023-08-04:
Thanks for the reply, good holiday!

pachi81, 2023-09-01:
Hi,
can you give me some more information, you have a android radio (not android automotive OS?). 
How Agama and the user radio interface are working? What can you access from the mobile phone? Also notifications?
Or only the floating widget?
Thank you!

Regards, Michael

lostboy86, 2023-09-01:
Android radio: atoto s8 premium
Android 12
Can install juggluco for mirror, and apk Glucodatahandler.

Agama its a Launcher

pachi81, 2023-09-01:
Ok, this means, you install Juggluco and GlucoDataHandler on the radio. Juggluco receives the values from the phone app and send them to GlucoDataHandler. Correct!?
What about the notifications in Juggluco and GlucoDataHandler? GlucoDataHandler also have normal widgets, can you access them? Or is the floating widget the only possibility?

lostboy86, 2023-09-01:
Normale widget can't be access on Android radio.

Not only with Agama or other Launcher, but when you drive and you have for
example gmaps or other, you can't have a widget overlay, for example

pachi81, 2023-09-01:
And notifications are also not possible!?
But I have checked an example for the floating widget, and it seems to be similar to the normal widgets. 
So maybe I can use the most of the code and layout from the normal widgets for that, which make it easier for me.

pachi81, 2023-09-01:
And popup notifications, like when you are receiving a new message from other apps. But yes, I also think, that the floating widget is the better solution...

lostboy86, 2023-09-01:
Popup notification can't display over Launcher or over some apps

Widget floating must be the best thing, for all android stock  car dashboard

pachi81, 2023-09-01:
Ok, thank you for the information. As soon, I have time, I will try to implement it.

pachi81, 2023-09-01:
Here ist the first try...
You can activate in settings in the notification section. 
On the first activation, you have to set the permission and go back. If it does not appear after setting the permission, disable and enable the setting in GlucoDataHandler again.
I know, it is a little bit small at the moment, but it should work...

[GlucoDataHandler_0.9.4.3.zip](https://github.com/pachi81/GlucoDataHandler/files/12499465/GlucoDataHandler_0.9.4.3.zip)

lostboy86, 2023-09-01:
On my Phone work, only i have two fix for suggest :

- Press long 5 sec to dismiss, like juggluco

- Press normally and launch Glucodatahandler or juggluco

- in my phone with normal text size, house, second and delta its half cut
maybe a settings for size of carachter?

 Next hour, After my phisioterapy for cistica fibrosis, i can test on the
radio.

Great work you are a magic!

pachi81, 2023-09-01:
Hi,

thank you for the feedback! I increased the size. For me, it is nearly to big, now. I have to find a good way...
But the bigger one is maybe better for car...

At the moment, I don´t know, how to implement the on-click event for calling the app or closing the widget, because at the moment I´m using the touch event for moving the widget around...

I have to check, maybe I find the way, how Juggluco have implemented it.

[GlucoDataHandler_0.9.4.4.apk.zip](https://github.com/pachi81/GlucoDataHandler/files/12500653/GlucoDataHandler_0.9.4.4.apk.zip)

pachi81, 2023-09-01:
Please also try this version for checking cut off the time...
Thank you!
[GlucoDataHandler_0.9.4.5.zip](https://github.com/pachi81/GlucoDataHandler/files/12501288/GlucoDataHandler_0.9.4.5.zip)

lostboy86, 2023-09-02:
On phone its cut, see attachments

lostboy86, 2023-09-02:
Cant reply with attachments, too big for github

pachi81, 2023-09-02:
Also with the 0.9.4.4 version?

pachi81, 2023-09-02:
What kind of phone do you have? Maybe I can reproduce it in an emulator.

lostboy86, 2023-09-02:
Oppo Find X2 pro.do you have an email for Send attachments?

It.s 5Mb , 7.5mb, etc

pachi81, 2023-09-02:
If the files are so big, so better upload it here:
https://1drv.ms/f/s!AvqtWFwQmBjwi2GueEaUdaIS-Ess
If it does not work, my mail: pachi81@web.de

pachi81, 2023-09-04:
New version with fixed size, so nothing should be cut...
I also add short click to open Juggluco and long click (4s) to close the floating widget (both only works, if you do not move the widget...).

[GlucoDataHandler_0.9.4.6.apk.zip](https://github.com/pachi81/GlucoDataHandler/files/12514603/GlucoDataHandler_0.9.4.6.apk.zip)

lostboy86, 2023-09-04:
Tested, not fix the cut.
Touch and launch juggluco work perfectly.

Press long and hide not work, Hide and  After 1 sec restart, sometimes work
and hide

lostboy86, 2023-09-05:
.6 its very small and cut on the radio

pachi81, 2023-09-05:
It is cut in width or height?

To close it, you have to touch at least for 4s and do not move it...
Maybe the movement detection is too sensitive, but it is hard to decide either the user want to move or want to close... And I don´t want to add a close button...

lostboy86, 2023-09-05:
Width and height.

I send in the previous message the collage screenshots

pachi81, 2023-09-05:
Which Android version your Oppo phone have?

pachi81, 2023-09-05:
So, next try...
I changed the layout completely...
I added also a resize slider in the settings, so you can increase the widget size.
Also improved the long click to close...
[GlucoDataHandler_0.9.4.7.zip](https://github.com/pachi81/GlucoDataHandler/files/12528133/GlucoDataHandler_0.9.4.7.zip)

lostboy86, 2023-09-06:
A 13 on phone, A12 on Android car radio atoto

pachi81, 2023-09-06:
Thank, unfortunately I could not reproduce the problem in emulator. 
But the new version .7 should work, now. (I hope...)

lostboy86, 2023-09-07:
I test many and many hours on phone and also in a road trip of 8 hours on
android radio.

For me it seems latest version fix all problem, sometimes not
stay persistent overlay on the agama launcher or, in the phone, but it's a
minor problem.

pachi81, 2023-09-07:
> sometimes not stay persistent overlay on the agama launcher or, in the phone, but it's a minor problem.

What do you mean? Don't really understand.

But good to know, that it works, now :-)

lostboy86, 2023-09-07:
I mean this :

Sometimes the widget disappear when i launch some apps, and reappear After,
or stay on background

pachi81, 2023-09-07:
That is really strange! I have not recognized on my phone, yet.
Do you have any Logreader App, like [this one](https://play.google.com/store/apps/details?id=com.conena.logcat.reader) ?
If possible can you activate logging for my app and when it dissappears send the log to me. Maybe I will see anything, which causes this problem.

pachi81, 2023-09-08:
One more question, does it always happen, when you launch the same app or is it more a sporadic issue?

lostboy86, 2023-09-08:
More sporadic issue

pachi81, 2023-09-08:
:-( Ok, if it will be reproduceable, it is fixable :-D

Have you activated Play Protect in the Google Play Store? If yes, try to deactivate it, because the Play Protect like to close apps, which are not coming from Play Store...

And it sound like, that my app is closed and on new value it is re-started (in the background). Otherwise I have to create a foreground service, which will not closed such often... 

One more thought: are you using the notification of GlucoDataHandler? If not, can you activate it and check, if it improves the problem?

Thank you very much for you testing and help!!!

pachi81, 2023-09-08:
And one more thing. I think, I have the same problem. In the version, where the last pos of the widget was not saved, it happens, that it re-appears in the center (sounds like a restart). So maybe I can test and log it on my device.
I will have a look.

lostboy86, 2023-09-08:
No play protect
Yes notification firm on notification center

lostboy86, 2023-09-08:
If i Remember correctly, same for me

lostboy86, 2023-09-08:
> So, I have checked with the old version, where the widget always starts in
> the center and check the logcat. The app restarts from time to time (don´t
> know why, maybe some android optimizations).
> If I have time, I will make the notification as foreground-notification,
> so the app should not be killed (I hope).

Thanks! Great news

pachi81, 2023-09-11:
Hi,
here is a new version with foreground service (Notification must be activated).
In the emulator, the notification gets colored background which looks strange, but on my phone I had not problems. So it is hopefully only a emulator problem...

[GlucoDataHandler_0.9.4.8.apk.zip](https://github.com/pachi81/GlucoDataHandler/files/12575245/GlucoDataHandler_0.9.4.8.apk.zip)

lostboy86, 2023-09-11:
I have a strange colour notification on the notification center, on dark
night mode

Others bug seems fixed, congrats, well done

I test tomorrow on the radio

pachi81, 2023-09-12:
> I have a strange colour notification on the notification center, on dark night mode 

Can you make a screenshot?

lostboy86, 2023-09-12:
also send this night in the previous mail

pachi81, 2023-09-12:
Please send direct to my mail (pachi81@web.de). It seems to be removed from GitHub, because there is no attachement.

lostboy86, 2023-09-13:
> Please try this version. It also add Tasker support for de-/activating the
> widget.
>
> GlucoDataHandler_0.9.4.9.zip

All seems fixed. Today i test more on the Road with radio.

lostboy86, 2023-09-13:
Tested on radio, there is some strange thing :

slider to adjust the size of the widget  : not adjust the dimension of the
entire widget, only the text

pachi81, 2023-09-13:
Ok, the slider only adjust the text-size, but the widget should be updated...
So, the text will be cut after resize? What happens, if a new value will be there?
If you disable and enable after resizing, does the dimensions are correct?

pachi81, 2023-09-13:
I create a new version, where I also had to do some changes on the layout (hopefully you will not recognize it).
I also added an option to change the style of the floating widget. 
And for you problem on the radio, I remove and re-add the view on changes to the settings. 
Please try, if this solves the problem on the radio.

[GlucoDataHandler_0.9.4.10.apk.zip](https://github.com/pachi81/GlucoDataHandler/files/12596464/GlucoDataHandler_0.9.4.10.apk.zip)

lostboy86, 2023-09-13:
The size its the same on radio only text was adjust with slider

on smartphone adjust size and text

This evening test the new version and report.

Thanks a lot for the hard work

lostboy86, 2023-09-17:
With latest version, all work on phone and radio.

Style of widget, size variable, on-top of the apps, etc.

You can release the latest version, work very good!

pachi81, 2023-09-17:
Great news! Thank you for your feedback!
I haven't expected, that it takes so much time...

Just one more question, you are coming from Italy? Do you like to translate the app into italian?

Regards, Michael

lostboy86, 2023-09-17:
Yes coming From Trento, how i can translate?

lostboy86, 2023-09-17:
Another questione :

Juggluco mirror Is very bad function for me, becauss ip of device change
every time on radio, with tethering.

You able to send value over ble From Glucodatahandler instance on One
device and another device with Glucodatahandler?

Same you act between phone and wear with Glucodatahandler.

This happens with every mirror, phone mother change ip on network home,
radio change ip when tethering phone ON, etc.

It's a wastw of time adjust ip on mirror every time, for example if i turn
on and off my car six times on a Road trip, i have to adjust the mirror ip
on radio because also tethering phone change ip phone

Same

pachi81, 2023-09-17:
I had a quick look and there seems to be a solution for bluetooth connection. 
But not sure, when I can spend some time for playing around.

Dor translation, I will send you a file, when I'm ready.

lostboy86, 2023-09-17:
Thank you wait for the file

closed 2023-12-06 by pachi81