pachi81/GlucoDataHandler #86: Watchfaces for WearOS

lostboy86, 2023-08-04:
Good morning,

Can you implement, like Juggluco watchfaces , but with color onrange value, etc?

The Juggluco watchfaces had central big value and trend arrow, locked to white, and some other complications you can change.
No dynamic change of color, no big value colored (also with you complications under) , very limited.

Thanks for your hard work

pachi81, 2023-08-04:
At the moment, I do not plan to implement watchfaces, as everyone likes different things...
That´s the reason, I provide colored complications, you can use in the watchface you like most.
And if you do not find the right one, use apps like Pujie Black, where you can create your own watchface and use my complications or the tasker integration.
Some information to this topic, you can find here: https://github.com/pachi81/GlucoseWatchfaces
I also provide some watchfaces in Pujie Black for Tasker integration or one for complications.

pachi81, 2023-08-06:
You can try Pujie, I do not recognized any problems.
I haven't used watchmaker for a long time.
But if I have some time, I can try to create a simple watchface, maybe it isn't such complicated.

lostboy86, 2023-08-06:
For me its a great idea custom watchfaces by Glucodatahandler, exspecially for colour range etc. Inviato da Posta per Windows Da: pachi81Inviato: domenica 6 agosto 2023 09:42A: pachi81/GlucoDataHandlerCc: lostboy86; AuthorOggetto: Re: [pachi81/GlucoDataHandler] Watchfaces for WearOS (Issue #86) You can try Pujie, I do not recognized any problems.I haven't used watchmaker for a long time.But if I have some time, I can try to create a simple watchface, maybe it isn't such complicated.

pachi81, 2025-02-03:
With the release of new Wear OS 5 watches, it is not possible anymore to provide watchfaces inside an app. So I will not investigate any work on this topic.

closed 2025-02-03 by pachi81