pachi81/GlucoDataHandler #87: Glucodata seems incompatible with xDrip and AndroidAPS

TheFlyingDave, 2023-08-11:
I use xDrip+ with AndroidAPS and am trying to get GlucoDataHandler working for a custom watchface.

I followed the instructions and set "Identify receiver" in xDrip+ to empty (was info.nightscout.androidaps).
Now the GlucoData app receives xDrip broadcasts but they are not getting to AndroidAPS. I have "Send glucodata broadcast" enabled and have tried it both with AndroidAPS checked and unchecked in "Identify Receiver". In both cases AndroidAPS never gets updated glucose values.
TIA

pachi81, 2023-08-11:
Please try for Identify receiver:
`info.nightscout.androidaps
de.michelinside.glucodatahandler`

TheFlyingDave, 2023-08-11:
That worked. Many thanks!

TheFlyingDave, 2023-08-11:
THis can be closed.

closed 2023-08-11 by TheFlyingDave

pachi81, 2023-08-11:
Thank you for the feedback! If I have some time, I will update the Readme...