pachi81/GlucoDataHandler #94: Very big colored complication color changes spontaneously

Igor-bubble, 2023-09-28:
Color of this complication is unstable. Sometimes color (in normal range) changes into white outline font regardless of  color selected. Attempts to select different colors leads to change to solid white. I verified this on 2 different watch faces from different sources, including Stretched Time from Samsung. Restarting the watch sets display to correct color, but not always. On one occasion color changed to selected one in a few hours. Color change usually goes wrong after complications manipulation as far as I can tell.
Many thanks to your efforts!

pachi81, 2023-09-28:
Oh, that sounds really strange! Can you make some screenshots? Thank you!

Igor-bubble, 2023-09-28:
Will do.Thank youIgor From: pachi81Sent: September 28, 2023 1:31 PMTo: pachi81/GlucoDataHandlerCc: Igor-bubble; AuthorSubject: Re: [pachi81/GlucoDataHandler] Very big colored complication color changes spontaneously (Issue #94) Oh, that sounds really strange! Can you make some screenshots? Thank you!

pachi81, 2023-09-28:
Great! 
What kind of watch are you using? And which Wear OS version?
Thank you!

Igor-bubble, 2023-09-28:
Resending the image without the video because it is too big for email. Attached are 2 screenshots showing alternating Juggluco V.B. colored and xDrip complications   Igor From: pachi81Sent: September 28, 2023 2:04 PMTo: pachi81/GlucoDataHandlerCc: Igor-bubble; AuthorSubject: Re: [pachi81/GlucoDataHandler] Very big colored complication color changes spontaneously (Issue #94) Great!What kind of watch are you using? And which Wear OS version?Thank you!

pachi81, 2023-09-29:
Hi,
sorry, but GitHub removes the attachments sent by mail. You have to add them direct in GitHub. 
From the mobile I think you have to activate the desktop mode of GitHub to get the "button" for adding files.

pachi81, 2023-09-29:
Hi, you can also use GDrive or something like this and send me only the link to the files.
So you can also include the video

Igor-bubble, 2023-09-29:
Below are 2 links to the same shared directory containing video and 2 screen captures. Video was recorded before the watch update. That update also broke Wearable Widgets generated tile on the watch, still not fixed by the developer. OS upgrades are a pain in the neck for everyone. https://www.4shared.com/folder/NpIaRSHx/Juggluco_GlucoDataHandler_and_.html http://GDH-complications.4shared.com RegardsIgor From: pachi81Sent: September 29, 2023 1:37 AMTo: pachi81/GlucoDataHandlerCc: Igor-bubble; AuthorSubject: Re: [pachi81/GlucoDataHandler] Very big colored complication color changes spontaneously (Issue #94) Hi, you can also use GDrive or something like this and send me only the link to the files.So you can also include the video

pachi81, 2023-09-29:
Hi,
I don't really see a problem. It seems, that the complications are changing between the colored one from my app and the one from xDrip+ with value, delta and trend arrow. 
I think that is not an issue, it seems to be the common behavior of this watch OS or watchface. But I would not like it, too.
Maybe there is any option to deactivate this strange behavior.

Regards, Michael

Igor-bubble, 2023-09-29:
Hello Michael As I explained before, video was taken not when GDH colour was missing. You are right, video shows expected behavior, green from GDH complication and smaller white from xDrip which does not have colour coded complication. The problem I was describing was that GDH Big colored complication was displayed in white or white outlineIt is rather sporadic and I still have no illustration to show you. Thanks for your help, I will try to catch and capture it the next time. Or, if it never happens again, we both will be very happy Best of luck to you Igor From: pachi81Sent: September 29, 2023 5:48 PMTo: pachi81/GlucoDataHandlerCc: Igor-bubble; AuthorSubject: Re: [pachi81/GlucoDataHandler] Very big colored complication color changes spontaneously (Issue #94) Hi,I don't really see a problem. It seems, that the complications are changing between the colored one from my app and the one from xDrip+ with value, delta and trend arrow.I think that is not an issue, it seems to be the common behavior of this watch OS or watchface. But I would not like it, too.Maybe there is any option to deactivate this strange behavior.Regards, Michael

pachi81, 2023-09-29:
Have you tried the "Colored AOD" setting?
If this is active, GDH does not provide a monochrom image for AOD. Maybe this solves your problem.

Igor-bubble, 2023-09-29:
I do not use AOD mode Thank you Michael  Igor From: pachi81Sent: September 29, 2023 6:47 PMTo: pachi81/GlucoDataHandlerCc: Igor-bubble; AuthorSubject: Re: [pachi81/GlucoDataHandler] Very big colored complication color changes spontaneously (Issue #94) Have you tried the "Colored AOD" setting?If this is active, GDH does not provide a monochrom image for AOD. Maybe this solves your problem.

pachi81, 2023-09-30:
If you do not use AOD, then you can activate this setting, so no monochrom image is part of the complication, also in non-AOD mode.

Igor-bubble, 2023-09-30:
Thanks, will do Igor From: pachi81Sent: September 30, 2023 2:09 AMTo: pachi81/GlucoDataHandlerCc: Igor-bubble; AuthorSubject: Re: [pachi81/GlucoDataHandler] Very big colored complication color changes spontaneously (Issue #94) If you do not use AOD, then you can activate this setting, so no monochrom image is part of the complication, also in non-AOD mode.

Igor-bubble, 2023-10-05:
> Hi, you can also use GDrive or something like this and send me only the link to the files. So you can also include the video

Igor-bubble, 2023-10-05:
To continue our previous discussion, today I restarted my phone, some time later noticed that complication on the watch is empty because GDH was not started on the phone (how do I automate the startup?). When I started GDH, it came up with b&w **outline** of the reading (1.png). GDH on the watch was **outline** too (2b.png), while colour selected in settings was green (2a.png). When I changed colour to aqua (2a.png) in settings, colour of complication changed to solid white (2b.png.
[image]
[image]
[image]
[image]
[image]

pachi81, 2023-10-06:
What kind of watch do you have and what OS is installed? Any custom one?

The behavior is really strange, even the text color (first picture does not work).

Can you try the G-Watch Watchface and check, if the color is working for this app?

I really have no idea and I think it is a special behavior on your watch. Unfortunately I can not reproduce this issue, so I have no option to check and fix it...

Igor-bubble, 2023-10-06:
My watch is Samsung Galaxy watch 4 classic, latest update of Wear OS installed. I am pretty sure the colour distortion occurs when the Watch is restated. It lasted until the phone was restarted as well, so now I can rectify the situation pretty simply. My phone is Samsung S20FE, also fully updated. I tried 2 customized watch faces, Main Time and Stretched Time by Samsung. Both behaved similarly. Will try G-Watch face next time the situation arises.

Kind regards

Igor

pachi81, 2023-10-06:
That is really strange. I know some people also using Galaxy Watch 4 with Wear OS 4 and they do not have such problems.
I also using Galaxy Watch 5 Pro without any problems...
The G-Watchface has its own colored glucose value and so it would be interested for me, if this value has also the problems with the color on your watchface. 

Have you tried to re-install GlucoDataHandler?

Igor-bubble, 2023-10-06:
I did try to reinstall GlucoDataHandler to no avail. Just now I restarted the watch trying to test G-Watch, but colours in GlucoDataHandler complications stayed normal. This is pretty sporadic, do not waste your time on it. Do you want me to close the issue on GitHub?
I noticed a new version on GitHub. Should I uninstall GlucoDataHandler first or just install over the older version on the phone and the watch?

Thank you

Igor

pachi81, 2023-10-06:
You can install it over the old version. 

I will close the issue for now.

Thank you und sorry, that I could not solve your problem.

pachi81, 2023-10-07:
Hi Igor,

one more thing 😀
What font are you using and which size?
Can you change to the default one and check, if it occurs again?
Maybe it is related to the font and I have to use a fix one...

igor-xdrip, 2023-10-07:
I use Gothic Bold font style with Bold turned on and Size cranked up to a maximum. I can switch to Default if this would help you test.

pachi81, 2023-10-07:
Please switch to default, if the problem occurs the next time and wait for a new glucose value. 
I will change it on my watch to your settings an see, what will happen...
Thank you!

Igor-bubble, 2023-10-08:
Of course
Igor

Sent from myMail for iOS

pachi81, 2023-10-24:
Hi Igor,

does it happened again? Otherwise I would close it.

Regards, Michael

Igor-bubble, 2023-10-24:
Hello Michael. Yes, unexpected color change happened again. I started a new sensor on Oct 17, color in range was normal (green). in 3-4 days, it changed to white outline. I configured Samsung's Stretched Time to use "Big Colored" complication, and it was the same as in "Main Time" that i normally use. Today (Oct 24) I had my sensor fail. Both faces displayed color correctly. I am at a loss. Unfortunately, there is not much I can do to help unless there is some log you want to see. I will do anything to help, but the problem is minor, and since you did not have it reported by others, please feel free to close the issue, Thanks again for all your work

Regards

Igor

Unrelated to the issue I reported: I participate in a few Facebook groups for people with Diabetes, and noticed, that since people like myself (older and diabetic) often have eyesight problems, there is a lot of questions re. how to increase readability of watches. Do you have any plans to include apps other then Juggluco as a source in Glucodatahandler? xDrip would be an obvious first choice as it is open source (GitHub) and has by far the most users.

Thanks again

pachi81, 2023-10-25:
Hi Igor,

have you tried to switch to default font, if the error occurs?
I think this problem is not related to my app, it seems a bug on Samsungs font implementation, but I´m not sure about.
At the moment, a solution in the app could be to force using default font for the images!? What do you think about?

To your second questions, I also have facebook (but not really use it at the moment), so if the groups are in english, I can join them.
To increase the size of items in a watch-face, I recommend using watch-creators like pujie-black. I´m using it and there you can at complication elements and increase there size. So you can add a small- or large-image comlication with the glucose value yourself.

My app already support xDrip+: https://github.com/pachi81/GlucoDataHandler/blob/master/INSTALLATION.md#configure-xdrip
I also thought about supporting Dexcom Patched App, but I haven´t found any users, yet, who are interested in. So, if you know someone, please request him to create an issue here. The implementation for Dexcom Patched App is not a big thing, but I can not test it...

If you have any other sources, feel free to point me to them. The problem is to get access to the data, because the official apps from play-store mostly have no good access to them. Therefor you need to add some internet accounts to request data... 
But if I found out how, I can also do that, if someone is interested in.

Regards, Michael

Igor-bubble, 2023-10-25:
Hello Michael. I tried to find you on the Facebook to send invitation to groups I thought would benefit from introduction to your app (it was mentioned in both groups, but only in passing).  I did not succeed.  So here are the groups I meant:
https://www.facebook.com/groups/xDripG5
https://www.facebook.com/groups/libregeeks

Regards,
Igor

Log in or sign up to view<https://www.facebook.com/groups/xDripG5>
See posts, photos and more on Facebook.
www.facebook.com

pachi81, 2023-10-26:
Hi Igor,

I already posted a link: 

https://www.facebook.com/groups/xDripG5/permalink/23963219879991515/

Thank you!

Regards, Michael

pachi81, 2023-10-27:
Hi Igor,

can you try this version on your watch. I forced to use roboto as font. Can you check, if the problem still occurs or not. Thank you!
[GlucoDataHandler-Wear_0.9.6.1.apk.zip](https://github.com/pachi81/GlucoDataHandler/files/13187749/GlucoDataHandler-Wear_0.9.6.1.apk.zip)

Igor-bubble, 2023-10-27:
Can't install. Samsung S20FE. Latest update.

pachi81, 2023-10-27:
Not on the phone, on the watch ;-)

Igor-bubble, 2023-10-27:
Sorry, I was still half-asleep and did not pay attention to the file name. Now I awoke and here we are:

Before installation I restarted the phone and Very Big Colored showed in correct colour. After installation complication remained coloured, but when I displayed GDH on the phone, it was b&w , and complication changed to b&w. Stayed b&w after watch restart as well.

Igor

pachi81, 2023-10-27:
Ok, so it does not solve the problem.
Than I have no idea...

Can you try the default font and size and check if it also occurs.

Igor-bubble, 2023-10-27:
I tried the following:

  1.  Changed font default and font size to smaller and not bold. VBC complication stayed b&w
  2.  Restarted the watch. VBC came up coloured
  3.  Started GDH on the watch. Shortly afterwards VBC changed to b&w
  4.  Changed font Samsung and max size. b&w
  5.  Restarted the watch. VBC coloured
  6.  Started GDH. Value displayed was b&w and shortly after VBC turned b&w
  7.  Restarted the watch. VBC colored will not display GDH on the watch to see how long colour will stay

pachi81, 2023-10-27:
Really really strange! 
And I have no idea. Have you a lot of content on the watch? Is it possible to reset the watch and start installation from the beginning 🙈
But the biggest problem is, I can not promise that it will work afterwards. Or do you have another watch for testing? As you are the only one with such a problem, maybe it is related to the watch, but I don't really know it and I don't know... 🤔

Sorry! ☹️

igor-xdrip, 2023-10-27:
Sorry, I have a lot of manually configured stuff on my watch, so resetting the watch for a relatively minor problem is not something I relish. To continue testing, I undertook the following further steps after about 7-8 hours (colour survived):
 8. Restarted the watch; after some delay colour VBC complication
 9. Started GDH **on the watch**. It came up with b&w value and colour trend arrow (this is probably graphics, not font?)
 10. When I returned to watch face, complication was b&w.
 11. Restarted the watch. Colour of VBC restored. 
 12. Changed colour in GDH settings on mobile. Colour on the watch changed accordingly and stayed.

My solution is: **_do not run GDH app on the watch, if run is required, restart the watch to restore colour_**

igor-xdrip, 2023-10-27:
Please, feel free to close the issue 👍

pachi81, 2023-10-27:
Ok, it was just a question 😉

But you do not have to start GDH on watch as it will be running in background and app itself does not provide a lot of information. 

But it is really strange... and I can't stop thinking about...

Maybe I can have a look in the logcat from my app on your watch, but it is not urgent.

igor-xdrip, 2023-10-28:
> Ok, it was just a question 😉
>
> But you do not have to start GDH on watch as it will be running in background and app itself does not provide a lot of information.
>
> But it is really strange... and I [...]

Yes, there is no need to run GDH on the watch, but running it changes colour of the complication, must be a bug. I'd gladly provide you with the logcat from my watch, but do not know, where to find it.

igor-xdrip, 2023-10-28:
> Hi Igor,
> 
> can you try this version on your watch. I forced to use roboto as font. Can you check, if the problem still occurs or not. Thank you! [GlucoDataHandler-Wear_0.9.6.1.apk.zip](https://github.com/pachi81/GlucoDataHandler/files/13187749/GlucoDataHandler-Wear_0.9.6.1.apk.zip)

Michael, I just realized that I did not mention that I did update GDH on my watch with the file you sent me, and all the testing described in the list above was done after it. However the "roboto" font you mentioned is not present on my Samsung watch, so the update probably had no effect. My watch was set to use SamsungOne font during testing. 
Is it not possible to somehow exclude GDH icon from app screen on the watch?

pachi81, 2023-10-28:
Yes, it is stange the arrow icon has not changed on my galaxy watch, too. But the complication value of VBC should use it, now.
After some re-install of the app, the font has been used, without changing anything...

I think, Samsung has some problems with creating bitmaps from text. But for the most galaxy watch users it is working, except you.
So maybe your watch has a bug.

But I can create a version without icon in the app and you can verify.
But the value of VBC is rendered the same way, because the complication does not support colored text.

pachi81, 2023-10-28:
For gettings logs of the watch, you need a computer with adb connected to your watch.
The type:
`adb shell "logcat --pid=$(pidof -s de.michelinside.glucodatahandler)"`

But I'm in holiday, now. So I haven't test the statement and can't say, if it is correct.

igor-xdrip, 2023-10-28:
> But I can create a version without icon in the app and you can verify. But the value of VBC is rendered the same way, because the complication does not support colored text.
If it is only for my sake, dont bother, I will simply not invoke it. If you want to implement it anyway, I am ready to test.

pachi81, 2023-12-25:
Hi,
merry x-mas!
Just one question, have you acitvated "High contrast fonts"?
If yes, can you deactivate it to verify, if this causes the problem.
Thank you!
[image]

igor-xdrip, 2023-12-26:
Merry Christmas and happy New Year to you and yours!
This was a great find. High contrast on= b&w font in complication, off= Christmas colour font. You are nothing if not persistent my friend.

Sent from myMail for iOS

Igor-bubble, 2023-12-26:
P.S. Here are a bit more details: fonts were in colour when I started checking. Turned off high contrast and colours remained. Ran GDH on Watch, fonts changed to b&w.. turned off high contrast and fonts changed to colour again, without restarting the Watch.

pachi81, 2024-01-16:
Unfortunately I do not find an option to create colors ignoring the high contrast setting...
Everything I have found does not work :-(
So I create a detection of this settings and show a message to the user to disable it for having colors. It will be available in next releast 0.9.9
Thank you!

closed 2024-01-16 by pachi81

igor-xdrip, 2024-01-18:
That’s a very usable solution. If you need a tester, just ask

Sent from myMail for iOS

pachi81, 2024-01-18:
You can join the open beta testing or wait for the release. I think it will be available next week.