pachi81/GlucoDataHandler #98: Pixel 8 Pro can't install

ChrisJSLH, 2023-10-12:
Hi, I can't install on Pixel 8 Pro. Says device isn't compatible.

pachi81, 2023-10-12:
Hi, which firmware version?
Maybe I have to increase the maximum supported version.
I have to check. Hopefully no deprecated or removed function, then it can take some more time.

pachi81, 2023-10-12:
I have red, that it should work also on Android 14.
Can you try to install using adb?

ChrisJSLH, 2023-10-13:
It works on Android 14 for sure. I had it installed on my Pixel 7 before and after the upgrade to Android 14 and no problems. Although only installed it on Android 13 before the upgrade.

The exact message is "App not installed as app isn't compatible with your phone".

pachi81, 2023-10-13:
I also expect that it should be working in Android 14. I have red in the docu, that Android 14 supports app with a target SDK of 23 and my app uses target SDK 33. 
If possible, please try to install it using adb from a computer:
`adb install -r -g GlucoDataHandler_0.9.6.apk`

I hope that adb will offer a better error message. 

Or you can try the fix of this video:
https://www.youtube.com/watch?v=ZYx9NEi1t2c
(Have you disabled Play Protect already?)

pachi81, 2023-10-18:
Hi, 
any news? Did you get it to work or do you have still the problem?

ChrisJSLH, 2023-10-18:
Still no joy. I can't get it to install from the phone and when I try and push it from adb it complains about a missing wear dependency. (This may be due to me not having the full SDK installed, just minimal ADB).

pachi81, 2023-10-18:
Can you post the whole adb error message. Have you tried to disable Play Protect?

ChrisJSLH, 2023-10-18:
I'll send the full adb error message later when I'm at home. Yes, definitely tried to disable Play Protect.

ChrisJSLH, 2023-10-18:
> One more question, are you using the apk without "Wear" in the name?

ADB Error Message:
Failure [INSTALL_FAILED_MISSING_SHARED_LIBRARY: Reconciliation failed...: Reconcile failed: Package de.michelinside.glucodatahandler requires unavailable shared library com.google.android.wearable; failing!]

 

Yes, definitely trying to install the Wear version.

pachi81, 2023-10-18:
You try to install the wear version on mobile?
That is not possible, you have to use the version without wear.

ChrisJSLH, 2023-10-18:
Omg. I was getting confused with G-Watch Wear App which has Wear version and a Tizen version. What an idiot I am...

pachi81, 2023-10-18:
😁
No problem, it happend to me, too...🤣

pachi81, 2023-10-24:
Close it as solved.

closed 2023-10-24 by pachi81