pachi81/GlucoDataHandler #100: license question

IzzySoft, 2023-10-26:
First thanks for providing your app here! May I ask what license it falls under? This repo doesn't declare one.

pachi81, 2023-10-26:
Hi IzzySoft,

there are too many license, I can´t decide ;-)
I think the normal is the apache 2.0, but I have to check it first.

Why are you asking? Do you want to change my app?

Regards, Michael

pachi81, 2023-10-26:
Ok, Apache is not a good one, as it allows to use my code and sell it...
I have to check, but this will take some time.

pachi81, 2023-10-26:
Apache seems not a good choice as it allows the sell my app. 
At the moment, I´m thinking about:
[CC BY-NC-SA 4.0](http://creativecommons.org/licenses/by-nc-sa/4.0/?ref=chooser-v1)

As I´m not so familiar with all the licenses, what do you think? Why are you asking?

Regards, Michael

IzzySoft, 2023-10-26:
The NC ones don't count as "libre" (violating the freedom to use/change/distribute for any reason, see [The Four Essential Freedoms of Free Software](https://en.wikipedia.org/wiki/The_Free_Software_Definition#The_Four_Essential_Freedoms_of_Free_Software). To give you a selection of options, you can check with [SPDX](https://spdx.org/licenses/): those approved by OSI/FSF are counted as "libre".

Reading between the lines (guessing what makes you consider NC), you'd probably like AGPL – which requires to open-source any changes even when no binaries are distributed but the code is used for a backend service). By the way, CC licenses are less suited for source code – they are rather intended for documentation, assets and media.

> Why are you asking?

I was considering adding your app to [my repo](https://apt.izzysoft.de/fdroid) (due to the dependency on GMS it would not be acceptable at [F-Droid](https://f-droid.org/) itself) to make it easier available to those outside of Google's walled gardens. A FLOSS license is one of the requirements for that (more details can be [found here](https://gitlab.com/IzzyOnDroid/repo/)).

pachi81, 2023-10-26:
Wow, great work!
Haven´t heart about, yet. I only know F-Droid. At the moment I´m planning to go to Google Play, but therefor I have to change some content, because of the restrictions (access battery optimization and target SDK for watch).

Related to the license. In the end it is ok, if others uses my code. 

The GNU license is to restrictive, as anybody only can use my code, if there also uses GNU.

So maybe, still using Apache 2.0 would be the easiest one and I do not have to take care :-)

pachi81, 2023-10-26:
After some checking and discussions with a friend, the best choice for open source is MIT, as Apache 2 does not contain any advantage for my type of project.

IzzySoft, 2023-10-26:
Ha! LOL – I just wanted to recommend you MIT, based on what you wrote. Yeah, that is fine! I'd also say forget about G (no such restrictions with F-Droid or with my repo) – but thanks to the Wear part you'd have no chance for a listing at F-Droid.org, and while my repo is already quite popular it's even smaller.

OK, then let me see if I can get your app in. MIT is set, there will be an anti-feature needed due to Wear and what it drags in – but apart from that it fits the inclusion criteria now:

```
Offending libs:
---------------
* Android Wear APIs (/com/google/android/gms/wearable): NonFreeComp
* Google Mobile Services (/com/google/android/gms): NonFreeComp
* Tasker Plugin Library (/com/joaomgcd/taskerpluginlibrary): NonFreeAdd
* Tasker (/net/dinglisch/android/tasker): NonFreeAdd

4 offenders.
```

GMS is dragged in by Wear – and the two Tasker libraries I'd count as one ("Tasker/Locale support") – that makes 3 < 4 = acceptable :smiley: Signature checks out (hoppla, Grüße nach nebenan! OK, 300 km…), and… oops:

    package: name='de.michelinside.glucodatahandler' versionCode='1' versionName='0.9.6'

`versionCode='1'`? This isn't the initial release, is it? I see 8 more there, so it should be `9` or higher. I can put it now as-is, but only if you promise to [increase it](https://github.com/pachi81/GlucoDataHandler/blob/master/build.gradle#L8) with each new version from now on. It's what Android uses internally to keep versions apart – and to tell if an APK provides an update to the installed version of an app or not. PlayStore won't accept updates either if you didn't increase this. Can we agree on that?

pachi81, 2023-10-26:
Hi,

> hoppla, Grüße nach nebenan! OK, 300 km…

Da können wir das ja auf deutsch machen 😉
Was bedeuted nebenan?

> Tasker libraries

Die Schnittstelle ist optional und meine App setzt Tasker nicht voraus.

> versionCode='1'? This isn't the initial release, is it? I see 8 more there, so it should be 9 or higher. I can put it now as-is, but only if you promise to [increase it](https://github.com/pachi81/GlucoDataHandler/blob/master/build.gradle#L8)

Ich schwör 🤚🤣
Ist mir auch bewusst, nur solange man nicht verpflichtet ist, habe ich es absichtlich nicht gemacht, damit man jederzeit downgraden kann.

Da du dich auskennst, ich habe für die App erstmal einen eigenen Key erstellt. Muss ich für den Playstore einen neuen über den Playstore generieren? Dann sollte man die App noch nicht in ein Store bringen, da das ja dann nicht mehr kompatibel ist...

IzzySoft, 2023-10-26:
> Was bedeuted nebenan?

München. Ich hatte bei Dir zuerst "Leoni" gelesen (Freud'scher Verleser :see_no_evil: das wäre direkt am Starnberger See gewesen, also S-Bahn Bereich) und erst dann das "St. Leon-Rot" gesehen…

> Die Schnittstelle ist optional und meine App setzt Tasker nicht voraus.

Hm, mein Scanner findet die beiden und jammert dann. Fragen wir mal so: Wird das "groß angepriesen", oder findet man es eher wenn man in den Settings, der Doku etc. sucht bzw. in Tasker "spielt"? In der Readme steht es ja auch nur "ganz unten". Also keine Pop-ups a la "Kannst auch Tasker…"? Dann würde ich die beiden Augen zudrücken, und die Warnung "übersteuern" (also quasi als "false positive" markieren).

> Ich schwör 🤚🤣

:rofl: Eine halbe Stunde früher, und es wäre vielleicht jetzt schon mit dem Sync hochgegangen – so muss es halt bis morgen warten.

> habe ich es absichtlich nicht gemacht, damit man jederzeit downgraden kann.

OK, solange man ohnehin manuell nach updates an den Tags suchen muss… Aber für ein "offizielles Listing" ist es ein Muss – sonst bekommt niemand Update-Benachrichtigungen.

Für den Spielzeugladen kannst Du m.W. den gleichen Key verwenden. Kugel verwendet dann einen so genannten "Upload-Key" und signiert intern sowieso um (frag mich nicht was das soll: als F-Droid selbst signierte war das böse, aber wenn G das macht ist das toll…) Bleibt damit kompatibel. Ich pinne übrigens das Zertifikat hier dann, sodass APKs nur akzeptiert werden, wenn sie mit eben dem selben Schlüssel signiert wurden.

IzzySoft, 2023-10-26:
OK, geht dann [morgen hier live](https://apt.izzysoft.de/packages/de.michelinside.glucodatahandler/) (gegen 20 Uhr). Preview:
[image]

pachi81, 2023-10-26:
Wow, sieht gut aus, bin jetzt etwas stolz 😁😎

Wie genau funktioniert das mit dem F Droid Client? Gibt es da auch ne App für Wear OS?

IzzySoft, 2023-10-26:
Nein, WearOS ist proprietär. Deswegen taucht das bei Deiner App auch als `NonFreeComp` auf: bei F-Droid.org gar nicht erlaubt so etwas – unfreie Komponenten in einer App. In meinem Repo "begrenzt geduldet". Mein Repo ist quasi eine Ergänzung zu dem von F-Droid, u. a. für die "close-but-no-cigar" Apps (die also die Kriterien von F-Droid nur *fast* erfüllen – aber aus Gründen entweder nie oder noch nicht erreichen können. Wie Deine mit Wear-Komponente).

Nebenbei: Wenn Du möchtest, kann ich Dir die erstellten Metadaten (Beschreibung, Bilder) per PR im Fastlane-Format zur Verfügung stellen. Dann kannst Du die entsprechend anpassen (Beschreibung erweitern, vielleicht auch per-Release Changelogs ergänzen) und das wird dann von meinem Updater bei jedem neuen Release von Dir mit gezogen. Mein [Fastlane Cheat Sheet](https://gitlab.com/-/snippets/1895688) kann Dir bei der Orientierung helfen – und Du kannst mich natürlich auch fragen, wenn Du irgendwo "festhängst".

pachi81, 2023-10-27:
Moin,

ja, das Fastlane kannst du gerne zur Verfügung stellen. D.h. sobald ich eine neue Release hier hochlade, wird das automatisch bei dir gezogen und entsprechend angezeigt. 👍 
Nur releases oder auch pre-releases?

Was für einen Client kann ich verwenden, dass er es auf meinem Telefon dann automatisch aktualisiert?

IzzySoft, 2024-01-07:
Oof, strange. Looks like I didn't receive a notification for your comment – but got one for your closing this issue? Woops…

OK, PR ist offen. Ich sollte eine Benachrichtigung erhalten wenn Du mergest, dann würde ich das bei mir aktivieren – und bei jeder Release würde mein Updater dann hier auch auf Änderungen prüfen. Ein kurzer Hinweis wenn Du neue Elemente hinzufügst, die es vorher noch nicht gab (z.B. gibt es jetzt noch keine per-release changelogs) wäre hilfreich, darauf prüft sonst nur ein monatlicher Job.

> Nur releases oder auch pre-releases?

Derzeit sind für die App nur Releases aktiviert.

> Was für einen Client kann ich verwenden, dass er es auf meinem Telefon dann automatisch aktualisiert?

Einen beliebigen [F-Droid client](https://android.izzysoft.de/applists/category/named/apps_markets#group_1181) Deiner Wahl :smiley: Wobei ich nicht für jeden davon sagen kann, welche Details er anzeigt. Ich selbst nutze auf manchen Geräten den offiziellen Client, auf anderen NeoStore. Droidify soll auch recht umfangreich sein.

Und Du kannst auch gern die Google-Badge um eine von mir ergänzen, die auf Deine App in meinem Repo verlinkt: [pick a badge](https://gitlab.com/IzzyOnDroid/repo/#if-my-app-is-available-here-do-you-have-a-badge-i-can-use-to-link-to-it)

pachi81, 2024-01-08:
Hi,
Was verstehst du unter per-release changelog? Ich trage das Changelog immer direkt beim Release hier mit ein...
Braucht es da noch eine extra Datei?

IzzySoft, 2024-01-08:
Siehe [Fastlane Cheat Sheet](https://gitlab.com/-/snippets/1895688). Ja, ist eine extra Datei pro Release. Deren Inhalt würde dann bei einigen/den meisten Clients direkt angezeigt, oft direkt unter der App-Beschreibung (auf meiner Website oder im offiziellen Client) oder in einer extra Sektion unterhalb (NeoStore). Ist optional, es "fehlt" also nichts. Aber ein "really nice to have" für diejenigen, die vor einem Update schauen wollen :wink:

pachi81, 2024-02-22:
Ich schließe das mal, da es aktuell funktionieren sollte.