pachi81/GlucoDataHandler #109: Add Support for BangleJS

phrogg, 2023-11-29:
Gude,

I've made an app a while ago, to be able to send blood glucose values to a BangleJS watch. I think this bit of code would be beneficial for your projet, or do you want to only support googles war os? If you are interested in this, give me a ping and I will create a pull request in the future. :) Here is a link to my app: https://codeberg.org/phrogg/BG2BangleJSApp already in F-Droid. 

Alles jute
Phil

pachi81, 2023-11-29:
Hi Phil,

I never heard about BangleJS watches, but why not... 
Just for information, is there any watchface, you will need for that? 

As I have seen, there is only an intent to send for the the BangleJS. 
This is not a big thing and I can implement.
I´m still planing to move the receivers out of the current service to an extra class. While doing that I can also add these lines.
But if you like, you can do it also...

Regards, Michael

phrogg, 2023-11-30:
> Hi Phil,
>
> I never heard about BangleJS watches, but why not... Just for information, is there any watchface, you will need for that?
>
> As I have seen, there is only an intent to send for the the [...]

However, you like. I would be glad to help :)

pachi81, 2023-11-30:
Can you explain, what exact is BangleJS and how this intent work on the watch? Is there a special widget or anything like this?

pachi81, 2023-12-20:
Hi, just for your information, I have not forgotten you, but I had a lot of problems getting the app in the playstore.
I don't think, that I can add it this year, but the is the next point on my list. As soon I can continue working on the app, you will get a version to test.
Regards, Michael

pachi81, 2023-12-21:
Hi Phil,

I had some time :-)
Here you can download and test the app:
https://1drv.ms/f/s!AvqtWFwQmBjwz6cnbx5R-92cIBDUbg

You find the setting at the bottom in the settings-menu.
This is what I'm sending:
GDH.GlucoDataServiceMobile : Send to bangleJS: require("Storage").writeJSON("widbgjs.json", {'bg': 145,'bgTimeStamp': 1703187989326,'bgDirection': 'FortyFiveDown'});

Is this ok? What about mmol/l?
Do you convert it or should I send it in bg value.

phrogg, 2023-12-22:
Hey, nice work. I will try it in the new year, since I don't have my watch with me right now. The code u send looks good. mmol/l will be calculated on the watch itself, no need to worry about that.

pachi81, 2024-01-10:
Hi Phil, 

had you some time for testing?

Regards, Michael

phrogg, 2024-01-18:
> Hi Phil,
> 
> had you some time for testing?
> 
> Regards, Michael

Hey sorry, not yet. Maybe on the weekend. I'm currently working a lot on my master's thesis and at work.

Sorry.

pachi81, 2024-03-13:
Hi, that makes no difference. 
Maybe it is related to the format of my value.
I will check later. 
But then I need you for testing...

pachi81, 2024-03-13:
If I compare your implementation with my, I do not see a real difference.
Can you check:
https://codeberg.org/phrogg/BG2BangleJSApp/src/commit/dcee7c30d36fb104b5db20504f6af9dab1be0664/app/src/main/java/de/phrogg/bg2banglejs/DataReceiver.java#L29
https://github.com/pachi81/GlucoDataHandler/blob/6382dccb6cb665489b4f1ef8c6fd5e1d894ff880/mobile/src/main/java/de/michelinside/glucodatahandler/GlucoDataServiceMobile.kt#L110

closed 2026-01-19 by pachi81