pachi81/GlucoDataHandler #114: Retrieve IOB, COB and some values aaps/loop from Nightscout

lostboy86, 2023-12-31:
Good holidays at all,

on widget or complications, can be retrieve iob, cob, etc from nightscout?

So many aaps user groups with android watch, or when drive, ask this function, nightscout provide this parameteres, uploaded by aaps.
Maybe display this values on widget mobile or complications watch?

Thanks

pachi81, 2023-12-31:
I can check, but I do not loop, so I do not know, what these values are used. Can you provide a complete list with their meanings?
Do you know, which nightscout endpoint supports these values?

lostboy86, 2023-12-31:
Support this values xdrip, androidaps, loop for ios (iAPS), many other cgm
apps

IOB : insulin on board active
COB : cho on board active

pachi81, 2024-01-01:
Do you know  which ns endpoint includes these data?
What unit and value range is used for?

lostboy86, 2024-01-01:
Umh what is ns endpoint?

Range are 0.00 - infinite

lostboy86, 2024-01-01:
@psonnera can explain well this parameters

psonnera, 2024-01-01:
@lostboy86 explain well is high expectation but you'll find some background here https://github.com/nightscout/cgm-remote-monitor?tab=readme-ov-file#iob-insulin-on-board but for more I need to seat in from of my computer (i.e. for an example query).

pachi81, 2024-01-01:
I have to check nightscout endpoints get the correct one, which offers these data. One question, can you test it, if I add it?

Another question, are both values need together or some people are using only iob or cob? Which is more important?

lostboy86, 2024-01-01:
Yes i can test, i use AaPS also and ns 15.

Need toghether, iob it's the insuline Active on board , cob it's the cho on
board Active.

pachi81, 2024-01-02:
Can you check the results of the nightscout API endpoints to check, which one contains the IOB and COB:
[NS_URL]/api/v1/entries/current.json
[NS_URL]/api/v1/entries/sgv.json?count=1
[NS_URL]/api/v1/treatments.json

-> [NS_URL]: your nightscout address, for example https://127.0.0.1 for localhost.

lostboy86, 2024-01-02:
Also with web ns?

pachi81, 2024-01-02:
Yes, it should work with every ns instance.

lostboy86, 2024-01-02:
i Think its in the treatment.

here's the link, check yourself, its open ns.

lost.soon.it/api/v1/treatments.json

lost.soon.it/api/v1/entries/sgv.json?count=1

lost.soon.it/api/v1/entries/sgv.json?count=1

pachi81, 2024-01-02:
Thank you! Great I can test it with your server. Just one more question. Are the IOB and COB time related? What do you expect to be visible on watch? Only the last value or also the timestamp?

pachi81, 2024-01-02:
I have checked your treatments.json and found different kind of eventType:

- "Temp Basal" seems to be the current basal value (-> ignored)
- "Meal Bolus" carbs you have ate (-> ignored)
- "Note" some note, don´t know (-> ignored)
- "Profile Switch" ? (-> ignored)
- "Temporary Target" ? (-> ignored)
- "Carb Correction" (-> ignored)
- "Bolus Wizard" seems to be the only one, containing data to analyze:
```
{
    "basalIOB": 0.769,
    "bolusIOB": 0.0,
    "carbs": 90.0,
    "carbsInsulin": 7.317073170731707,
    "cob": 0.0,
    "cobInsulin": 0.0,
    "dateCreated": 1704192471335,
    "glucoseDifference": -31.0,
    "glucoseInsulin": -0.62,
    "glucoseTrend": 0.0,
    "glucoseValue": 69.0,
    "ic": 12.3,
    "id": 112,
    "ids": {},
    "isValid": true,
    "isf": 50.0,
    "note": "",
    "otherCorrection": 0.0,
    "percentageCorrection": 90,
    "profileName": "Regolato 30/12/23 00:16",
    "superbolusInsulin": 0.0,
    "targetBGHigh": 100.0,
    "targetBGLow": 100.0,
    "timestamp": 1704192471302,
    "totalInsulin": 6.7,
    "trendInsulin": 0.0,
    "utcOffset": 3600000,
    "version": 0,
    "wasBasalIOBUsed": true,
    "wasBolusIOBUsed": true,
    "wasCOBUsed": true,
    "wasGlucoseUsed": true,
    "wasSuperbolusUsed": false,
    "wasTempTargetUsed": false,
    "wasTrendUsed": false,
    "wereCarbsUsed": false
}
```

cob seems to be 0.0
iob, which one? 
Is the format a common one, or does any pump has its own format?
If I check for this eventtype, there are only some results per day:
https://lost.soon.it/api/v1/treatments.json?find[eventType]=Bolus+Wizard&count=5
Is this correct?

pachi81, 2024-01-02:
Ok, now I make official: I hate the nightscout api docu!!! There is a better docu for the closed librelinkup and nightscout is a open source software!

There is only one(!!!!) github issue, which points to the correct url.
The last iob you can find here:
https://lost.soon.it/api/v2/properties/iob
and the last cob here:
https://lost.soon.it/api/v2/properties/cob

lostboy86, 2024-01-02:
Jesus Christ, great result, i try to find myself but no success

pachi81, 2024-01-02:
One more question. Are you also using xDrip+ for your values, IOB and COB?
Because xDrip+ offers the same endpoint for pebble as nightscout, which also supports IOB and COB:
NS: `https://lost.soon.it/pebble`
xDrip+: `http://127.0.0.1:17580/pebble`

So, if you have xDrip+, you can also use it "offline" for getting these values.

And I have to add some options to get the values from the different sources.

lostboy86, 2024-01-02:
@psonnera can support for xdrip

psonnera, 2024-01-02:
@pachi81 never tried to grab iob or cob from xDrip+, not sure they're provided either... https://github.com/NightscoutFoundation/xDrip/blob/master/Documentation/technical/Local_Web_Services.md

lostboy86, 2024-01-02:
Also, maybe my answer above was wrong, iob and cob maybe are time-relate,
because AAps send every 5 minutes, correct?

psonnera, 2024-01-02:
Good question... probably... I never tried the 1 minute BG. AAPS is still 5 minutes based AFAIK. Does it upload BG from Juggluco minute by minute in Nightscout?

lostboy86, 2024-01-02:
Wait, Glucodatahandler retrieve BG every minutes From juggluco, no problem.

But if display on widget the iob and cob, maybe Is Better to display some
info about the time of there?

For example :

104 --∆ 2 mins

IOB : 6.2 UI -- 4 minutes ago
COB : 34 CHO -- 4 minutes ago

Or something like this?

pachi81, 2024-01-02:
This is too long for complications...
And the pebble endpoint only supports one timestamp for all values...

@psonnera do you use IOB and COB in xDrip? Can you check the pebble endpoint?
I have tested it and it only support iob, but no cob. But I do not use loop...

lostboy86, 2024-01-03:
For the question of too long : 

104 trend arrow
I : 6.2 | C : 35 | 5ago

Same other watch complications.

psonnera, 2024-01-04:
@pachi81 not really because xDrip+ doesn't use AAPS data directly. Actually I don't know if the endpoint returns xDrip+ calculation (probably) instead of the real values sent by AAPS.

psonnera, 2024-01-04:
[image]

pachi81, 2024-01-04:
Ok, thank you! Not so important, I will change my endpoint to the pebble one and so it can either used for nightscout or xDrip+.

But at the moment, I do not have much time, so it will take some time until I can provide a test version. But with the endpoint of lostboy86 I can test it.

lostboy86, 2024-01-05:
Good, test whatever you want with my ns

pachi81, 2024-01-12:
I created a version with iob and cob support for nightscout.
Can you test it.
I uploaded to playstore for internal beta tests.
At the moment it only supports complications, no widgets....
You can also install from here:
https://onedrive.live.com/?authkey=%21AG8eUfvdnCAQ1G4&id=F01898105C58ADFA%211299367&cid=F01898105C58ADFA

lostboy86, 2024-01-12:
I have some medical problems in this day.

I test now on my phone mother.

All ok From main app

Fix :

Not show cob and iob on lockscreen notification

In main app, iob must be two values After point (es. -0.10 instead -0.1)

Add unit to cob and iob (add gr. For grams and UI for con)

Great job man!

lostboy86, 2024-01-12:
See this notification for example :

pachi81, 2024-01-12:
Yes, at the moment only for complications. 
Notification, widgets and so on will come later.
Step by step ;-)
Units I did not know yet...
2 decimal place I also did not know yet.
As I do not use it  I need your input.

pachi81, 2024-01-12:
And get well soon!

lostboy86, 2024-01-12:
Tell me what you need also.

Thanks

pachi81, 2024-01-13:
I created a new version with notification support.
I added units like in AAPS and nightscout: IOB: "U" and COB: "g"

lostboy86, 2024-01-13:
In previous link i only see 0.988 dev

pachi81, 2024-01-13:
Playstore!?

pachi81, 2024-01-14:
@lostboy86 Did you get the latest version 0.9.8.9 from PlayStore? As you are set as internal tester, you should get the newest version, first... (I do not mean the open beta test, this comes later, after google checked it...)

lostboy86, 2024-01-14:
Can you share the apk for my mother phone?

Because i have 9.8.9 but my Glucodatahandler take value From juggluco and
send locally to aaps, cannot retrieve iob and cob

lostboy86, 2024-01-14:
Tested, work correctly

pachi81, 2024-01-14:
Do you still need it?
I haven't built an apk.
I will do with the next version.

lostboy86, 2024-01-15:
Yes because cannot test iob and cob on phone parents and android radio.

On my phone i have Glucodatahandler source juggluco wear and send
interapp.from Glucodatahandler to AAps.

Is there a way to have local source and internet download iob and cob?

Or , AAps can send iob cob etc tò xdrip protocol to xdrip.

Can Glucodatahandler received them From APS locally?

pachi81, 2024-01-15:
You can try to add nightscout as "second" source to you phone with some delay.
If iob and cob are present and have changed, it will overwrite it and the other values should stay the same.

Can you test it?

lostboy86, 2024-01-15:
Umh can you explain well?

Yesterday i setup my Glucodatahandler phone  source not juggluco , but
nightscout.

Stopped APS not recept BG From Glucodatahandler, and done the test

I send the photo in last mail of Yesterday

pachi81, 2024-01-15:
I did not receive any photo (maybe Github removed attachments...).

But you can use your phone GDH to receive values from Juggluco and you can also setup nightscout at the same GDH.
For normal, GDH will ignore the values from nightscout if there not newer than the local ones from Juggluco, but if iob or cob has changed, it force update the values includings iob and cob.

So, that mean, that you can use two sources on your phone GDH, one for Juggluco and one for NS.

There is one thing I do not know, if GDH receives the values twice, it will also forward each time. Not sure, if AAPS handle this correct, normally it should ignore the second value sent with the same timestamp....

lostboy86, 2024-01-15:
Glucodatahandler can recept some values trough xdrip broadcast?

AAps can send values trough xdrip broadcast protocol interapp

pachi81, 2024-01-15:
Yes, GDH can also receive values from xDrip, but this broadcast does not include iob and cob.
You can also connect to the xDrip+ local webserver as nightscout source.

lostboy86, 2024-01-15:
Not From xdrip, but From AAps trough xdrip broadcast

AAps can receive BG From xdrip, Glucodatahandler, juggluco and can send
treatments, iob, BG, etc tò nightscout or to xdrip app, trough xdrip
broadcast protocol

lostboy86, 2024-01-15:
> Can configure any target app in AAPS? It can be, that AAPS sends the xDrip
> broadcast only to xDrip-package (app).
> What for a source do you have to specify in xDrip for such a broadcast?
> Maybe I can add to GDH, but only, if AAPS does not specify a package or if
> you can select a package (like in Juggluco) for sending broadcasts.

xDrip+ settings — AndroidAPS 3.2 documentation
<https://androidaps.readthedocs.io/en/latest/Configuration/xdrip.html>

Never use xdrip with aaps

pachi81, 2024-01-15:
These are the xDrip-settings. I mean, what for settings AAPS provide to send data to xDrip?
I have checked the source code and it seems, that AAPS sends IOB and COB in one broadcast without specify a target, so maybe I can receive it.
AAPS also sends glucose values in an extra broadcast, where you have to specify receivers. 
Can you check AAPS for such settings. I can add AAPS as source in GDH, but I can not test it...
So you have to find out, if there is any setting in AAPS to send data to xDrip.

Have you checked using Juggluco + Nightscout in GDH to receive values and IOB/COB?

lostboy86, 2024-01-15:
Here are the screens of settings AAps.

pachi81, 2024-01-15:
Sorry, but the screenshots are filtered out by github. Can you send them to my private mail?

closed 2024-01-18 by pachi81

lostboy86, 2024-01-18:
> completed via #118 <https://github.com/pachi81/GlucoDataHandler/pull/118>.

All works in AA radio, widget, cob iob etc

pachi81, 2024-01-18:
Great!!! Thank you for testing and feedback!

 
 

Gesendet: Donnerstag, 18. Januar 2024 um 10:18 Uhr

PR #118 (2024-01-18): Develop
Resolves #114