pachi81/GlucoDataHandler #115: mmol/l unchecked constantly

alexey6600, 2024-01-07:
Option "Use mmol/l" constantly unchecking. 
For last day- twise. 
Using last version from Play store ( 0.9.8)

pachi81, 2024-01-07:
Which source are you using?
For xDrip and nightscout I already have created a bugfix:
https://onedrive.live.com/?authkey=%21AG8eUfvdnCAQ1G4&id=F01898105C58ADFA%211299367&cid=F01898105C58ADFA

Can you check it? If it occurs again, can you make a screenshot, where I can see the value?
Thank you!

alexey6600, 2024-01-07:
Source is XDrip+.
Thanks.  I'll try big fix

pachi81, 2024-01-07:
Really strange! Your are the second one with this issue. The other one had the same problem, but mmol was checked once a day and she is using mg/dl.
Maybe there is a special broadcast, which causes the problem.
Have you detected a special value, when it happened?
I think I have to remove the auto-detection. Only for the first value...

alexey6600, 2024-01-07:
ActuallyI just installed it yesterday. And not detect a special value, when it happened.
Just installed version 0.9.8.3, and keep watching.
Thanks a lot! 
Really good software in combination with Wear watch!

pachi81, 2024-01-07:
Thank you for testing! Is it hard to fix something, I can not reproduce...
But I have changed the handling dor xDrip, that it should not happen. 
But I'm still not sure, what causes the problem...

alexey6600, 2024-01-07:
I will definitely write about the test results!

alexey6600, 2024-01-09:
With new version there are no problems with unchecked option.
But found new issue (on attachment )
[image]

pachi81, 2024-01-09:
What happened? The screenshot is gone!? Maybe that was the reason for the problem with unchecking. There seems to be a message without value. I also have activated xDrip as source, but have not had this problem until now.
I also check in the next version (not online, yet) also for the existence of the value.

alexey6600, 2024-01-09:
Removed, as problem gone. 
But problem appeared again. 
[image]

pachi81, 2024-01-09:
Do you know how to use logcat or do you have any logcat client on your phone?

alexey6600, 2024-01-09:
Know only with ABD

pachi81, 2024-01-09:
Can you create logcat for my app? Maybe it contains the problem, just have a look, if the time of the problem is inside...

alexey6600, 2024-01-09:
Maybe there are application for this ?

alexey6600, 2024-01-09:
[logcat.txt](https://github.com/pachi81/GlucoDataHandler/files/13874923/logcat.txt)

pachi81, 2024-01-09:
Strange, it contains a lot of logs of my app, but only from Android System, not from my app itself.
I think you have to give the app the permission to read all logs.

`adb shell "pm grant com.dp.logcatapp android.permission.READ_LOGS && am force-stop com.dp.logcatapp"`

See here for more details: https://github.com/darshanparajuli/LogcatReader

alexey6600, 2024-01-09:
I used adb with grep on macos) It's simpier)

pachi81, 2024-01-09:
Try `adb shell logcat | grep GDH.`

pachi81, 2024-01-09:
Or using pid (not tested!)
```
adb logcat --pid=`adb shell pidof -s de.michelinside.glucodatahandler`
```

alexey6600, 2024-01-09:
[logcat2.txt](https://github.com/pachi81/GlucoDataHandler/files/13875279/logcat2.txt)

pachi81, 2024-01-09:
Great! Now, I see the values of my app. Unfortunately only the last half an hour. 
Can you create it again, if the issue occurs again? 
And as you broadcast looks different to my one, which source do you use in xDrip+?

alexey6600, 2024-01-09:
Ok. When issue occurs I create logcat again.
In XDrip source is Sync Follower

alexey6600, 2024-01-09:
Happened again
[logcat3.txt](https://github.com/pachi81/GlucoDataHandler/files/13876670/logcat3.txt)

pachi81, 2024-01-09:
Great! Thank you!
I have received these data:
`01-09 18:46:35.785 20911 20911 I GDH.XDripBroadcastReceiver: Glucose: 0.0 - Time: 18:46:35 - Slope: 0.0 - SlopeName: null - Raw: 172.88371999999998 - Source: null (Follower)`
What has xDrip shown at 18:46:35? The raw value is set to 9,6 mmol/l.
Is that correct.

At the moment I'm not sure, what I will do, either ignore or using the raw value.

The raw value I do not really understand, for example I always get a raw value of 0.0.

I have to check other sources.

alexey6600, 2024-01-09:
[image]

pachi81, 2024-01-09:
Thank you very much!
I use the raw value, but I do not have trend information, so I will display a ?
You can download this version 0.9.8.5 here:
https://onedrive.live.com/?authkey=%21AG8eUfvdnCAQ1G4&id=F01898105C58ADFA%211299367&cid=F01898105C58ADFA

alexey6600, 2024-01-10:
Problem appeared again with new version
[image]

pachi81, 2024-01-10:
:-(
Can you provide logcat?

alexey6600, 2024-01-10:
[logcat4.txt](https://github.com/pachi81/GlucoDataHandler/files/13887198/logcat4.txt)

alexey6600, 2024-01-10:
I noticed that such problems arise with a sharp increase or decrease in sugar levels

pachi81, 2024-01-10:
Have you already created an issue for xDrip? If not I can also do it.
Here is the data, I have received:
```
{ 
  com.eveningoutpost.dexdrip.Extras.NoiseBlockLevel => 200
  com.eveningoutpost.dexdrip.Extras.VersionInfo => 2123092521 ab4c6032b-2023.09.25
  com.eveningoutpost.dexdrip.Extras.Raw => 175.88371999999998
  com.eveningoutpost.dexdrip.Extras.Time => 1704887351817
  com.eveningoutpost.dexdrip.Extras.CalibrationInfo => 1704748394164
  com.eveningoutpost.dexdrip.Extras.NsNoiseLevel => NULL
  com.eveningoutpost.dexdrip.Extras.Noise => 250.99240296515094
  com.eveningoutpost.dexdrip.Extras.SensorBattery => 0
  com.eveningoutpost.dexdrip.Extras.SourceDesc => Follower
}
```

There is missing some informations, like `com.eveningoutpost.dexdrip.Extras.BgSlope`, `com.eveningoutpost.dexdrip.Extras.BgSlopeName` and `com.eveningoutpost.dexdrip.Extras.BgEstimate`.

At the moment, I check for the presence of BgSlope. I can also remove this to show at least the raw-value without trend direction...

pachi81, 2024-01-10:
I have checked the source code of xDrip and it seems to be not a bug, it is a "feature".
The problem is the NoiseBlockLevel which is set to 200 on your device. The current noise level is around 250.
So xDrip blocks these levels and do not send any estimated values (even the trend).

Please go to settings - inter app settings and change the noise block level (3. entry) to send anything.

alexey6600, 2024-01-10:
Done.
Thanks! I'll test, and write about the test results!

pachi81, 2024-01-10:
Hopefully, you will not detect it, as all should work like expected, now.
I will create a hint for this problem in GDH, if it occurs.

Thank you so much for your help!!!

pachi81, 2024-01-12:
Hi!
Does it happened again?

alexey6600, 2024-01-12:
Hello.
No. Work well!
Thanks!

Ps. A small wish, if possible)
On the watch, make the option "glucose and trend" colored, and not big.
Now there is such an option, but it’s a big one.

pachi81, 2024-01-12:
Great to hear. 
Can you do me one more favor, can you install the last beta version 0.9.8.6-dev and change the xDrip Noise Block settings to lowest one to force this error happens again? Than you should receive a question mark instead of the trend arrow and there should be some hint on main screen of my app. 
After you have received a valid value, this hint should disappear... 

I tried to reproduce this issue, but for my source the highest noise was around 4. So I did not get into this problem.

> Ps. A small wish, if possible)
On the watch, make the option "glucose and trend" colored, and not big.
Now there is such an option, but it’s a big one.

Wow, I did not expect such a whish... The most people like to have it very big. The normal, small ones are no images and I can only add colors to images. But I can limit the font size while creating the images, so it would be smaller. 
What do you expect, only the glucose value to be smaller or both?

alexey6600, 2024-01-12:
With big ones I have this:
[image]

With normal more nice look:
[image]

alexey6600, 2024-01-12:
Yep. With latest dev there are question mark and hint. 
[image]

pachi81, 2024-01-12:
Yes the normal is text only with icon, where I can not change the color, but I can add an additional image with smaller size.

pachi81, 2024-01-12:
Great, that is what I expected!

pachi81, 2024-01-12:
Hi,
I created a new version with small colored complications. Can you check?
You can also join my beta playstore for easier installation on watch, but you know, how to use adb.

alexey6600, 2024-01-12:
Thanks!
I'm allready joined to beta on playstore. But I don't see a newer version yet

pachi81, 2024-01-12:
Don't know, how it works. At first step, I added a mail list. But there are 3 areas until release, maybe you will are on a higher layer...
But for the higher layers, I need google confirmation.

pachi81, 2024-01-12:
I moved it on a higher level. Now I have to wait for google...
If I have it, maybe you will receive it as beta tester...

alexey6600, 2024-01-12:
Ok. Thanks a lot!
I will wait.

alexey6600, 2024-01-14:
Just received update. 
Thanks! Nice look now on watch.

pachi81, 2024-01-14:
Can you make a screenshot? 
On my watch it looks different to the emulator.
So I'd like to see, how it looks on your watch.
Thank you!

alexey6600, 2024-01-14:
[image]

alexey6600, 2024-01-14:
Option "foreground" disappeared. 
This is normal?

pachi81, 2024-01-14:
It looks like on my watch....
In the emulator I have less space.

Yes, I removed the foreground option as some user deactivated it and complaint about problems...
Now it is mandatory for every one.

pachi81, 2024-01-18:
@alexey6600 I create a new beta and uploaded it. After google accepted it, you will get the new version.
I changed the small trend-glucose complication to have less space between the value and the trend arrow.
Please provide some feedback and screenshot, how do you like. 
Thank you!

alexey6600, 2024-01-18:
Thanks!
After I receive update - give you feedback!

alexey6600, 2024-01-18:
Is version 0.9.8.8 (on watch) new ?
Now is very small space between value and trend. 
[image]

pachi81, 2024-01-18:
Update is available, now...

alexey6600, 2024-01-18:
[image]

pachi81, 2024-01-18:
Is it ok? Or still to big?

alexey6600, 2024-01-18:
Still too big. But it can be like that)

pachi81, 2024-01-18:
Ok, because, the next size, which looks good, I think is too small.
So I would let it be like that...

alexey6600, 2024-01-18:
Thank you! 
Really good software?

pachi81, 2024-01-18:
Thank you!
On my watch the small is nearly the same like the normal one:
[image]

[1 more reply]

closed 2024-02-21 by pachi81