pachi81/GlucoDataHandler #134: question on permissions

IzzySoft, 2024-02-21:
My scanner got a few new checks in January, and with today's release reported:

```
! repo/de.michelinside.glucodatahandler_1022.apk declares flag(s): usesCleartextTraffic
! repo/de.michelinside.glucodatahandler_1022.apk declares sensitive permission(s): android.permission.SYSTEM_ALERT_WINDOW
! repo/de.michelinside.glucodatahandler_1022.apk contains signature block blobs: 0x504b4453 (DEPENDENCY_INFO_BLOCK; GOOGLE)
```

Could you please help me clarify those?

* `usesCleartextTraffic`: guess that's needed to connect to the gadgets as they don't have certificates and thus use plain http?
* `android.permission.SYSTEM_ALERT_WINDOW` is used for… what?
* `DEPENDENCY_INFO_BLOCK` is easy to get rid of:

```gradle
android {
    dependenciesInfo {
        // Disables dependency metadata when building APKs.
        includeInApk = false
        // Disables dependency metadata when building Android App Bundles.
        includeInBundle = false
    }
}
```

For some background: that BLOB is supposed to be just a binary representation of your app's dependency tree. But as it's encrypted with a public key belonging to Google, only Google can read it – and nobody else can even verify what it really contains.

pachi81, 2024-02-22:
* `usesCleartextTraffic`: Yes it is used to access local http webserver provided by other apps to receive data from
* `android.permission.SYSTEM_ALERT_WINDOW` is used for the floating widget
* `DEPENDENCY_INFO_BLOCK` I have no idea and will try your fix...

IzzySoft, 2024-02-22:
Thanks for the details, added:

[image]

> I have no idea

Android Studio (and also IntelliJ IDEA) add that by default as Google likes to have it. I've summed up some details on those checks on the info page in my repo, see [What about security?](https://apt.izzysoft.de/fdroid/index/info#security) There will hopefully be a full and detailed article on this available soon (I have the draft ready but it needs some "fine tuning" to not be too techy and "overwhelming" :wink:)

> will try your fix...

Thanks! Worked out on many projects already, I'm not aware of any where it didn't.

closed 2024-05-04 by pachi81