pachi81/GlucoDataHandler #136: Broadcast to WatchDrip+

infossk, 2024-02-22:
If possible, please create a broadcast to WatchDrip+ (like Juggluco does) so I don't have to use xDrip or Juggluco. Thx.

pachi81, 2024-02-22:
I already have checked the Juggluco sources and it seems not a big deal! 
I have installed version 0.2.6 of WatchDrip+ and it works with Juggluco without having a watch. So I can test my implementation :-)

pachi81, 2024-02-22:
The only problem is, I can not send a graph, yet.
So the graph on watch will not work!
@infossk  Is this ok for you?

infossk, 2024-02-22:
I will try how it will work without the chart.  As far as I know, the graph is drawn directly by the WatchDrip+ application.

pachi81, 2024-02-22:
I don't know, but in Juggluco, there is an extra function for adding the graph to the watchdrip broadcast.
But if I'm ready, you can try it...

pachi81, 2024-02-23:
I have created a first version. The values will be received in WatchDrip, but without watch, I cannot say, what is missing:
https://onedrive.live.com/?authkey=%21AG8eUfvdnCAQ1G4&id=F01898105C58ADFA%211299367&cid=F01898105C58ADFA
@infossk can you test it? Thank you!
You have to enable WatchDrip+ in setting in GlucoDataHandler and after that, you have to (re-)enable the service in WatchDrip+.

Regards, Michael

infossk, 2024-02-23:
Hello thank you.  It works without problems.  The difference is that the graph, last bolus and IOB are not really displayed.

Marian

[image]

[image]

pachi81, 2024-02-23:
Yes, as I already said, I do not have a graph for sending, yet. And the other parts I have to check.
But Juggluco also does not send the IOB and Bolus!?
If I find a way, I can include IoB, but I do not support bolus, yet.

infossk, 2024-02-23:
I understand.  Juggluco also does not send this data.  I can only send them from xDrip.  The graph is only a design problem.  🙂 IOB would be useful though.

infossk, 2024-02-23:
Just for your interest, I'm sending you a photo of my car radio.  🙂 You can see your floating widget below.  It would be great if I could lock his position.

[image]

pachi81, 2024-02-23:
I uploaded a new version with IoB. Not sure, if it is the correct setting. Can you try?

The floating widget position should be locked until you move it. Or what do you mean?

infossk, 2024-02-23:
The IOB is displayed.  👌

Regarding the position of the floating widget: When I get in the car and turn on the radio, the widget is always in the middle and I have to keep moving it.  I don't know if it is possible to save after moving and restore the position after activating (starting) GlocoDataHandler.  Maybe it's a problem with the launcher I'm using (FCC launcher).

pachi81, 2024-02-23:
Great! But in the moment, I do not support alarms. I will add, if I have some time.

Normally the position is stored...
You can try on phone. Maybe there is any problem on the radio. Which Android version is it?
Do you can connect the radio with adb for getting logs?

pachi81, 2024-02-23:
@lostboy86 Do you also have problems with the position of the floating widget on car radio?

lostboy86, 2024-02-23:
No problems with widget,
But i tested only few times.

Me and my medical dog , ar 18/02/2024, attacked from a free dog in the
night whithout leash, my dog arm was esploded, same my hand etc

infossk, 2024-02-23:
The version in the car is Android 10.
Sorry, but I won't be able to connect via adb today.
Thank you for your great work.

Regards, Marian.

pachi81, 2024-02-23:
> No problems with widget, But i tested only few times. Me and my medical dog , ar 18/02/2024, attacked from a free dog in the night whithout leash, my dog arm was esploded, same my hand etc

@lostboy86 WTF! I wish all the best! Get well soon!

infossk, 2024-02-23:
I tested it.  Alarms on watches also work.

pachi81, 2024-02-23:
Thank you for testing. 😀👍

lostboy86, 2024-02-24:
> > No problems with widget, But i tested only few times. Me and my medical dog , ar 18/02/2024, attacked from a free dog in the night whithout leash, my dog arm was esploded, same my hand etc
> 
> @lostboy86 WTF! I wish all the best! Get well soon!

I reply above for the widget

lostboy86, 2024-02-24:
> Just for your interest, I'm sending you a photo of my car radio. 🙂 You can see your floating widget below. It would be great if I could lock his position.
> 
> ![IMG_20240223_160328753.jpg](https://private-user-images.githubusercontent.com/8257881/307365299-ffe08df9-0796-4853-8ada-123fdf0f804e.jpg?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDg3NTc3NTQsIm5iZiI6MTcwODc1NzQ1NCwicGF0aCI6Ii84MjU3ODgxLzMwNzM2NTI5OS1mZmUwOGRmOS0wNzk2LTQ4NTMtOGFkYS0xMjNmZGYwZjgwNGUuanBnP1gtQW16LUFsZ29yaXRobT1BV1M0LUhNQUMtU0hBMjU2JlgtQW16LUNyZWRlbnRpYWw9QUtJQVZDT0RZTFNBNTNQUUs0WkElMkYyMDI0MDIyNCUyRnVzLWVhc3QtMSUyRnMzJTJGYXdzNF9yZXF1ZXN0JlgtQW16LURhdGU9MjAyNDAyMjRUMDY1MDU0WiZYLUFtei1FeHBpcmVzPTMwMCZYLUFtei1TaWduYXR1cmU9YjQxODAzNmRkMmNhMDI2NWFjZWU4OTVlODM2NGZkZGJmNWNjN2IyZWQ1NzljNzgyNmM0ZDUxYzhmNzg2MDMzNiZYLUFtei1TaWduZWRIZWFkZXJzPWhvc3QmYWN0b3JfaWQ9MCZrZXlfaWQ9MCZyZXBvX2lkPTAifQ.1t4rYUzp7x_om08dOW0zmpx-wVdPdC1on6Zj2wTQy_o)

How you make the aaps widget, floating?
I also have a similar A10 radio, atoto s8 gen2, with agama launcher.

Thanks for the help

lostboy86, 2024-02-24:
> The IOB is displayed. 👌
>
> Regarding the position of the floating widget: When I get in the car and turn on the radio, the widget is always in the middle and I have to keep moving it. I don't know if [...]

Same in my radio, with agama launcher or stock launcher

pachi81, 2024-02-24:
Ok, is it possible to attach adb to get the logs?

What happen, if you change the colors, will they stay the same after restart? 
Either it is a problem with saving the settings or a bug with the floating widget. Will check in emulator.

infossk, 2024-02-24:
I tried all available launchers (including Agama) and the only one that can work well with widgets for the FCC launcher. 

> How you make the aaps widget, floating?
> I also have a similar A10 radio, atoto s8 gen2, with agama launcher.
>
> Thanks for the help

infossk, 2024-02-24:
> Ok, is it possible to attach adb to get the logs?
> 
> What happen, if you change the colors, will they stay the same after restart? 
> Either it is a problem with saving the settings or a bug with the floating widget. Will check in emulator.

Today, when I got into my car, the floating widget was exactly where I left it yesterday.  It's probably fine.  Sorry, but I won't have much time to devote to this in the next few days.  I'm also going on a long trip and I don't want to run out of battery in the car.  🙂

infossk, 2024-02-24:
I found one more problem with WatchDrip.  No delta is displayed.  Zero is still displayed there.

pachi81, 2024-02-24:
> I found one more problem with WatchDrip. No delta is displayed. Zero is still displayed there.

I have added a new 0.9.9.4 version. Can you check for delta. Not sure, if I have the correct value, now... (no docu).

I also added a change to the floating widget. If you disable it, it also save the position. So can you try to enable it, move it to the position you want it and disable it. Then re-enable it and it should stay at this position.

infossk, 2024-02-24:
Unfortunately, the delta is not displayed even in version 0.9.9.4.  I will be able to test the widget in the car only tomorrow.

[image]

infossk, 2024-02-24:
This is what the output from xDrip looks like.

[image]

pachi81, 2024-02-24:
Mmmh, really strange! This was the only value I have found, to put the delta in. Have you tried it for longer time? 
I have to re-check the xdrip code, if there is anything else...

pachi81, 2024-02-24:
I have checked the values, my app is sending and there seems to be an error, related to delta. I have to check. But not today... 😉

infossk, 2024-02-25:
Yes, upload it, pls.

pachi81, 2024-02-25:
Is online! Haven't increased the version... 😉

infossk, 2024-02-25:
Did you upload it yesterday at 19:37?  Because that's what I see in the Onedrive shared folder.

[image]

pachi81, 2024-02-25:
No, today, an hour ago. Please reload the page.

pachi81, 2024-02-25:
Thanks, next time I will change the version ;-)

infossk, 2024-02-25:
I modified the watch face on my watch.  I removed the graph and centered the blood glucose data.  I am satisfied now. 🙂 

Thank you, Michael.

[image]

[image]

pachi81, 2024-02-25:
Looks good 👍
What kind of watch it is?
You have created your own watchface?

infossk, 2024-02-25:
Amazfit GTR2e.  I edited the config.json on the watch face created by Artem (author of WatchDrip+)

pachi81, 2024-02-25:
Thank you! Cool, good to know!

pachi81, 2024-03-12:
Available in new release 0.9.10

pachi81, 2024-05-15:
@infossk 
Hi, I´m working on a new GDH version including alarms. 
It is nearly finished, except the Watchdrip+ part. 
Could you test this part, if I provide you a test version?

Just one question, is it possible to dismiss or snooze the alarm on watch?
I have found something related to this in the xDrip code...

Thank you!

infossk, 2024-05-15:
Hi, of course I will test it.
The alarm can be turned off in xDrip using the watch. I think they use a phone call for that, but I'm not sure.

pachi81, 2024-05-15:
@infossk 
Perfect, thank you! As I´m ready, I will provide a test version.
I have checked the xDrip code, I found an Snooze Alert command, which is received from watchdrip. I will use this to stop the alert.

pachi81, 2024-05-15:
@infossk 
Can you try this version:
https://onedrive.live.com/?authkey=%21AG8eUfvdnCAQ1G4&id=F01898105C58ADFA%211299367&cid=F01898105C58ADFA

In the watchdrip app I do not see any alarm, so I can not test it. It should also solves your problem on Android Radio.
You can use the test alert in the alarm settings for the related alarm type.
Very low and very high are sent as alarms, the other ones as "other alarm".

Thank you!

infossk, 2024-05-15:
> @infossk Can you try this version: https://onedrive.live.com/?authkey=%21AG8eUfvdnCAQ1G4&id=F01898105C58ADFA%211299367&cid=F01898105C58ADFA
>
> In the watchdrip app I do not see any alarm, so I can not [...]

I tried it (alarms on the watch). I'll try it in the car later. Everything works as expected. Canceling the alarm on the watch also works. Perfect job!
 
Another useful functionality would be if an alarm could be set in the event of a sharp drop or rise in BG. Depending on what the delta is. But watch out for the minute reading. :)

pachi81, 2024-05-15:
@infossk Great! Thank you for testing. Can you please send me the logs from phone, so I can check the commands received from watchdrip.

> Another useful functionality would be if an alarm could be set in the event of a sharp drop or rise in BG. Depending on what the delta is. But watch out for the minute reading. :)

I already got a request for this. But step by step ;-)

infossk, 2024-05-15:
> @infossk Great! Thank you for testing. Can you please send me the logs from phone, so I can check the commands received from watchdrip.
>
>
> I already got a request for this. But step by step ;-)

https://gist.github.com/infossk/038dcd9cec2fb3878366fa98d8149d34

pachi81, 2024-05-15:
@infossk Thank you! From the logs, I only see, that you stopped the alarm from the notification!? 
Have you also tried to stop it from the watch?

infossk, 2024-05-15:
> @infossk Thank you! From the logs, I only see, that you stopped the alarm from the notification!? Have you also tried to stop it from the watch?

I stopped the alarms from the watch.

infossk, 2024-05-15:
> @infossk Thank you! From the logs, I only see, that you stopped the alarm from the notification!? 
> Have you also tried to stop it from the watch?

I am sending other log:

--------- beginning of system
05-15 14:41:42.937 26259 26259 D VRI[MainActivity]: update {(0,0)(fillxfill) sim={adjust=pan forwardNavigation} ty=BASE_APPLICATION wanim=0x10302fd
05-15 14:41:42.937 26259 26259 D VRI[MainActivity]:   fl=LAYOUT_IN_SCREEN LAYOUT_INSET_DECOR SPLIT_TOUCH HARDWARE_ACCELERATED DRAWS_SYSTEM_BAR_BACKGROUNDS
05-15 14:41:42.937 26259 26259 D VRI[MainActivity]:   pfl=NO_MOVE_ANIMATION FORCE_DRAW_STATUS_BAR_BACKGROUND USE_BLAST FIT_INSETS_CONTROLLED
05-15 14:41:42.937 26259 26259 D VRI[MainActivity]:   bhv=DEFAULT
05-15 14:41:42.937 26259 26259 D VRI[MainActivity]:   fitSides=} by setLayoutParams
05-15 14:41:45.016 26259 26259 D ViewRootImpl: update {(523,117)(546x0) gr=TOP START CENTER DISPLAY_CLIP_VERTICAL sim={state=unchanged} ty=APPLICATION_SUB_PANEL fmt=TRANSLUCENT surfaceInsets=Rect(45, 45 - 45, 45) (manual)
05-15 14:41:45.016 26259 26259 D ViewRootImpl:   fl=LAYOUT_NO_LIMITS ALT_FOCUSABLE_IM WATCH_OUTSIDE_TOUCH SPLIT_TOUCH HARDWARE_ACCELERATED FLAG_LAYOUT_ATTACHED_IN_DECOR
05-15 14:41:45.016 26259 26259 D ViewRootImpl:   pfl=WILL_NOT_REPLACE_ON_RELAUNCH LAYOUT_CHILD_WINDOW_IN_PARENT_FRAME
05-15 14:41:45.016 26259 26259 D ViewRootImpl:   bhv=DEFAULT
05-15 14:41:45.016 26259 26259 D ViewRootImpl:   fitTypes=STATUS_BARS NAVIGATION_BARS CAPTION_BAR} by setView
05-15 14:41:45.028 26259 26259 D VRI[PopupWindow:3032352]: update {(523,117)(546x1132) gr=TOP START CENTER DISPLAY_CLIP_VERTICAL sim={state=unchanged} ty=APPLICATION_SUB_PANEL fmt=TRANSLUCENT surfaceInsets=Rect(45, 45 - 45, 45) (manual)
05-15 14:41:45.028 26259 26259 D VRI[PopupWindow:3032352]:   fl=LAYOUT_NO_LIMITS ALT_FOCUSABLE_IM WATCH_OUTSIDE_TOUCH SPLIT_TOUCH HARDWARE_ACCELERATED FLAG_LAYOUT_ATTACHED_IN_DECOR
05-15 14:41:45.028 26259 26259 D VRI[PopupWindow:3032352]:   pfl=WILL_NOT_REPLACE_ON_RELAUNCH LAYOUT_CHILD_WINDOW_IN_PARENT_FRAME
05-15 14:41:45.028 26259 26259 D VRI[PopupWindow:3032352]:   bhv=DEFAULT
05-15 14:41:45.028 26259 26259 D VRI[PopupWindow:3032352]:   fitTypes=STATUS_BARS NAVIGATION_BARS CAPTION_BAR} by setLayoutParams
05-15 14:41:51.915 26259 26259 D VRI[PopupWindow:3032352]: update {(523,117)(546x1132) gr=TOP START CENTER DISPLAY_CLIP_VERTICAL sim={state=unchanged} ty=APPLICATION_SUB_PANEL fmt=TRANSLUCENT surfaceInsets=Rect(45, 45 - 45, 45) (manual)
05-15 14:41:51.915 26259 26259 D VRI[PopupWindow:3032352]:   fl=NOT_FOCUSABLE NOT_TOUCHABLE LAYOUT_NO_LIMITS WATCH_OUTSIDE_TOUCH SPLIT_TOUCH HARDWARE_ACCELERATED FLAG_LAYOUT_ATTACHED_IN_DECOR
05-15 14:41:51.915 26259 26259 D VRI[PopupWindow:3032352]:   pfl=WILL_NOT_REPLACE_ON_RELAUNCH LAYOUT_CHILD_WINDOW_IN_PARENT_FRAME
05-15 14:41:51.915 26259 26259 D VRI[PopupWindow:3032352]:   bhv=DEFAULT
05-15 14:41:51.915 26259 26259 D VRI[PopupWindow:3032352]:   fitTypes=STATUS_BARS NAVIGATION_BARS CAPTION_BAR} by setLayoutParams
05-15 14:41:58.099 26259 26259 D ViewRootImpl: update {(523,117)(546x0) gr=TOP START CENTER DISPLAY_CLIP_VERTICAL sim={state=unchanged} ty=APPLICATION_SUB_PANEL fmt=TRANSLUCENT surfaceInsets=Rect(45, 45 - 45, 45) (manual)
05-15 14:41:58.099 26259 26259 D ViewRootImpl:   fl=LAYOUT_NO_LIMITS ALT_FOCUSABLE_IM WATCH_OUTSIDE_TOUCH SPLIT_TOUCH HARDWARE_ACCELERATED FLAG_LAYOUT_ATTACHED_IN_DECOR
05-15 14:41:58.099 26259 26259 D ViewRootImpl:   pfl=WILL_NOT_REPLACE_ON_RELAUNCH LAYOUT_CHILD_WINDOW_IN_PARENT_FRAME
05-15 14:41:58.099 26259 26259 D ViewRootImpl:   bhv=DEFAULT
05-15 14:41:58.099 26259 26259 D ViewRootImpl:   fitTypes=STATUS_BARS NAVIGATION_BARS CAPTION_BAR} by setView
05-15 14:41:58.115 26259 26259 D VRI[PopupWindow:4cf3b5f]: update {(523,117)(546x1132) gr=TOP START CENTER DISPLAY_CLIP_VERTICAL sim={state=unchanged} ty=APPLICATION_SUB_PANEL fmt=TRANSLUCENT surfaceInsets=Rect(45, 45 - 45, 45) (manual)
05-15 14:41:58.115 26259 26259 D VRI[PopupWindow:4cf3b5f]:   fl=LAYOUT_NO_LIMITS ALT_FOCUSABLE_IM WATCH_OUTSIDE_TOUCH SPLIT_TOUCH HARDWARE_ACCELERATED FLAG_LAYOUT_ATTACHED_IN_DECOR
05-15 14:41:58.115 26259 26259 D VRI[PopupWindow:4cf3b5f]:   pfl=WILL_NOT_REPLACE_ON_RELAUNCH LAYOUT_CHILD_WINDOW_IN_PARENT_FRAME
05-15 14:41:58.115 26259 26259 D VRI[PopupWindow:4cf3b5f]:   bhv=DEFAULT
05-15 14:41:58.115 26259 26259 D VRI[PopupWindow:4cf3b5f]:   fitTypes=STATUS_BARS NAVIGATION_BARS CAPTION_BAR} by setLayoutParams
05-15 14:42:00.842 26259 26259 D VRI[PopupWindow:4cf3b5f]: update {(523,117)(546x1132) gr=TOP START CENTER DISPLAY_CLIP_VERTICAL sim={state=unchanged} ty=APPLICATION_SUB_PANEL fmt=TRANSLUCENT surfaceInsets=Rect(45, 45 - 45, 45) (manual)
05-15 14:42:00.842 26259 26259 D VRI[PopupWindow:4cf3b5f]:   fl=NOT_FOCUSABLE NOT_TOUCHABLE LAYOUT_NO_LIMITS WATCH_OUTSIDE_TOUCH SPLIT_TOUCH HARDWARE_ACCELERATED FLAG_LAYOUT_ATTACHED_IN_DECOR
05-15 14:42:00.842 26259 26259 D VRI[PopupWindow:4cf3b5f]:   pfl=WILL_NOT_REPLACE_ON_RELAUNCH LAYOUT_CHILD_WINDOW_IN_PARENT_FRAME
05-15 14:42:00.842 26259 26259 D VRI[PopupWindow:4cf3b5f]:   bhv=DEFAULT
05-15 14:42:00.842 26259 26259 D VRI[PopupWindow:4cf3b5f]:   fitTypes=STATUS_BARS NAVIGATION_BARS CAPTION_BAR} by setLayoutParams
05-15 14:42:01.913 26259 26259 D ViewRootImpl: update {(523,117)(546x0) gr=TOP START CENTER DISPLAY_CLIP_VERTICAL sim={state=unchanged} ty=APPLICATION_SUB_PANEL fmt=TRANSLUCENT surfaceInsets=Rect(45, 45 - 45, 45) (manual)
[... cut ...]
05-15 14:43:31.011 15382 15382 D GDH.WearPhoneConnection: check for connected nodes without data
05-15 14:43:31.027 15382 15382 D BufferQueueConsumer: [](id:3c1600000006,api:0,p:-1,c:15382) connect: controlledByApp=false
05-15 14:43:31.029 15382 15402 E OpenGLRenderer: Unable to match the desired swap behavior.
05-15 14:43:31.038 15382 15402 D BLASTBufferQueue: [VRI[MainActivity]#6](f:0,a:1) acquireNextBufferLocked size=1080x2400 mFrameNumber=1 applyTransaction=true mTimestamp=580621805509992(auto) mPendingTransactions.size=0 graphicBufferId=66065186947102 transform=0

pachi81, 2024-05-15:
Perfect! In these logs I have seen everything I need! Great! Thank you! So it seems to work. :-)

pachi81, 2024-05-15:
@infossk Just one more last thing ;-)
Can you make a picture, how the alarms looks like on the watch? I´m thinking about also add the trend arrow or delta to alarm message, but not sure, how much space there is...
What do you think? Do you want to have some more information?

infossk, 2024-05-15:
> @infossk Just one more last thing ;-)
> Can you make a picture, how the alarms looks like on the watch? I´m thinking about also add the trend arrow or delta to alarm message, but not sure, how much space there is...
> What do you think? Do you want to have some more information?

[image]

closed 2024-06-05 by pachi81