pachi81/GlucoDataHandler #142: Size of text on lock screen

sgmoore, 2024-03-18:
GlucoDataHandler has a very nice feature where my current blood glucose level is shown on my lock screen.

However the size of the text is extremely small, which means that it is hard to read without my glasses and especially in the middle of the night when I am only half-awake. 

It would be great if the size of this text could be increased. Is that possible?

I did pull the source code, but didn't see anything obvious about how or where this is implemented and don't know enough about android programming to follow your logic.

pachi81, 2024-03-18:
Hi,
which part do you exactly mean? The notification text is quite big, I think. And the icons are limited in their size an I can not increase them. What kind of smartphone are you using? For Samsung for example, there is another solution to bring the Widget on Lockscreen and AOD.
Regards, Michael

sgmoore, 2024-03-18:
I'm using a Moto G13 running Android 13.

The following is a screenshot of my lock screen showing my glucose levels at 10.8.

[image]

The size of the text looks to be the same as the size of the text on the notification/status bar.

pachi81, 2024-03-18:
Yes, this is the icon of the notification. 
I can not increase it.
Can you check, if it is possible, to show the whole notification on lockscreen? Maybe it is a setting anywhere.
For example my lockscreen looks like this:
[image]

sgmoore, 2024-03-18:
If I understand you correctly you are saying that your app provides the notification and it is up the phone whether or not the notification appears on the lock screen.
If so, it looks like I am out of luck. I can find an option to hide notifications from the lock screen, but no option to display the full notification as your phone is doing.

pachi81, 2024-03-18:
Yes, it is related to Android settings and not to my app. What happens, if you touch on the lockscreen or the icon of my app? Maybe this will show the full message...
Have you already checked these settings:
https://en-us.support.motorola.com/app/answers/detail/a_id/172907/~/lockscreen-security---moto-g13#:~:text=Go%20to%20Settings%20%3E%20Display%20%3E%20Advanced,or%20not%20to%20show%20notifications.

It also seems, that the moto g13 has an AOD screen. Maybe this will help you: https://www.youtube.com/watch?v=3Ov2SliJ07A

sgmoore, 2024-03-18:
No joy.
Nothing happens when I touch either the lock screen or the icon.
I have checked the moto settings.
I tried the app in the youtube link and another app that is supposed to do something similar but neither or them work.

pachi81, 2024-03-18:
Oh, thats bad... But I have no other idea, yet. Maybe you can ask in any Moto G13 forum as it is a commom problem with notifications. But I can not help you...
I will also close this topic as it is not an issue of GDH.
Good luck!

pachi81, 2024-03-18:
Hi,

maybe, there is a solution. Can you try this app:
https://play.google.com/store/apps/details?id=com.jee.timer

Start a timer, maybe for 10 seconds and lock the phone.
On my phone I get a fullscreen notification on lockscreen.
Maybe this can also be done by my app...

sgmoore, 2024-03-18:
Sorry to have caused you such trouble.

With that app, during the 10 seconds countdown, there is a normal notification message but this does not appear on the lock screen. 
After the 10 seconds countdown are up, then the phone wakes up with a fullscreen display until I stop the alarm.

pachi81, 2024-03-19:
> After the 10 seconds countdown are up, then the phone wakes up with a fullscreen display until I stop the alarm.

Great, this is what I wanted to know. Maybe it is possible to create such a fullscreen for lockscreen for the whole time, so I can put a bigger value on. But then you only have this screen without other notifications. What do you think about?
But I have no idea, if this will drain the battery... (it is just an idea)

> Sorry to have caused you such trouble
No problem, it is no trouble as I also want to add alarms to my add and therefor I want to use such an full lockscreen display to snooze alarms, like in the MultiTimer App...

sgmoore, 2024-03-19:
I'm not sure I like the idea of taking over the full lock screen. It would be fine for something like an alarm as it only takes over the screen when it is needed, but for GlucoDataHandler to do so would mean it would be taking over it over permanently and could interfere with other notifications and apps etc.

I notice there is an option in GlucoDataHandler to forward data to xDrip so I am going to try that as xdrip has an option to draw the glucose value on the lockscreen and it allows you to choose the size and position of the text.

pachi81, 2024-03-19:
Ok, so I will only add it for alarm notification. 

Good luck with xDrip. I know there is an option for AOD in xDrip. Do you mean this one?

sgmoore, 2024-03-19:
Xdrip seems to work well.
This is the feature I mean - https://xdrip.readthedocs.io/en/latest/use/display/#numbered-wall-on-locked-screen.
There is another section for Always on Display settings, but I haven't figured out what it does yet.

This is my current lockscreen, which is big enough for me to read even when half-asleep.

[image]

If you are wondering about the 4.8 vs 5.0, I presume Xdrip is doing some smoothing which is why it displays a slightly different figure from GlucoDataHandler. Haven't quite figured out if that is a good or bad thing.

Thank you again for your help.

pachi81, 2024-03-19:
Great, good to know! I will check the xDrip code. Maybe I can offer a similar thing. 

xDrip has a smoothing algorithm and supports only 5 minute interval as default. If you receives new values every minute, you have to activate it in xDrip:
https://github.com/NightscoutFoundation/xDrip/issues/2680

pachi81, 2024-03-19:
Just a remark for myself:
Link to the code of xDrip: https://github.com/NightscoutFoundation/xDrip/blob/e5113d973dee89ee2af78601cea0b6621fe3b0fc/app/src/main/java/com/eveningoutpost/dexdrip/ui/LockScreenWallPaper.java#L95

pachi81, 2024-03-20:
@sgmoore Hi, I have added a similar wallpaper option to GDH. Can you test it?
Please use version 0.9.11.4 from here:
https://onedrive.live.com/?authkey=%21AG8eUfvdnCAQ1G4&id=F01898105C58ADFA%211299367&cid=F01898105C58ADFA

Thank you!

sgmoore, 2024-03-20:
Brilliant.

This is working well (once I disabled xDrip's version)

Thank you very much.

pachi81, 2024-03-20:
Great to hear. So please have a look, if there is any issue or it drains the battery or something like that.

sgmoore, 2024-03-21:
24 hours later, no problems and I still have over 50% battery left, so I don't think it is going to cause any problems.

pachi81, 2024-03-21:
Perfect, thank you! I also have not recognized any problems, yet.
Maybe I will find a solution to add delta and time, like in the widgets, but everything I tried does not work, yet.

closed 2024-04-23 by pachi81