pachi81/GlucoDataHandler #145: Translations

dinizmauricio, 2024-03-24:
Hello!
I am in awe of your app!
I would like to contribute by inserting the translation of the strings into Brazilian Portuguese. If you agree, can you tell me would you prefer me to send the files?

pachi81, 2024-03-25:
Hi,
great! Thank you for your translation. 
Just one remark, this is an ongoing process, so you should translate new changes before release. I do not send every translation, even if I´m ready with my changes, then I will contact you and it would be great if you can change them direct in the pull request.
I add all new stuff at the end in English.
You do not have to do it immediately, because I normally test my app over a week before release.  If you like, I can add you to the list of internal testers, so you can verify the translation before the beta tester will see them.

And just one more question, is there a big difference between Brazilian Portuguese and normal Portuguese? 
If not, I would offer it as pt only for all Portuguese users!?

Thank you!

Regards, Michael

dinizmauricio, 2024-03-25:
Sure, I'll be happy to collaborate on new versions. Could you add me to the list of people testing the app?

Regarding the difference in translation, there are some words in Brazil that are different in Portugal and can cause wrong interpretations. From what little I know, applications tend to make different translations: pt and pt_br.

I will send the updated files using the pull request, as you suggested. I'll send you the other missing files soon :)

pachi81, 2024-03-25:
With the list of testers, I mean the playstore internal testers. So I need your gmail address, which you are using for playstore. Please send to my mail: glucodatahandler@michel-inside.de

If the difference is not so big, I can also use your string as pt, so it would be used for all pt variants. I think this is even better for Portuguese people then using English...

dinizmauricio, 2024-04-12:
Hello!
Sorry not answering. I'm having a busy week here. Can I send you the translation by Sunday?

pachi81, 2024-04-12:
Yes, of course! I only wanted to be sure,  that you received a notification.

dinizmauricio, 2024-04-14:
I just updated the file, could you check please?

dinizmauricio, 2024-04-19:
Hello Michael!

After a few days, without making any changes, the watch started showing the
results. Now, I'm trying to get the Android Auto running (see video
https://youtube.com/shorts/Z83Ob0wi5_w?si=c2uPU3OjuCmRQjl4). Do you know
why the values don't appear on the screen? I'm forgetting something?

Regarding the new alarm features, I will send you the translation by
tomorrow. If you want, I can translate from the car too ;)

Regards,

Maurício
[image: print2.jpeg]

pachi81, 2024-04-19:
Have you checked all these steps:
https://github.com/pachi81/GlucoDataAuto?tab=readme-ov-file#setup-android-auto

The strings are also for GDA, the next version will contain it

closed 2024-04-23 by pachi81