pachi81/GlucoDataHandler #147: Data colour in widget if LibreLinkup does not supply data

Peter-Jacob, 2024-03-30:
First of all, I would like to say how fantastic I think your app is! I use Libre 3 and a Samsung A34 as my reader, and I can now connect it to my Galaxy Watch 6 using your Gluco Data Handler. 
The Libre Linkup app you use to retrieve the data is slow as hell. If it doesn't provide data, you turn the colour grey and cross out the value. 
1/ Is it possible to keep the normal colour as if it were connected and just cross it out? My watch face background is black, gray is hard to read. 
2/ After a while, the value is dropped completely, and it re-appears after Linkup sends new data. Can you just display the old value (crossed out)?

I am unfamiliar with how Linkup operates, but it seems to transmit data every five minutes. If the value remains unchanged, it does not transmit data at all (just a feeling).

pachi81, 2024-03-30:
Hi,
Libre 3 should offer a new value every minute, also over LibreLink. Sometimes the server has problems, but normally there should be a new value every minute. 
Please change the interval to one minute and disable battery optimizations for my app.
If it does not work, please check the LibreLinkUp App, if there is a value every minute. If yes, it is a problem with GlucoDataHandler 😬
If not, it is a problem with the libre app. Then you should reinstall the libre app.

I just recommend to use Juggluco instead of the libre app, so you get the data direct from the app.

Regarding the color of an obsolete value. Good point, I could offer a setting to change it.

Regards, Michael

pachi81, 2024-03-30:
Just one more remark, there was another one using Galaxy A52s with the same issue. After re-installing the Libre 3 app, he received the value every minute.

Peter-Jacob, 2024-03-31:
I have tried your suggestions, reinstalling the libre app and disabling battery optimisation. There was no change in behaviour. I don't think it's a problem with your app, it's Librelinkup that's behaving strangely. When I open it I see that it doesn't receive any data for several minutes, my best guess is 5 minutes. You're not getting any data during that time either! I guess there is nothing you can do. 
But my initial request was to change the colour in this situation from grey to something else. It would be easier to read on my watch. It's just a thought, no urgency.

Regards

Peter (Munich)

pachi81, 2024-03-31:
Hi Peter,

I will provide a setting for changing the color.
I also thought about a setting to increase the time for an obsolete value.
At the moment I have a lot to do with implement alarms. So it will take some time...

Michael

Peter-Jacob, 2024-03-31:
Hi Michael, as I said, no urgency, I am completely happy to have something that is working. If I can be of any help to you, just let me know. These are my usual activities: [https://github.com/mvslovers/brexx370](url).  But I have no idea about app development. 

Peter

ojalaj, 2024-03-31:
@Peter-Jacob are you using LibreLink credentials in GlucoDataHandler or have you created a LibreLinkUp (follower) account and use those credentials with GlucoDataHandler? To my experience, these issues where some 3rd party app is not receiving Libre readings (slowly or not at all) are usually solved by either re-logging in to LibreLinkUp account (sometimes there are updated user rights agreements that need to be approved etc.) or then Libre3 app needs to be re-installed, as suggested by @pachi81, since there are some issues in sending Libre3 readings to Libre cloud, where the 3rd party app is pulling/polling them. I had similar issues than @Peter-Jacob and re-installing Libre3 helped.

pachi81, 2024-03-31:
Just use Juggluco instead of the nightmare of the libre app 😉🙃
No problems, you can connect the sensor direct to the Galaxy Watch, if you like, ....

Mongoose69, 2024-05-28:
> Just use Juggluco instead of the nightmare of the libre app 😉🙃
> No problems, you can connect the sensor direct to the Galaxy Watch, if you like, ....

I tried to set up Juggluco a while ago but it could never talk to my Libre 2 sensor IIRC

pachi81, 2024-05-28:
> > Just use Juggluco instead of the nightmare of the libre app 😉🙃
> > No problems, you can connect the sensor direct to the Galaxy Watch, if you like, ....
> 
> I tried to set up Juggluco a while ago but it could never talk to my Libre 2 sensor IIRC

Maybe you should try again or try using xDrip.
But in the next version I have fixed another bug with LibreLinkUp and also added a setting to change the color of obsolete values.
It is already available as beta and will released next week.

closed 2024-06-10 by pachi81

Mongoose69, 2024-06-14:
> Maybe you should try again or try using xDrip.
> But in the next version I have fixed another bug with LibreLinkUp and also added a setting to change the color of obsolete values.
> It is already [...]

Lol. I've tried xdrip, blose, diabox, everything I could. For all it's faults, at least Libre Link was simple to get started on, it just doesn't stay like that. Eg, It took 5 months before I managed to get the sensor to last the whole 14 days 🤷 and I've managed that feat only once