pachi81/GlucoDataHandler #152: Reconnect to NS source

infossk, 2024-05-15:
Hi, I would like to ask you to solve a problem.
I use GDH in the car in Android radio. After starting the car, the radio connects to the 4G network using the built-in SIM, but it always takes a while. However, it often happens that the glucose level is not displayed on the widget. Only turning the data source off and on again will help. But it is inconvenient while driving, as it is on the menu. :) A button on the main screen would help me to reconnect to the selected data source. 

Thank you.

pachi81, 2024-05-15:
Hi, that sounds strange! Can you send me the logs from the car? Is this possible to get access for.
Have you checked all battery optimization to be off?

pachi81, 2024-05-15:
Maybe it is related to the missing timeout handling, which will be fixed in the next version. You can already use the beta version of the playstore or wait for the test version I will provide you for watchdrip.

pachi81, 2024-05-15:
@infossk I have created a beta4 version, where I force update in the interval, even if the previous connection is still active. So you do not have to reconnect ns anymore.

closed 2024-06-10 by pachi81