pachi81/GlucoDataHandler #154: Graphs in app

pubmania, 2024-06-09:
Thanks for the app. It really os very nicely done. I was wondering if you have any plans to include graphs and logs for the data ot recieves on the phone app? It really will be very helpful especially because the visibility of data on librelink app is pretty basic and does not allow view of reading at the time, just graph and logs are only shown for manual scans. 

I think even if the value received is stored in a simple dqlite database that will be sufficient really.

Also is there a possibility to include a widget that shows values in both units not just mmol/l or mg/dl... thats helpful when sharing info with fellow diabetics from across the pond. :)

Once again, Many thanks for the lovely app.

pachi81, 2024-06-09:
Yes, I will add a graph. I already checked some libraries for a good one.
In the next version, I will add Dexcom Share.
Afterwards doing some optimization and then check adding the graph...

Do you really need a widget for the other unit? 
At the moment I already show the raw value in mg/dl, if you are using mmol/l in the details of the app. I can also add showing mmol/l, if you are using mg/dl, there. Will it be enough? Because normally you do not need the other unit for yourself all the day!?

pubmania, 2024-06-10:
>Do you really need a widget for the other unit?

No not at all. As you said its already shown in raw data. It is strongly in the category of "Nice to have" and will be unique to this app of added. No other app, that I know of, does it.

pubmania, 2024-06-10:
> As a widget is not a big deal, I have thought about adding a value only widget for the other unit. Do you have an idea, how this can be named? "other unit" or "alternate unit" or "second unit" or "complementary unit"?

Hmm as I was just hoping it will be displayed in main widget I did not give the naming much thought but if it will be a separate widget, you are right it will need a name. I think "Other Unit" might be least ambiguous.

pachi81, 2024-06-10:
> Hmm as I was just hoping it will be displayed in main widget I did not give the naming much thought but if it will be a separate widget, you are right it will need a name. I think "Other Unit" might [...]

Adding it to the main widget, will be more difficult as I have to take care for the layout with and without the other unit. 
But the big advantage for widgets is, that you can use multiple widgets and it is much easier for me to use just the simple value widget to show the other unit. That will be a very quick change, because I already have the related layout...

dinizmauricio, 2024-06-10:
I also think the work developed by @pachi81  is admirable!
I agree that the graph is still missing. For families using GlucoDataHander as a follower, the graph makes a significant difference.

pachi81, 2024-06-10:
> I also think the work developed by @pachi81 is admirable! I agree that the graph is still missing. For families using GlucoDataHander as a follower, the graph makes a significant difference.

Yes, the problem is, that I have developed the app mainly for myself :-) And I do not really need the graph, because it is the past and not really relevant for the future. Only if you want to analyze your insulin affect for meals and so on, but then you need a better tool, where you can see all events.... 
I only need the value, delta and trend-arrow of the present value to make my decisions... But that is my personal opinion.
But you are right, especially for followers, it makes really sense and also for other people, who are using the graph for their decisions.

BrandonGoodman, 2024-07-21:
Another vote for the graph.  
I used the [g-watch watchface](https://play.google.com/store/apps/details?id=sk.trupici.gwatch.wear&hl=en_US) and widget for my own daily monitoring. The mini graph on **_both_** the widget and the watch are great for at-a-glance review of the recent past readings.  I love that your app allows us to to choose the shortcut for the widget too.  The graph is the last think to make it a superior replacement for me.

Thank you, @pachi81 for the great work in any case.  I know its difficult to find motivation to implement features that you don't intend to use yourself ;)

pachi81, 2024-07-21:
Thank you for your feedback!
I already checked some libraries for graph implementation. 
For the next release (1.2) I already planed to add some smaller wishes for alarms and so on. After I have finished this and nothing important happens, I will start the work for the graph. 
Yes, I do not really need it for my self, but I know, a lot of people are using it and I'm interested in the development part for creating it 😉

Arakon, 2024-12-20:
I'll also add my +1 to the hope of a graph display, now that g-watch wear no longer works on newer wearOS versions. My BG can change rapidly and it's often critical for me to know the previous ups and downs to see if I need to react or if it's a misread/compression issue.

pachi81, 2024-12-20:
I already start playing with the graph libraries, but at moment it a little try and much fail 🤣

If someone has experience with graph library, please contact me. At the moment I'm playing with MPAndroidChart...

SpeedoPasanen, 2024-12-24:
> I already start playing with the graph libraries, but at moment it a little try and much fail 🤣
> 
> If someone has experience with graph library, please contact me. At the moment I'm playing with MPAndroidChart...

A charting lib might be overkill? Should be simple to draw a few dots and lines. But I don't know anything about Android dev, bring an old ass full stack Typescript dev. Wish I could help. 

I'd love a small circle complication with 12 dots showing the last hour, if that's technically possible...

Arakon, 2025-03-08:
Not sure if this is a quirk of Watchface Studio or WearOS, but I compiled the current version of the graph branch and the graph appears very small and off-center in a "Largebox" complication slot. In order to have it centered and actually filling up the Largebox complication, I had to severely oversize the Largebox. When sizing it to the frame in the center of the watchface, it would instead be squished vertically and very small (maybe filling 1/4-1/3 of the frame).

[image]

[image]

[image]

I'm running a Galaxy Watch 5 (on WearOS 4 still).

pachi81, 2025-03-08:
Complications are always in square format, so yes, you have to oversize them, because the graph is not in square format.

Arakon, 2025-03-08:
Thanks for the quick answer!
I was just messing around and found that using "SmallBox" at least requires less of an oversizing and may be preferable for my purpose.
Can you tell me at which actual pixel measurements the graph is generated?

pachi81, 2025-03-08:
It is not final, yet. I will also try again with rectangle. I used the square also to be compatible to watchfaces for Blose.
I will try to change it and see what happens.

pachi81, 2025-03-13:
@Arakon I changed the complication and added a new one:
1. 3x3 Graph complication, where the graph is centered vertically with fixed height of 1 -> to be compatible to existing watchfaces
2. 3x1 Graph complication, which can be used for new watchfaces like your one.

Will you publish your watchface in Playstore? If yes I can link it inside the app.

Arakon, 2025-03-13:
I likely won't publish it on playstore, but I suppose I could add it on github. 

It seems to be slightly stretched in my current setup:

[image]

[image]

With 3x1 you mean i.e. 450x150 would result in a properly scaled image, for example?

pachi81, 2025-03-14:
Hi,
I´m not a watchface expert, but in Pujie it looks as expected. 
Even in Samsung Watchface Studio the graph looks as expected for DMM watch faces:

[image]

The creator is using 400x133 for its graph complication...

So maybe you have to play around with the settings, but I can not help you, as I have no experience with Samsung Watchface Studio...

Arakon, 2025-03-14:
I think I got it down to a fairly decent look now.

[image]

Arakon, 2025-03-16:
I just uploaded the [watchface](https://github.com/Arakon/Hex-GlucoWatchFace) to github, if you want to link to it. Now comes with 3 color themes and a slightly updated background to allow for the color variations.

[image]

sgmoore, 2025-04-03:
Would it be possible to add the axes to the graph on the widget?

Particularly the x-axis as differs from the graph in the app.

pachi81, 2025-04-03:
@sgmoore which widget you mean? It is already possible to add them to the phone widgets.

pachi81, 2025-04-03:
There is also the floating widget and the "widget" for AOD and complication and so on. That is the reason, why I´m asking.
And you can enable the axes for this widget. So what is the problem?

pachi81, 2025-04-03:
Perfect, I was afraid, that there is s.th. wrong with axes... 🫣

RoadyFPV, 2025-05-12:
Hi, great work guys. 
I am looking for something exactly like this.
Is it possible to try out the new graph somehow?

pachi81, 2025-05-12:
Just wait for the release.
Hopefully this week...

RoadyFPV, 2025-05-12:
Sounds great.
Thanks for your efforts