pachi81/GlucoDataHandler #158: Broken connection to LibreLinkUp (2/3 times a day)

mak2710, 2024-07-10:
Hi, 
I've installed the GlucoDataHandler and GlucoDataAuto on my Asus Zenfone 8 (Android 13) and sometimes (2/3 times a day) the connection with my LibreLinkUp account stops working.
When I open the application the status of the connection of LibreLinkUp is NotConnected.
I've to stop the application and restart it.
I don't known if the issue is related to some of my phone settings.
I faced the issue when I installed the 1.0.0 version (and it has continued with 1.0.1).
I don't rembere this kind of issue with the 0.9.x versions.
Has anyone else reported a similar issue?
Thanks, 
Bye 
Marco

pachi81, 2024-07-10:
Hi Marco,
with version 1.0 I have added some improvements/fixes to the LibreLinkUp connection, maybe it causes another issue.
Have you disabled battery optimization and so on?
Can you send me the logs, if it happens again. 
Have you tried the "reconnect" switch, if it has happened? 
What is the correct status? "No internet connection"? 
If yes, this is related to the internet connection of you phone and is checked each time, before a request is made. 
Is there any battery saving mode, which disables the internet connection, if the phone goes to sleep?

Regards, Michael

mak2710, 2024-07-11:
Hi Michael,
> Have you disabled battery optimization and so on?
Yes, battery optimization is disabled
> What is the correct status? "No internet connection"?
Sorry, the correct status is "2:notAuthenticated"
> Is there any battery saving mode, which disables the internet connection, if the phone goes to sleep?
No
> Have you tried the "reconnect" switch, if it has happened?
Where is it?

Attached you can find the log file (downloaded when the error appears) and the screenshot of the error.

[GDH_PHONE_APP_20240710_204608.txt](https://github.com/user-attachments/files/16175547/GDH_PHONE_APP_20240710_204608.txt)
[image]

pachi81, 2024-07-11:
This error means, that you user and/or password is wrong. Can you re-check it. For user, you have to take the e-mail address for the follower account. 
Does it is working in the LibreLinkUp app?

mak2710, 2024-07-11:
> This error means, that you user and/or password is wrong. Can you re-check it. For user, you have to take the e-mail address for the follower account.

Ok but the problem is not the authentication because to get the connection working again I simple stop the GlucoDataHanlder application and restart it without change anything in the LibreLinkUp login settings.

> Does it is working in the LibreLinkUp app?

Yes

pachi81, 2024-07-11:
That is really strange as the error is received by LibreLink... 
Can you send me the logs after restarting GlucoDataHandler.

mak2710, 2024-07-11:
Hi!
Here attached 2 log files.
The first one 182441 was taken when the application was in error with state
notAuthenticated.
After I've stopped the application, restarted it and I've downloaded the
second log file 182545.

Thank you

--------- beginning of system
07-11 17:50:13.209 29786 29786 I ActivityThread: Performing receive of Intent { act=de.michelinside.glucodatahandler.GLUCODATA flg=0x30 pkg=de.michelinside.glucodatahandler cmp=de.michelinside.glucodatahandler/.common.receiver.InternalActionReceiver (has extras) }: ***@***.***, appName=de.michelinside.glucodatahandler, pkg=de.michelinside.glucodatahandler, comp={de.michelinside.glucodatahandler/de.michelinside.glucodatahandler.common.receiver.InternalActionReceiver}, dir=/data/app/~~6e6VNj5u89F4s4TEqRNAeg==/de.michelinside.glucodatahandler-YoTbiYzHzGR6Cwdv9tA7TA==/base.apk, duration=48
07-11 17:51:13.190 29786 29786 I ActivityThread: Performing receive of Intent { act=de.michelinside.glucodatahandler.GLUCODATA flg=0x30 pkg=de.michelinside.glucodatahandler cmp=de.michelinside.glucodatahandler/.common.receiver.InternalActionReceiver (has extras) }: ***@***.***, appName=de.michelinside.glucodatahandler, pkg=de.michelinside.glucodatahandler, comp={de.michelinside.glucodatahandler/de.michelinside.glucodatahandler.common.receiver.InternalActionReceiver}, dir=/data/app/~~6e6VNj5u89F4s4TEqRNAeg==/de.michelinside.glucodatahandler-YoTbiYzHzGR6Cwdv9tA7TA==/base.apk, duration=37
07-11 17:52:13.259 29786 29786 I ActivityThread: Performing receive of Intent { act=de.michelinside.glucodatahandler.GLUCODATA flg=0x30 pkg=de.michelinside.glucodatahandler cmp=de.michelinside.glucodatahandler/.common.receiver.InternalActionReceiver (has extras) }: ***@***.***, appName=de.michelinside.glucodatahandler, pkg=de.michelinside.glucodatahandler, comp={de.michelinside.glucodatahandler/de.michelinside.glucodatahandler.common.receiver.InternalActionReceiver}, dir=/data/app/~~6e6VNj5u89F4s4TEqRNAeg==/de.michelinside.glucodatahandler-YoTbiYzHzGR6Cwdv9tA7TA==/base.apk, duration=33
07-11 17:53:14.171 29786 29786 I ActivityThread: Performing receive of Intent { act=de.michelinside.glucodatahandler.GLUCODATA flg=0x30 pkg=de.michelinside.glucodatahandler cmp=de.michelinside.glucodatahandler/.common.receiver.InternalActionReceiver (has extras) }: ***@***.***, appName=de.michelinside.glucodatahandler, pkg=de.michelinside.glucodatahandler, comp={de.michelinside.glucodatahandler/de.michelinside.glucodatahandler.common.receiver.InternalActionReceiver}, dir=/data/app/~~6e6VNj5u89F4s4TEqRNAeg==/de.michelinside.glucodatahandler-YoTbiYzHzGR6Cwdv9tA7TA==/base.apk, duration=36
07-11 17:54:13.171 29786 29786 I ActivityThread: Performing receive of Intent { act=de.michelinside.glucodatahandler.GLUCODATA flg=0x30 pkg=de.michelinside.glucodatahandler cmp=de.michelinside.glucodatahandler/.common.receiver.InternalActionReceiver (has extras) }: ***@***.***, appName=de.michelinside.glucodatahandler, pkg=de.michelinside.glucodatahandler, comp={de.michelinside.glucodatahandler/de.michelinside.glucodatahandler.common.receiver.InternalActionReceiver}, dir=/data/app/~~6e6VNj5u89F4s4TEqRNAeg==/de.michelinside.glucodatahandler-YoTbiYzHzGR6Cwdv9tA7TA==/base.apk, duration=32
07-11 17:55:14.186 29786 29786 I ActivityThread: Performing receive of Intent { act=de.michelinside.glucodatahandler.GLUCODATA flg=0x30 pkg=de.michelinside.glucodatahandler cmp=de.michelinside.glucodatahandler/.common.receiver.InternalActionReceiver (has extras) }: ***@***.***, appName=de.michelinside.glucodatahandler, pkg=de.michelinside.glucodatahandler, comp={de.michelinside.glucodatahandler/de.michelinside.glucodatahandler.common.receiver.InternalActionReceiver}, dir=/data/app/~~6e6VNj5u89F4s4TEqRNAeg==/de.michelinside.glucodatahandler-YoTbiYzHzGR6Cwdv9tA7TA==/base.apk, duration=40
07-11 17:56:13.184 29786 29786 I ActivityThread: Performing receive of Intent { act=de.michelinside.glucodatahandler.GLUCODATA flg=0x30 pkg=de.michelinside.glucodatahandler cmp=de.michelinside.glucodatahandler/.common.receiver.InternalActionReceiver (has extras) }: ***@***.***, appName=de.michelinside.glucodatahandler, pkg=de.michelinside.glucodatahandler, comp={de.michelinside.glucodatahandler/de.michelinside.glucodatahandler.common.receiver.InternalActionReceiver}, dir=/data/app/~~6e6VNj5u89F4s4TEqRNAeg==/de.michelinside.glucodatahandler-YoTbiYzHzGR6Cwdv9tA7TA==/base.apk, duration=32
07-11 17:57:14.187 29786 29786 I ActivityThread: Performing receive of Intent { act=de.michelinside.glucodatahandler.GLUCODATA flg=0x30 pkg=de.michelinside.glucodatahandler cmp=de.michelinside.glucodatahandler/.common.receiver.InternalActionReceiver (has extras) }: ***@***.***, appName=de.michelinside.glucodatahandler, pkg=de.michelinside.glucodatahandler, comp={de.michelinside.glucodatahandler/de.michelinside.glucodatahandler.common.receiver.InternalActionReceiver}, dir=/data/app/~~6e6VNj5u89F4s4TEqRNAeg==/de.michelinside.glucodatahandler-YoTbiYzHzGR6Cwdv9tA7TA==/base.apk, duration=38
07-11 17:58:13.178 29786 29786 I ActivityThread: Performing receive of Intent { act=de.michelinside.glucodatahandler.GLUCODATA flg=0x30 pkg=de.michelinside.glucodatahandler cmp=de.michelinside.glucodatahandler/.common.receiver.InternalActionReceiver (has extras) }: ***@***.***, appName=de.michelinside.glucodatahandler, pkg=de.michelinside.glucodatahandler, comp={de.michelinside.glucodatahandler/de.michelinside.glucodatahandler.common.receiver.InternalActionReceiver}, dir=/data/app/~~6e6V [... cut ...]
07-11 18:24:49.604 29786 29786 I GDH.InternalNotifier: add notifier ***@***.*** - filter: [BROADCAST, IOB_COB_CHANGE, IOB_COB_TIME, MESSAGECLIENT, CAPILITY_INFO, NODE_BATTERY_LEVEL, SETTINGS, CAR_CONNECTION, TIME_VALUE, ALARM_STATE_CHANGED, SOURCE_STATE_CHANGE] - timechanged: true
07-11 18:24:49.604 29786 29786 I GDH.Main: Schedule exact alarm is active
07-11 18:24:49.604 29786 29786 I GDH.Main: Battery optimization is inactive
07-11 18:24:49.604 29786 29786 I GDH.Main: High contrast is inactive
07-11 18:24:49.616 29786 21496 I GDH.Utils: Getting logcat with command: logcat -t 4000

mak2710, 2024-07-11:
[Uploading GDH_PHONE_APP_20240711_182545.txt…]()

mak2710, 2024-07-11:
[GDH_PHONE_APP_20240711_182441.txt](https://github.com/user-attachments/files/16181943/GDH_PHONE_APP_20240711_182441.txt)

[GDH_PHONE_APP_20240711_182545.txt](https://github.com/user-attachments/files/16181945/GDH_PHONE_APP_20240711_182545.txt)

pachi81, 2024-07-11:
Ok, it seems your login settings are wrong and if there is any error in the response, there is a new login, which fails.
But, I do not reset the token in the saved settings, after the restart, the token is used without login and this seems to be valid until the next error occurs and a re-login is tried. You can force the re-login, if you go to the source settings for librelink and enable reconnect. Then during the next interval, a login is executed, which will fail.
Have you changed your password?

I will check, if I have to remove the token from the saved settings during re-login.

mak2710, 2024-07-12:
I've entered again my LibreLinkUp credential yesterday evening, until now the application is working without problem.
I'll keep you informed if the issue comes again.
Thank you for your support

pachi81, 2024-07-12:
Perfect, have you also tried the reconnect switch which force a re-login?
So you can be sure, that your login data is correct and the issue does not happens again.