pachi81/GlucoDataHandler #166: App disconnecting and 'Battery optimizer is enabled' pixel 8 pro

ComptonaPrime, 2024-08-10:
Hey, great app!

The app keeps disconnecting from librelinkup up (not updating) even when forcing a new connection. Not sure if it's linked too battery optimisation is enabled. Although the app (watch and phone) has no restrictions

Kind regards
Andy
[image]
[image]
[image]

pachi81, 2024-08-10:
Have you tried to disable battery optimization?
I still had another one with two phones, on the one everything was fine, even with battery optimzation and on the other one he needs to disable it.
Not sure if it was a Pixel 8 Pro.

ComptonaPrime, 2024-08-16:
Battery optimisation is disabled, but coincidentally since the latest security update for pixel, I've had zero issues.

App still says that battery optimisation is on (it isn't) but been faultless

ComptonaPrime, 2024-08-16:
Issue no longer present

closed 2024-08-16 by ComptonaPrime