pachi81/GlucoDataHandler #167: Startup minimized

lostboy86, 2024-08-21:
Good morning.
Try to explain an example : 

on Android car radio (NOT android auto), you can have the widget mobile over the entire gui or app (google maps, music, etc.)
When you start the radio, quick boot, you can have to autostart / autolaunch Glucodatahandler on boot, ma launche the app itselft, then you scroll up, multitasking, minimize the app, and you have only the widget to move where you want.

Same situation after resume the radio after shutoff the car and power on the car, or if you close the app accidentally.

Can implement an option to start the widget directly, and also auto-minize the app, or start the app minimized, and if user enabled the widget, appear only the widget and not first the entire app?

Same situation for tablet or raspberry, but on this device you not have to not distract when you drive.

Thanks a lot

pachi81, 2024-08-21:
Hi,
not sure, if I understand it correct. After reboot, GDH does not start automatically? I normally have a startup registered after boot for GDH. 
Or is it only related to the floating widget?

pachi81, 2024-08-21:
But after the start of the app, the floating widget is also started or do you have to enable it each time?

pachi81, 2024-08-21:
Is there an option in the autostart app to start minimized? As I´m not sure, if I can detect, that it is not a normal start...

lostboy86, 2024-08-21:
Nothing, no option tò start minimizer

lostboy86, 2024-08-21:
But maybe i misunderstood, i try to explain :

You can implrment a checkbox for user, If cheked, the app start minimized?

froster82, 2024-08-29:
@lostboy86, I tried to read your description and have had a bit of a trouble wit understanding your problem. Maybe this is a matter of information being lost in translation. Can you please try another approach? Can you write everything once more in your native language and then translate the whole message using https://www.deepl.com/ Maybe we can extract a bit more information from this approach. And describe in more detail the system you are using.

lostboy86, 2024-08-29:
I ask if possibile to start the app, without main app screen prompted,
directly to widget.

The system of android car its android 10.

pachi81, 2024-08-29:
My problem is, if I add such a checkbox it will never be possible to start the app again, if it always start minimized. 
I don't think it is possible and the app already receives the boot command, which starts the background work...
Maybe the car implementation does not send such commands.

lostboy86, 2024-08-29:
Maybe a long Press tò the widget to enable/disable this?

Strange because on phone start directly to widget, on radio start the app,
popup the main screen app Glucodatahandler,then the widget

pachi81, 2024-08-30:
Hi @lostboy86,

can you try this version:
https://drive.google.com/drive/folders/1SFBdrZBXvgZwPBfHaiCpssyuxZOXZwRk

I added some more boot receiver. Maybe this already solves the problem...

lostboy86, 2024-09-04:
Fixed in the last version in the link above, thanks mate!!

pachi81, 2024-09-04:
Perfect! Will be available in version 1.2

closed 2024-09-04 by pachi81