pachi81/GlucoDataHandler #168: LibreLinkUp - notAuthenticated

gregorgregor25, 2024-08-22:
Hi,

downlaoded app and installed on s23 ultra and installed companion app on galaxy watch 4.  I used the _Sources_ menu to add my LibreLinkUp account and chose to enable.  Within the phone app under _Connections_  it shows LibreLinkUp 2: notAuthenticated.

LibreLinkUp is receiving the Data from the LIbre 2 app. 

I cannot figure out what is going wrong here. I've addded my logs to a pastebin. 

https://pastebin.com/UZv1ZXfZ

[image]

Any help is appreciated. Thank you

pachi81, 2024-08-22:
For libre 3 you have to enable LibreLinkUp first and connect it in the libre app. Not sure about libre 2.
Please download the LibreLinkUp app and check, if it is working there.

gregorgregor25, 2024-08-22:
Hi thanks for replying :) I am using Libre 2. I have LibreLinkUp, installed and it is recieving data from the LIbre 2 app. 

[image]

pachi81, 2024-08-22:
Ok, so please re-enter your account data, maybe there is a small typo.

pachi81, 2024-08-22:
Just one question, are you using libre 2 or libre 3? Does the libre 2 app support continuous reading?

gregorgregor25, 2024-08-22:
I read online that there has been some issues with LibreLinkUp over the last few days. I reset my password for LibreLinkUp and re-added it to the glucodatahandler app and it's now working. Sorry to bother you and thanks for your help! Your app is great!

closed 2024-08-22 by gregorgregor25

pachi81, 2024-08-22:
Great!
Just for interest, does the libre 2 app supports continuous reading and upload or are you using libre 2?