pachi81/GlucoDataHandler #171: bigger floating widget dimension

ghino18, 2024-09-24:
hi, it is not really an issue but i ask if it is possibile to have big dimension.
now it arrive to 10, i ask  if it is possible to double until 20.
best regards

pachi81, 2024-10-04:
This is how it look like with 30:
[image]

And this with 20:
[image]

pachi81, 2024-10-04:
I think 20 is enough. What do you think?

ghino18, 2024-10-04:
I used it on a android Watch and i have to try with Its screen to be bigger as possible to be and all screen. If 30 Is not a problem to inser i would like try , i have more choice

pachi81, 2024-10-04:
Ok, thank you for the feedback.
As you can see, it is not a problem, but I have to check the slider, as it has a lot of steps, now.
So to set the setting is a little bit more complicated.

closed 2024-11-11 by pachi81

pachi81, 2024-11-24:
@ghino18 just a question, what size are you using, now?