pachi81/GlucoDataHandler #172: LibreLink error 4

ScottyTM, 2024-09-28:
I'm using Librelink and had some issues in the past couple of days with the data sync.

I found this in the logs:
09-27 23:08:21.024 30338  5065 I GDH.Task.Source.LibreLinkTask: Send request to https://api.libreview.io/llu/auth/login
09-27 23:08:21.439 30338  5065 W GDH.Task.Source.DataSourceTask: Set state for source LIBRELINK: ERROR - Error (4)

A [comment](https://gist.github.com/khskekec/6c13ba01b10d3018d816706a32ae8ab2?permalink_comment_id=4301967#gistcomment-4301967) on a gist about the LibreLinkUp API pointed me in the right direction.

Apparently the Terms of Use have been updated. I had to log out of the LibreLinkUp App and log back in. Then I had to accept the new terms. 

It's been working since then.

Maybe this can be outmated somehow.

pachi81, 2024-09-28:
Hi,
thank you for posting the solution! I already know this problem from some facebook threads. There are several solutions provided, so I wasn't really sure, what is the correct one.

Unfortunately there is a GDH user, which accepted the terms of use, but the error occurs again after some hours.
Maybe it is related to error 429 (Too many requests), as he also using LibreLinkUp App in parallel. 

We will see.

Best regards, 

Michael

fpalab, 2024-10-25:
Hello. Installed yesterday and was greeted by the error 4: solved by accepting the Tou.

This step can also be performed from an app or a web service by calling the libreview.io api.
When the the login call returns a status 4, it also adds to the reply something like: 
`		"step": {
			"type": "tou",
			"componentName": "AcceptDocument",
			"props": {
				"reaccept": true,
				"titleKey": "Common.termsOfUse",
				"type": "tou"
			}
		},
`
basically asking to reaccept the Tou. 

Calling the service https://api.libreview.io/auth/continue/tou will fix that (returned status is 0). Just use the token returned by the half-failed login in the Bearer header.

You can find more details at this site [https://libreview-unofficial.stoplight.io/](https://libreview-unofficial.stoplight.io/), in case you don't already know it.

Anyway your app solved my issue to get the glucose on the car screen, as all of my Tasker customizations were basically neutralized by android auto.
Thanks!
Francesco

fpalab, 2024-10-25:
I went through my notes and I could find this full reply to user/pass login request: 
```
{
  "status": 4,
  "data": {
    "step": {
      "type": "tou",
      "componentName": "AcceptDocument",
      "props": {
        "reaccept": true,
        "titleKey": "Common.termsOfUse",
        "type": "tou"
      }
    },
    "user": {
      "accountType": "pat",
      "country": "CH",
      "uiLanguage": "en-US"
    },
    "authTicket": {
      "token": "the token here",
      "expires": 1719337966,
      "duration": 3600000
    }
  }
}
```

This is a reply for a registered account that needs to (re)accept the tou. This should happen whenever new Terms of Use are reissued, but it only shows when doing a user/pass login. That's why it's so difficult to test it.

The key here is data.step.type: its value defines the page to call for acceptance. In this case it is "tou" and thus refers to the url https://api.libreview.io/auth/continue/tou

In another case it was "pp" (I think it was for token renewal) and the url becomes 
https://api.libreview.io/auth/continue/pp.  I can't find the original info for this though.

In both cases reusing the provided token is mandatory. I assume the old one is invalidated. 
I hope it helps you.

pachi81, 2024-10-25:
@fpalab thank you very much!!!
At the moment, I´m still testing the next version, so I will add this in the version after.
Regards, Michael

pachi81, 2024-11-07:
@fpalab Just one more question. Do you also know, if there is an API endpoint to accept the mail invitation?

fpalab, 2024-11-07:
@pachi81  I have no idea, I never found something about that.

However, something can be inferred from the process. 
The registration to Libre LinkUp requires getting an email for confirmation: I doubt there's an alternative Api. Although it's always possible to read the email and fetch the link (quite complex and invasive).

An alternative for those who just wish to read their own glucose on your app, is to use the same account as the LibreLink, and yes it works. Just register yourself for listening on LibreLink and then log in with the Api.
The drawback back here is the the user gives the main password to a third party app, which is not a secure practice.

Hope it helped a bit.

pachi81, 2024-11-07:
@fpalab thank you for your information.
I already use it with Librelinkup, but only for testing.
So I hoped, there is also a solution for the invitation, because some user does not understand it and also some users are thinking, that they need the LibreLinkUp app for my app, not only for setting up the user.

But if I can offer the tou, this will help.

Thank you for your information.

pachi81, 2024-11-15:
@fpalab: Very important! Abbott changed their interface. 
You need to use a new min version, but it seems, that you also need more header-data. 
Do you know anything about???

pachi81, 2024-11-23:
@fpalab: one more question to accept tou.
Do I need to set anything special? I´m using the token from the message, but get an error in response:
`Code 401 received with message Unauthorized  {"message":"MissingCachedUser"}`

Thank you!

commit 8d7a7238 (2024-11-24): Fix #172: error4 handling
closed 2025-01-22 by pachi81