pachi81/GlucoDataHandler #174: GlucoDataHandler is not showing up the values at all on the lock screen randomly

cruiserusa, 2024-11-02:
The GlucoDataHandler is not showing up the values at all on the lock screen randomly and I don't know the cause.

I installed GlucoDataHandler on an android Samsung A23, set it to show the values from a LibreLinkUp connection on the on the lock screen, and everything was fine for a couple of days. After that, the lock screen started not showing up the values anymore, I mean it looked like the widget simply didn't load up. I opened and lock the screen and no success as the lock screen started showing me up a generic colorful background from Samsung with the time and day and NO MORE A BLACK BACKGROUND WITH THE GLUCOSE VALUES. It looked like the widget didn't load or was hidden.

So to fix it I had to fully power off the phone and power it on, wait for some minutes and unlock the phone and the locked it again to have the BLACK BACKGROUND WITH THE GLUCOSE VALUES back again. In any moment I had to open the GlucoDataHandler app to change its settings.

I've checked if there is a Software Update option to install new versions automatically and couldn't see this option on the Samsung A23, so I don't see this could be the cause. BUT could be some app from the Google Play that are messing up with updates on the user interface. However the phone has basic and standard apps.

This has been happening often and randomly. Any help is appreciate. Thanks.

https://www.reddit.com/r/Freestylelibre/comments/1gi539o/glucodatahandler_is_not_showing_up_the_values_at/

pachi81, 2024-11-02:
One more question, have you granted all permission and have you disabled battery optimization?

If you do not use the playstore version, please also disable Play Protect.

cruiserusa, 2024-11-08:
Thank you for the feedback. I've checked the battery optimization was not disabled, so now I disabled it and so far so good for some days. I'll back if the issue returns.
Just answering your previous questions, the background had changed to a default one with a picture. It was download and installed from the official Google PlayStore and the phone is under Android 14.

pachi81, 2024-11-08:
Ok, thank you for the feedback.
That is really strange, the part, that the image was set. I had expected, that the value will not change any more, but it will stay.
Unfortunately I can not really test it for a long time, because I use lockstar to get the data on AOD, too. And the AOD in lockstar will only work, if you also use the lockscreen in this app...
Hopefully it will work, now.
If there are any problems, please send me the logs.
Thank you!

Regards, Michael

closed 2025-01-22 by pachi81