pachi81/GlucoDataHandler #175: Wear OS 5 - Unable to select source

Bi0-delete, 2024-11-04:
The app appears to work and I can add complications but the source section on the pixel 3 appears to be greyed out for the sources and doesn't expand to input the details like it would for the phone. 

Here's a screenshot of what is shown:  
[image]

pachi81, 2024-11-04:
For normal usage (not follower of another one), you do not need to activate the source on watch, as the watch receives the data from phone.
In your case, the connection to the phone seems not to work correct, you may see in main screen on phone and/or watch for connection "?%".
Please try to restart your watch and check if it solves the problem.
If not, you can try the current beta version, where I try to fix this problem:
https://play.google.com/apps/testing/de.michelinside.glucodatahandler

Bi0-delete, 2024-11-04:
I tried the restart and there was no change.  So I installed 1.2 beta and no change as well I do have the watch yelling at me to reset now.  lol  Could this be related to the same issue Blose is having with OS 5:  https://issuetracker.google.com/issues/365493102?pli=1

pachi81, 2024-11-05:
But I think, it is only yelling at you to reset connection...
But it won't work, because there seems to be another problem.
Does the watch also show a connection to phone with ?% or does the watch show nothing?
If there is no connection to phone shown on watch, it may be one of these problems:
- you are using private profile on phone
- there is no phone watch connection at all, no notification from any app is shown

Solutions: 
- move to normal profile on phone, as there is no connection to watch in private profile 
- you have to do a factory reset of the watch
Another user had the problem with the notifications not showing on watch. After he did a factory reset, everything is working, now.

The problem with the exact alarm is only relevant for my app, but is not related to the connection issue. The app is also working without this permission, but you can grant it, if you know how to use adb.

Regards, Michael

Bi0-delete, 2024-11-05:
Thanks for the information, that might make sense as the watch app isn't seeing my watches configuration changes and it was saying I needed to reset the device.  Which is odd because I just got the new pixel 3 today and set it up for the first time.

pachi81, 2024-11-05:
Ok, seems to be also an issue of the new pixel watch 3...
You are already the second one.
So you may have to setup your watch again.

Bi0-delete, 2024-11-05:
I'm going through the factory reset now.  So we shall see but yeah it's seeming like maybe there's some issues with OS 5 and the pixel 3.  Already got an error that it couldn't connect to fitbit.

Bi0-delete, 2024-11-05:
I had to reset it a 2nd time, BUT it looks like it's working now.  I see glucose readings in the app on the watch.  That said I do see the "?%" mentioned now.

closed 2024-11-05 by Bi0-delete

pachi81, 2024-11-05:
Great to hear! Yes, I added some self-fix for the connection and it seems to work :-)
Thank you for your response!
Enjoy the app!
Regards, Michael