pachi81/GlucoDataHandler #176: Tapping watch glucose complication

Tronutopia, 2024-11-06:
Any plans to show a graph when the watch complication is tapped ?

Both G-watch and Blose display a nice glucose graph when their complications are tapped.

I really love glucosedatahandler, but I miss not seeing the graph.

pachi81, 2024-11-06:
Yes, graph is planned, but it will take some time to implement. Maybe at the beginning of next year.
I already tried to open the G-Watch graph as tap action, but it is not exported to external apps.

Tronutopia, 2024-11-06:
The G-Watch graph works well (on my Galaxy Watch 6), BUT unfortunately G-Watch is no longer supported and does NOT work on my new Galaxy Watch 7.
So I went looking for a new watch app that works on Galaxy Watch 7 and found Blose.
But I didn't know that GlucoseDataHandler even existed until I read a facebook comment, so I thought I'd try it.

I love it. So many great options that work well.
Love the floating widget.

I love these apps that help us diabetics (I'm type 1), to manage our life.

Well done!!

pachi81, 2024-11-06:
Thank you very much! I´m also type 1 and have looked for a solution. As I didn´t found a good working solution, which fits my need, I decided to try out developing my own app. This is my first app and I do not have any experience in app developing. 
I also do it in my free-time, so every feature will take some time...
As I do not really need/use the graph, there is no one at the moment.
But there are so many requests to add it, so I will start doing it, after the next version is released.
Enjoy the app, and you can also ask me using the GlucoDataHandler facebook group.

Regards, Michael

pachi81, 2024-12-18:
I will close this as duplicate/extension of #154

closed 2024-12-18 by pachi81