pachi81/GlucoDataHandler #177: Wear OS: Sound at beginning of Alarm before vibration

kdupke, 2024-11-07:
Thank you for this great piece of work!

Samsung Watch 6
* UI 5.0
* GDH 1.1

Xiaomi
* LineageOS 20 / Android 13
* GDH 1.1

On my phone I have configured a low alert with no sound. Everything is fine.  Works the same on my watch.

On my phone I have configured very low alert with 30 seconds delay. During this delay the phone vibrates. Then the sounds kick in. Everything is fine.

On my watch at the begin of an alarm I get somewhat like a notification sound. Then 30 seconds of vibration. Then the full sound.

However, this notification sound already is enough to wake me my wife up. Which of course I would like to disable and have it the same way as on my phone.

Any idea what I may have configured wrong? 
For me it would be OK to have 30 secs of vibration on my watch and then alarm on my phone. But as soon I have my watch running alerts are only active on the watch - which I also see as a cool feature.

If more information or clarification is needed I would be glad to provide this.

regards, kai

pachi81, 2024-11-07:
Hi,

to have sich a custom vibration and sound, I can not use the vibration and sound of the notification, as this can not changed on watch.
So I'm using a notification with a short vibration, before I'm start the vibration and sound you have configured. So this is played independent from the notification.

So I think the short sound is not related to the watch notification.
But Wear OS also shows the notification from phone and for these notification it normally plays a short notification sound, which I can not change. 

You can test it: on phone, go to alarm settings -> advanced and enable "no notification if watch connected".
Then trigger a test alarm to find out, if there is no short sound.

The problem is, you can not disable the notification from phone on watch only, as this will also disable the watch notification. 
As you see, it is really complicated...

Michael

kdupke, 2024-11-07:
Let me thank you for the quick reply!

Of course I have played a bit with it now that I have started to file this issue.

Setting about 'no notification if watch is connected' does not make a difference.

However, as part of this I also updated everything. And after some problems on the watch I was able to update the watch as well (had to disable Bluetooth and run the update from the watch via WIFI alone).

After this, the problem disappeared. 

I will close this issue now. In case the problem will come back, I will reopen it.

pachi81, 2024-11-07:
Ok, thank you for the feedback!
So it seems, that the watch also have a sound for the notification, also as I not provide any.

If the problem occurs again, you can also try to disable sound (only vibration) on watch and enable in GDH on watch to force sound, so hopefully the short sound will not appear, but the GDH sound after 30s.

Regards,
Michael

kdupke, 2024-11-08:
I am sorry, I have to reopen this.

If I run the test everything is fine, but in real life it still has the issue.

I will do some more research with the phone and with the watch connected and report it back here.

Sorry for the loop & hope an update would have fixed it.

pachi81, 2024-11-08:
Have you triggered the test from phone?

If the problem occurs, was there also a phone notification or the watch notification only!?

kdupke, 2024-11-08:
Thank you for the very quick response.

Triggering a test from phone and watch works fine.

My watch is set to vibration only (always).
During night is is set to power-saving + vibration only.
And my phone is then set to don't disturb.

I will do more tests. I already found that 'low glucose' - which for me has no sound but vibration - only vibrate 4 times. But that seems to be a different issue.

pachi81, 2024-11-08:
The default vibration is such "short". But you can increase by setting the sound delay, even if you do not use a sound.

One more question, you are using the watch in vibration, but there is a sound on watch? Have enabled to force sound? 
Because I always use my watch in vibration mode and never get a sound, only if I activate sounds on watch.

kdupke, 2024-11-08:
(what is about having a button in the phone app to have an easy c&p setting report for phone *and* watch? Sorry, having a background with enterprise support :-)

Watch settings are
(manual translated from German)
Alarm on

extended setting
only vibration off
Use volume of alarm clock
Force alarm sound on
no popup with phone connected: no
Delay 1.5 sec

very deep value setting
volume -1
external sound off

kdupke, 2024-11-08:
`The default vibration is such "short". But you can increase by setting the sound delay, even if you do not use a sound.`

Means I can set this to 30sec max, right? I wonder if it is possible to configure it in a way that it vibrates till I actively confirm?

Might be I create another issue for this (question)? Also to make it easier for others to find.

pachi81, 2024-11-08:
OK, am besten weiter auf deutsch...
Mach mal Alarm Ton erzwingen aus, dann vibriert die Uhr nur... 
Du könntest das auch über Tasker ändern, wenn du es zum Beispiel es nur nachts haben möchtest.

Wegen der unendlichen Vibration hatte ich überlegt, aber erstmal verworfen, das Problem ist, dass ich aktuell schon so viele Einstellungen für die Alarme habe, dass das langsam keine mehr versteht, daher habe ich das verworfen. 
Kannst aber das "Wiedererscheinen" nach einer Minute wählen.

kdupke, 2024-11-08:
Ich würde mein Setup ungern verkomplizieren. Jetzt komme ich damit zurecht, aber in 20 Jahren möchte ich das auch noch nutzen können.

Ich will ja laute Alarme haben, aber quasi einen Vorlauf in Vibration.

Dann vibriert meine Uhr, im Normalfall merke ich das.
Wenn ich es nicht merken würde (weil ggf. richtig tief unterzuckert), dann will ich natürlich den Alarm, der dann meine Frau wecken würde.
Mit Vorlauf kann ich reagieren, ohne dass meine Frau aufwacht. 
Dazu liegt Traubenzucker/Müßliriegel bereit, da brauche ich noch noch nicht mal Licht anzumachen.

Ohne Vorlauf bzw. mit dem beep-beep, wacht meine Frau ggf. schon mit mir auf.

kdupke, 2024-11-08:
Alarm bis zum abwinken.

Ich würde das anders machen, als globale Option, die sagt, Alarm muß explizit bestätigt werden.
Das passt gut zu dem Bereich mit force Alarm würde ich sagen.

NB: Ich nutze die Alarme ggf. etwas anders als andere/ursprünglich geplant.
Ich bin Typ 2 und habe meine tiefe Schwelle bei 80 und meine sehr tiefe bei 75.
Das reicht in der Regel, damit ich nicht unter 70 komme und damit in meinem Zielbereich bleibe. 
Daher bei 80 dann nur Vibration.
Und bei 75 dann mit Sound - aber halt gerne so, dass ich darauf reagieren kann, bevor meine Umgebung das mitbekommt oder meine Frau aufwacht.

kdupke, 2024-11-08:
Ich bin noch auf dem Handy am testen gerade mit den verschiedenen Einstellungen. Danach mache ich das mit Handy und Uhr. Dann reporte ich mal meine Ergebnisse (bzw. meine Probleme, die ich dann noch habe). Das braucht etwas, weil ich muß ja immer warten, bis mein BZ dann wieder unter die Schwelle (die passe ich derzeit laufend an and meinen echten Wert) fällt für den entsprechenden Alarm.

pachi81, 2024-11-08:
Ja, das mit den Anpassen kenne ich sehr gut :-D
Aber eigentlich solltest du den Ton auf der Uhr hören, wenn du den Alarm über Telefon testest. Zumindest konnte ich es so nachvollziehen. Ich bin aktuell am Release von Version 1.2, werde aber schauen, ob ich den Fix noch einbauen kann.

pachi81, 2024-11-08:
So, ich nochmal, also ich habe keinen Einfluss auf den Standard-Benachrichtigungston auf der Uhr. Egal was ich setze, es kommt immer der eingestellte Ton von der Uhr.
Was ich auf meiner Seite machen kann: wenn Ton erzwingen eingestellt ist, mache ich das sofort, auch wenn eine Verzögerung aktiv ist... 
Das müsste ich erst nach der Verzögerung machen, dann wäre es möglich, dass die Uhr solange die Verzögerung dauert, im Vibrationsmodus bleibt und erst danach der Ton für meinen Alarm aktiviert wird. 
Das wäre aber keine Änderung mehr, die ich für Version 1.2 machen kann, da die erst getestet werden muss.

Workaround: du kannst in den Uhr-Einstellungen als Benachrichtigungston Stumm auswählen, dann bekommst du den Ton nicht mehr. Allerdings gilt das dann für alle Benachrichtigungen auf der Uhr.

kdupke, 2024-11-08:
Danke für Deine Nachforschungen.

Handy habe ich jetzt durch, da klappt alles, wie es sein sollte. Also nur Vibro macht nur Vibro und Vibro + Ton macht genau das.

Uhr kommt als nächstes dran.

Ich habe schon überlegt, ob ich mir nicht eine MP3 erzeuge, die vorne 30 Sekunden Stille hat und dann erst den normalen Ton.

Das sollte doch zusammen mit Vibration klappen, oder?

Benachrichtigungstöne auf der Uhr brauche ich nicht, vor allem, da ich die normal auf Vibration habe. Aber ich habe nirgends eine Einstellung für Benachrichtigungstöne gefunden. Da muß ich nochmal suchen.

pachi81, 2024-11-08:
Auch wenn du eine MP3 mit stille nimmst, ändert das nix daran, dass die Uhr immer für jede Benachrichtigung ihren Standard-Ton abspielt.

Ich habe die Galaxy Watch 5 Pro, daher wird die Einstellungen bei dir genauso sein:
Einstellungen -> Töne und Vibration -> Scrollen bis "Benachrichtigungston" -> "Stumm" auswählen.

kdupke, 2024-11-08:
Mit Benachrichtigungston auf stumm geht es, wie es soll. 

Damit habe ich einen Workaround.

Danke!

pachi81, 2024-11-08:
Super! Und ich habe noch etwas Zeit, damit rum zuspielen 😀

closed 2025-01-22 by pachi81