pachi81/GlucoDataHandler #178: Request for Libre 2 Plus with Tandem T:connect

bsaunder2002, 2024-11-12:
Tandem t:slim, looping, t:connect app on phone, pump loops straight with Libre 2 Plus sensor, no need for phone app.
Need more robust data handling / alerting.
Could GlucoDataHandler be a second Libre 2 plus reader?  
Since tandem runs the t:connect and has nothing to do with Libre, I dont believe the LibreLink will work.
Would be willing to invest in this.
Data management options really stink right now.

pachi81, 2024-11-12:
No GDH is not able to connect direct to a sensor. If you are looking for an alternative app, you can try Juggluco.

bsaunder2002, 2024-11-12:
Thanks for the quick reply.  I've got a request in for Juggluco already.  Unsure if Libre 2 Plus can handle 2x simultaneous readers.
Wasnt sure if GlucoDataHandler could talk to the t:connect app or the Tandem online dohickey.

pachi81, 2024-11-12:
Never heard about these apps...