pachi81/GlucoDataHandler #181: Librelink 3 keine Verbindung

Holger4711, 2024-11-15:
Keine Verbindung mehr mit Libre 3 .
Beim mir und meiner Frau. 
Alle Einstellungen zurückgesetzt und probiert. 
Kommt immer Error. 
Wir haben ein Samsung S23 EE

pachi81, 2024-11-15:
Bei mir auch. Ich schaue gerade...

Im schlimmsten Fall hat Abbott die Schnittstelle dich gemacht...

Liedmeister, 2024-11-15:
Betrifft denke ich tatsächlich die Schnittstelle @pachi81. Sweet Dreams auf iOS hat auch keine aktuellen Werte 🥺. Ich hoffe es gibt noch irgendeine Lösung, ich liebe deine App.

Wassermann73, 2024-11-15:
> Bei mir auch. Ich schaue gerade...
> 
> Im schlimmsten Fall hat Abbott die Schnittstelle dich gemacht...

Also, bei mir funktioniert das auch nicht, allerdings läuft meine Abfrage auf meiner Garmin Fenix 7 pro noch. Hier läuft das auch über LibreLinkUp, der Kanal kann also nicht ganz zu sein.

Skullghost2003, 2024-11-15:
Id guess this would have to do with the new 2FA for libreview. Logging into libreview directly requires this now. I found no way to disable 2FA in libreview yet

guenn75, 2024-11-15:
Seit heute ist es ja notwendig, die neue Version der librelinkup App zu nutzen. Wahrscheinlich hängt es damit zusammen.
Hab mit Abbot telefoniert, librelinkup funktioniert. Zu xdrip oder glucodatahandler können sie nichts sagen.

The 2FA was already necessary before at liebreview. I am using it since beginning of the year. So this is not the Problem in my opinion, because the Data is coming from librelinkup

svenlohberg, 2024-11-15:
2FA was active for me for a long time, bit GlucoDataHandler was able to connect. So I don't think it is related to the 2FA. I also get the 403 error since a few hours.

pachi81, 2024-11-15:
I have checked API and there is a new version in header needed. But even with this version I get error "RequiredHeaderMissing"
So there is anything new. I don´t know what. Tried to capture LibreLinkUp App without success....

gui-dos, 2024-11-15:
@pachi81:

You can take a look to my very quick Swift fix: https://github.com/JohanDegraeve/xdripswift/commit/13e4a1e7418364fbdb68e5bd54cac888fd2eb9b8  ;-)

pachi81, 2024-11-15:
Thank you! I will check!!!

Liedmeister, 2024-11-15:
@pachi81 
Thank you very much for your commitment

pachi81, 2024-11-15:
@gui-dos it works!!!!!!!!!!!!!!!!!!!!!!!

Thank you sooooo much!!!!!!!
I will create an update as fast as possible!!!

pachi81, 2024-11-15:
Version 1.2.1 is online: https://github.com/pachi81/GlucoDataHandler/releases/tag/1.2.1

guenn75, 2024-11-15:
Thanks for the quick fix, you saved my day

Holger4711, 2024-11-15:
Neue Version geht wieder. Nur einmal bei der Libre up App abmelden und wieder anmelden. 
Natürlich auch wieder alles bestätigen. Sie da es geht. 
Tolle schnelle Arbeit. Klasse

pachi81, 2024-11-15:
> > Version 1.2.1 is online: https://github.com/pachi81/GlucoDataHandler/releases/tag/1.2.1
> 
> Hallo, gibt es auch eine Version für die Uhr?

Ist doch dort, die Datei, welche mit Wear endet, aber die Uhr erhält die Daten von Telefon, ein Update dort ist nur nötig, wenn du die Uhr als standalone follower verwendest.

Jumpy1976, 2024-11-15:
> Ist doch dort, die Datei, welche mit Wear endet, aber die Uhr erhält die Daten von Telefon, ein Update dort ist nur nötig, wenn du die Uhr als standalone follower verwendest.

Ich konnte die alte Version auf der Uhr installieren mit der neuen geht das nicht die finde ich da nicht

Jumpy1976, 2024-11-15:
Und wenn die mit Wear runterlade kommt der Fehler das sie mit meinem Telefon nicht kompatibel ist

pachi81, 2024-11-15:
Ist ja auch für die Uhr...
Die kannst über Wear Installer installieren.
Google muss die Version erst für den Playstore freigeben, das kann dauern... 😕

Jumpy1976, 2024-11-15:
> Ist ja auch für die Uhr... Die kannst über Wear Installer installieren. Google muss die Version erst für den Playstore freigeben, das kann dauern... 😕

Ok dann warte ich ein bisschen 😉

pachi81, 2024-11-15:
> @pachi81:
> 
> You can take a look at my very quick Swift fix: [JohanDegraeve/xdripswift@13e4a1e](https://github.com/JohanDegraeve/xdripswift/commit/13e4a1e7418364fbdb68e5bd54cac888fd2eb9b8) ;-)

@gui-dos just one more remark, you also have to update your version to 4.12.0 at least for the german servers.
One more question, how did you get this information? You can also contact me direct: GlucoDataHandler@michel-inside.de 

Thanks again, you saved my day!!! 😉

gui-dos, 2024-11-15:
@pachi81 

I reported it already on https://github.com/timoschlueter/nightscout-librelink-up/issues/155 but it was ignored until today: you can spot the new header in the beautified _bundle.js_ and _vendor.js_ since LibreLinkUp is written in multi-platform NativeScript ;-)

pachi81, 2024-11-15:
> @pachi81
> 
> I reported it already on [timoschlueter/nightscout-librelink-up#155](https://github.com/timoschlueter/nightscout-librelink-up/issues/155) but it was ignored until today: you can spot the new header in the beautified _bundle.js_ and _vendor.js_ since LibreLinkUp is written in multi-platform NativeScript ;-)

Ok, I have to check! Thank you for the hint. I have tried half a day to get a proxy server to work to capture the packages, but I had no success...

FloJun88, 2024-11-16:
Hallo,

ab wann ist die neue Version im Playstore erhältlich? Darf auf meinen Firmentelefon nur darüber installieren -.-

Vielen Dank im Voraus. 

BG

pachi81, 2024-11-16:
Fixed. New Playstore version is also live.

closed 2024-11-16 by pachi81

Liedmeister, 2024-11-16:
Thank you very very much. @pachi81 
Great job.

svenlohberg, 2024-11-17:
Thanks a lot @pachi81.