pachi81/GlucoDataHandler #182: LibreLinkUp Datenaktualisierung

nilremehm, 2024-11-16:
Ich benutze LibreLinkUp als Datenquelle.
Leider kommen da nur alle 15 Minuten neue Werte.
Es wäre schön, wenn die Zeit zur Markierung veralteter Werte auf mindestens 15 Minuten eingestellt werden könnte.
Außerdem wird bei mir kein Delta-Wert angezeigt. Ich vermute mal, dass das auch an dem langen Datenintervall liegt.
Könnte hier vielleicht einfach die Differenz zwischen neuem und alten Wert angezeigt werden?

pachi81, 2024-11-16:
Hast du das Intervall auf 15 Minuten stehen? 
Normalerweise sollten minütlich neue Werte kommen. Wenn nicht, mal in der Libre App den Follower neu einrichten.

nilremehm, 2024-11-16:
Mir ist nicht bekannt, wo ich in der Libre 3 App einen Intervall einstellen kann.
In der Libre App kommen die Daten minütlich. Den Follower habe ich schon mehrmals neu eingerichtet.
Mir genügt ja der 15 Minuten Abstand, aber da die Einstellung für veraltete Werte auf 12 Minuten begrenzt ist bekomme ich 3 Minuten lang einen veralteten Wert angezeigt.

pachi81, 2024-11-16:
In GlucoDataHandler kannst du das Intervall konfigurieren. 
Ich persönlich empfinde 15 min schon lange, daher habe ich die Zeiten so gewählt.
Bei Libre App kann man entweder den LibreLinkUp Account entfernen und neu hinzufügen oder die Libre App neu installieren. Ist ein bekanntes Problem. 
Aber ich könnte natürlich auch die Zeit auf 15 Minuten erhöhen.

nilremehm, 2024-11-16:
Das habe ich alles schon mehrfach versucht. Ohne Erfolg.
Kürzere Abstände wären mir auch lieber, aber 15 Minuten sind besser als keine Werte auf der Smartwatch.
Warum bekomme ich denn keinen Delta-Wert?

pachi81, 2024-11-16:
Ich glaube bei Delta sind es auch 10 Minuten als maximum.
In GDH hast du aber 1 min Interval aktiv?

nilremehm, 2024-11-16:
Ja, habe ich. Das Problem liegt bei LibreLinkUp. 
In der LibreLinkUp App kommen die Werte auch nur alle 15 Minuten.

pachi81, 2024-11-16:
Ok, dann werde ich mal das Intervall auf 15 min erhöhen.
Aber schon komisch. 
Ich nutze Juggluco, da gibt es solche Probleme nicht... 😉

nilremehm, 2024-11-16:
Vielen Dank.
Ich werde mich wohl auch mal mit Juggluco vetraut machen.

closed 2025-01-22 by pachi81