pachi81/GlucoDataHandler #183: Allow setting alarm repeat interval in seconds

WebFreak001, 2024-11-22:
I would like to repeat notifications more often than a minute, until they are dismissed manually by the user. Currently the minimum repeat interval is 1 minute, it would be nice to have this be 1 second instead.

This is especially important for me on a smart watch where I want it to keep vibrating until the notification is dismissed and never otherwise stop.

Currently the alarm for "very low" is never ending and always repeats, which is good. But I would like to mirror this for the other alarms as well. Currently it seems like "very low" is just special since even with the same exact settings for other alarms, they don't repeat infinitely like it does.

pachi81, 2024-11-22:
Ah, sorry, my mistake. You mean the retrigger time not the alarm interval...

But I think with the repeat until close it will be solved!?

Retrigger in seconds can not be handled as the slider is really ugly...

WebFreak001, 2024-11-22:
> the new repeat until close setting will solve this problem

I don't see this setting anywhere. (1.2.1)

I want the audio and vibrations to repeat until I dismiss the notification. Right now the notification just plays the sound just once and then stops, same with vibration that plays during the sound.

In my mind the easiest way to achieve this would be to change the setting that re-triggers the sound & vibration, which I think people would only really set to either 0 (no repeat), 1 minute or at most 2 minutes to a scale of 0-120s - however if there is another setting to do this that would work too

pachi81, 2024-11-22:
Repeat until close will be part of version 1.3 which is still in development. I started the implementation of this feature some minutes before I get your post... 🤣

At the moment, you can use the sound delay setting for long vibration. For having a long sound, you can use a custom sound, which is quite long...
For example, if you are using a 3 minutes sound, you have 3 minutes sound + vibration.

WebFreak001, 2024-11-23:
thanks, that suits my use-case

closed 2024-11-23 by WebFreak001