pachi81/GlucoDataHandler #185: No values on Samsung watch for almost a week

FlorisQK, 2024-11-25:
GlucoseDataHandler was working fine for months.
(sometimes an error but always back on after restarting the watch)
But now since last Weddnesday not getting any values anymore.

Getting different errors:
when I restart my watch
first LinbrerLinkUp error 2:not authenticated 
then LibreLinkUp error (just general)
then LibreLinkUP 429 error (the most)

I tried reconnecting with LibreLinkup, reset connection within DataGlucoseHandler app, restarting the watch, leaving it off for a while etc. etc.
Would love to have this back on my watch because it makes daily life easier.
Anyone a suggestion how to fix this or another solution.
Thanks for any help in advance.

pachi81, 2024-11-25:
Hi,
it seems Abbott did some more changes to their servers and blocks too many requests (429).
So it is important to use only one follower app for your account! Please stop or uninstall LibreLinkUp app.
Also stop the source in GDH for some minutes.
If you have several devices, following the same account, please check to disable them or use different accounts.
Please also stop the data source in watch version as it is not needed, if you already receive the data in phone version.

Regards, Michael

FlorisQK, 2024-11-25:
> Hi, it seems Abbott did some more changes to their servers and blocks too many requests (429). So it is important to use only one follower app for your account! Please stop or uninstall LibreLinkUp [...]

Thanks. No luck yet but I keep trying and let you know. Now only running GDH on my watch.

FlorisQK, 2024-11-28:
Still working so looking for other solutions. Thanks. And maybe I can ask abbott about the 429..

pachi81, 2024-11-28:
Abbott will not support 3rd party apps.
What about LibreLinkUp? Is this app working?

FlorisQK, 2024-11-28:
yes LibreLinkUp app is working. (Has always been working. Did a reinstall with a new login just to try if this helps with the 429).

pachi81, 2024-11-28:
Are you have always used LibreLinkUp App?

FlorisQK, 2024-11-28:
yes always. thanks

pachi81, 2024-11-28:
> Hi, it seems Abbott did some more changes to their servers and blocks too many requests (429). So it is important to use only one follower app for your account! Please stop or uninstall LibreLinkUp [...]

Have you red this???

FlorisQK, 2024-11-28:
yes and tried all. made a whole new account and only used GDH on phone with it. But no luck.

pachi81, 2024-11-28:
Maybe it is not related to the account, but to the device...
Please completely stop the LibreLinkUp app for at least an hour and set the interval to 5 minutes in GDH...

FlorisQK, 2024-11-28:
Thanks will try this later. (need to find the right moment). I'll let you know.

pachi81, 2024-12-18:
Close it as it is not an issue of GDH.

closed 2024-12-18 by pachi81

FlorisQK, 2024-12-18:
yeah sorry tried all not able to solve, thanks for your help anyway

FlorisQK, 2024-12-18:
thanks maybe I try during the holidays but you can close the issue here..thanks