pachi81/GlucoDataHandler #186: Cannot enable android.permission.SCHEDULE_EXACT_ALARM on Pixel Watch 3 running WearOS 5

lisalovespiral, 2024-12-06:
Hello!
I am using gluco datahandler on a pixel 9 phone and pixel 3 watch. I have the error message "Schedule exact alarm is disabled!!! GlucoDataHandler may not work correct!!! Press here to go direct to the permission setting" on my watch.

When I press there I get Alarms and Reminders, it shows loading apps for a few seconds and then that disappears. If I go to app permissions through settings I also have no option to grant permission for alarms and reminders for any apps. 

It appears this is a known issue, and there are some possible fixes in the google issue tracker thread, but they are beyond my limited capabilities to implement.

Here is the google issue tracker: https://issuetracker.google.com/issues/365493102

My alarms are working most of the time but sometimes they don't work in the night and that is not good. I was using gluco datahandler for about a year with my pixel 1 and it was stable and wonderful. 

Please help!  :)

pachi81, 2024-12-06:
Hi,
yes the only solution is using the adb command as I wrote already on in this issue.
But if you don´t know, how to use adb, it should normally work without this permission as long you are using the phone version for receiving data.
What source are you using? Do you use the watch to receive the data from source or the phone?
As long you are using the phone, the phone should also trigger the alarm, even on watch.

Michael

lisalovespiral, 2024-12-06:
Thank you for responding so quickly! Yes, I am using juggluco on the phone
as the source and it does seem to work most of the time. Occasionally it
will just vibrate once instead of a sustained vibration, but maybe that is
unrelated to the permissions issue. Thank you for the info, I feel safer
trusting it now.

pachi81, 2024-12-06:
Is the short vibration on watch or phone?
For watch, please go to advanced alarm settings and increase the delay. 
You can also test the alarms on watch and phone, just go to the related alert, scroll down and press test. I recommend to do it on phone, this will also trigger the alarm on watch.
And verify, that you have enabled alarms on watch.

lisalovespiral, 2024-12-06:
Thank you! That was really helpful. I increased the delay and now the
vibration is longer and more likely to alert me. Thanks for making this
great app!

Lisa

pachi81, 2024-12-06:
Great! What delay are you using?
Maybe I have to increase the default value...

lisalovespiral, 2024-12-06:
I have it on the full 3 seconds. I need the longer vibration to wake me up
at night. I love having the vibration on my watch at work too, I don't have
to be that weird diabetic with all the beeping all the time. I can be
undercover diabetic, lol!

pachi81, 2024-12-07:
The delay is not related to vibration itself, it is only the time before starting the vibration. Because the watch notification itself has a short vibration, so I can not start it before. If you want to have a longer vibration, you can use the sound delay of the alarm itself, for example 10s vibration before the sound starts...

pachi81, 2024-12-18:
Close as it is not an issue on my app. Hopefully Google will fix it.