pachi81/GlucoDataHandler #187: Watch battery level widget on phone

rwalker-bluebeck, 2024-12-10:
Any chance the phone battery level could be added to any of the widgets? Or even a new widget..
A notification when the phone battery level went below a certain level would be awesome as well.

I of course do realise I could install a separate app for this but thought it would be a nice feature to have in the app. I like to run my phone/watch as lean as I can...

Many thanks for your hard work on this awesome app :)

RobertoW-UK, 2024-12-20:
Hi. Would you welcome a pull request for this feature?
Decided to learn some Android dev and this seems like a good thing to try (I'm an iOS dev usually)

[image]

pachi81, 2024-12-20:
Hi, if you like, you can provide a change. 
Thank you for your support!

Michael

RobertoW-UK, 2024-12-23:
HI Michael, my pull request is ready to add but I don't have permission to push changes.. Do I need to create a fork?  (I've only worked in private repo's before)

pachi81, 2024-12-23:
I added you as contributor: https://github.com/pachi81/GlucoDataHandler/invitations

But please do not merge to branch 1.3!
Use your own one, so I can check.

RobertoW-UK, 2024-12-23:
Done :)

closed 2025-05-13 by pachi81