pachi81/GlucoDataHandler #188: Do not disturb is automatically set

lolokraus, 2024-12-17:
When you have the case that do not disturb is currently deactivated but the phone is in silent mode.
Then an alarm is triggered and you dismiss the alarm. In this case do not disturb is activated by the app even though it was disabled before.

pachi81, 2024-12-18:
Ok, thank you! I will test.

pachi81, 2024-12-18:
Not sure, what you mean, my test with force alarm sound:
No DnD, Silent -> Alarm disable Silent -> Silent is re-created, DnD stays disabled

DnD+Silent -> Alarm disable Silent+DnD -> Silent+DnD activated again

So what is your setup and what exact is the problem?

lolokraus, 2024-12-18:
I am on Android 15, Pixel 8 and my steps to reproduce are exactly as you described in the first part, only difference is it enables DnD on my phone.

It seems to relate to this: https://stackoverflow.com/questions/58044974/enable-silent-mode-in-android-without-triggering-do-not-disturb

pachi81, 2024-12-18:
Ok, I have to verify in emulator. Thank you for providing already a solution. At the moment, I´m testing new version 1.3, so I will add it.

lolokraus, 2024-12-18:
Thank you! 
I have additional problems getting my phone to vibrate when the current setting is silent, I may look at it in the next days if I can identify the problem :)

pachi81, 2024-12-18:
Great, vibration is something, I can not test in the emulator...

pachi81, 2024-12-18:
@lolokraus Can you do another test for me, because in emulator it seems, that there is also a problem if the ringermode is only set to vibration:
- Enable force alarm sound
- Select an alarm without sound (or set sound to silent), but keep vibration active
- DnD off, Silent on
- Trigger test alarm
- Check ringer mode
In Emulator, it the result would be DnD on, Vibration on. With you fix, it would be DnD off, but Vibration on, instead of Silent.
I have a workaround to set ringermode to normal, before set to silent, but want to make sure, if this behavior is the same in real world...

lolokraus, 2024-12-18:
@pachi81 I tested it and I had the same behavior you described in real life. I also tested the fix you already merged and it then worked correctly. Thx

pachi81, 2024-12-18:
@lolokraus Thank you very much for testing!!! I wanted to provide you a test version, but you already did it! Great! Thank you very much!

lolokraus, 2024-12-18:
No worries, and really great app you provide here. I will soon look a bit at the vibrations because they don't seem to work correctly on my phone, but if you need some help anytime you can write me.

PR #189 (2024-12-18): Reset do not disturb
Fixes #188

Reset the interruption filter to all after resetting the ringer mode.
This should fix the issue. If do not disturb was disabled it will be disabled again.
If do not disturb was in another state it will be reset to the previous state by the following logic of lastDndMode.
commit 6ad20372 (2024-12-18): Fix recreate silent mode

Fixes #188
closed 2025-01-22 by pachi81