pachi81/GlucoDataHandler #190: WearOS/Samsung: Alarm message not on top when in battery saving mode

kdupke, 2024-12-19:
Good Day!
[Antwort in Deutsch ist OK, using English for potential readers outside of Germany]

I run the data handler on my Samsung Watch 6. In principal it works very well  - thank you for this great tool!

During night I switch the watch to battery saving mode, to not have the AOD illuminating my sleeping room. (BTW, any idea how to reduce the brightness below what the standard minimum is?)

In this mode I can't confirm a glucose alarm.
- the alarm sounds
- the screen keep blank
- touching the screen gives me the regular watch face
- I have to use the dial to open up the notification to silence this

Of course, getting up during night, till I am able to silence the alarm, my wife also gets awake. 

What I would like to get is alarm breaks the battery saving mode and wakes up the display with the notification on top OR as soon as I wake up the display (which is a click or tip) the notification is on top.

Thank you for reading so far and looking forward for your help.

pachi81, 2024-12-19:
Hi,

are you really sure, this is related to battery saving? On my Watch 5 Pro (still Wear OS 4) the notification pops up, if the screen is off and battery optimization is on. 
I only have a problem, if the screen is locked, so I have to unlock the screen first and then I have to swipe to get to the notification.

Michael

kdupke, 2024-12-20:
Since yesterday it behaves different. I do know you are ahead of time with fixes! :-)

Seriously, yesterday I updates my watch to the latest version. 
Last night I got an alarm and it seems to work fine.
I will close this now, and reopen with more details if I still experience issues.

kdupke, 2024-12-20:
not reproducable.

pachi81, 2024-12-20:
Ok, thank you for the feedback!!

pachi81, 2025-02-03:
Close as not reproducable.

closed 2025-02-03 by pachi81