pachi81/GlucoDataHandler #193: Resetting setting to default

kdupke, 2024-12-21:
Hi there!

I am still struggling with my alarms. As you may have seen it looks like I now can confirm an alarm without issues after tapping the watch - which is fine for me. However, now the vibration isn't working anymore the way it is expected.

Before creating an issue I would like to reset GDH to the defaults to make sure it is not any of the changes I made in the past to try to fix my issues.

Is there an easy way to reset GDH to defaults? Optimal would be a Button in the app, but in absence of this I also could handle ADB and so (both on my phone and my watch).

TIA!

pachi81, 2024-12-21:
Just uninstall and reinstall. 

But there is a similar issue #191 
So if you know, how to use adb, you can try the beta:
https://drive.google.com/drive/folders/1SFBdrZBXvgZwPBfHaiCpssyuxZOXZwRk

pachi81, 2025-01-22:
You can also go to app settings - GlucoDataHandler - Storage: Clear/Delete storage.
This will remove all custom changes to the app, like a re-install.

I will close this issue, as there is a solution inside Android.

closed 2025-01-22 by pachi81