pachi81/GlucoDataHandler #194: Vibration on DND

usernamebeep, 2024-12-23:
Not sure if I don't know how to configure app or if it is something else but I have enabled DND on my phone and watch but both still vibrates, how do I prevent my devices from vibrating when Do Not Disturb is enabled?

Is there a way to make some alarms override DND and some don't?

Using Galaxy Watch 7 and S24 Ultra

pachi81, 2024-12-23:
You can force vibration and sound in advanced settings. This will overwrite DnD.
It is not possible to enable it only for one alarm.

pachi81, 2024-12-23:
You have to check also your phone DnD settings, as the alarm sound is enabled for DnD. Just play around with the settings. You can test the alarm using the "Test it" button for each alarm.
You can also disable using Alarm sound for alarms in advanced settings.

usernamebeep, 2024-12-23:
Okay, I removed GDH Alarms and notifications special permission. Let's see how that goes.
Thanks.

pachi81, 2024-12-23:
You can test the settings using the Test button for each alarm.
I'm thinking about to add "Ignore for DnD" for each alarm. But as there are so many options, it may confuse the most people 🫣

usernamebeep, 2024-12-23:
Everything is described quite well but there are some duplications for settings. 
Other than that, everything seems to be working except one alarm that xdrip is executing because it can override DND.
Thanks

pachi81, 2024-12-23:
Ok, so everything is fine and this can be closes?

usernamebeep, 2024-12-23:
Yes

closed 2024-12-23 by pachi81

usernamebeep, 2024-12-24:
Well, seems like app still ignores DND even tho I have not given any permissions to it.
If DND is enabled, watch still vibrates. Workaround is to disable alarms in app.
Aldo Very rarely watch vibrates with pattern I have set in GDH.

pachi81, 2024-12-24:
Not possible at the moment.
But for such settings, you can use Tasker app. GDH supports Tasker and with Tasker you can automate nearly everything.
For example disable specific alarm while charging.

usernamebeep, 2024-12-24:
Clear but why does watch vibrate even when DND is enabled?

pachi81, 2024-12-24:
Have you enabled force on watch? 
Maybe it is related to your DnD settings. You can also try, what happens, when you disable alarm sound.
I will test it on my watch.

usernamebeep, 2024-12-24:
Yup, I just did disabled alarm sound. I'll let you know how that works out.

pachi81, 2024-12-24:
You also have to check your phone settings for DnD, as it also has affect on watch.

pachi81, 2024-12-24:
But, you have time as I will not check or do anything for the rest of the year 😉
Have a nice Christmas time.

usernamebeep, 2024-12-24:
You too!

pachi81, 2024-12-24:
Ok, I did some tests 😂🫣

At least in the current test version there is a bug...
You have to remove the permission for DnD for GDH, as GDH disable DnD every time. But I'm not sure, if this is only a bug of my current teat version...

Also DnD exception for media sound is used for vibration.

After disabling all these settings, no vibration during DnD on watch and phone.

But on watch there was the short vibration for new notification from phone, but this is not related to my app, as it is a system behavior for every notification from phone.

usernamebeep, 2024-12-24:
No alarms permissions or DND permission given but phone and watch still vibrates (sometimes with pattern I have chosen and sometimes not)
I'm not sure what i'm doing wrong.

pachi81, 2024-12-24:
Just disable notification vibration in Android settings on watch and check again, as there is also the normal system vibration.

usernamebeep, 2024-12-24:
Could you help me finding that setting?

pachi81, 2024-12-24:
In Galaxy, settings, vibration and sound, notification vibration

pachi81, 2024-12-24:
Yes, but you can not disable default system vibration for only one app...