pachi81/GlucoDataHandler #196: [enhancement] Remove some unused space from largest widget.

sgmoore, 2024-12-23:
There seems to be a lot of wasted space when you resize the home-screen widget to have the text as large as possible.

The following is a screenshot showing two widgets (on a background which shows the true size of the widgets).

[image]

The larger widgets wastes quite a lot of space and there does not seem to be anything in-between these two sizes.

The larger widget can be made longer but that only increases the unused space at the top and between the sections. 
It should be possible to reduce the height to remove more of the unused space, but the widget won't keep that height until it needs to shrinks the text to fit as in the smaller widget.

P.S. My real home screen is obviously more cluttered than the screenshot otherwise this would not be an issue.

sgmoore, 2024-12-24:
>  But the layouts are not so easy to fit each resolution, so I will not do it.

Fair enough.

I must admit I looked at the code to see if I understand how it works and hack it just to suit myself, but it looks beyond me. I can't even figure out how it works out what sizes are valid when you try resizing the widget.

pachi81, 2024-12-24:
The problem with widget is the main screen raster. For example you have a raster of 5x6, so you can only size the widget in this raster, like 1x1 or 1x2 and so on.
But 1x1.5 is not possible. 

I really tried a lot and I'm also not satisfied, but always try to change it, is still frustrating.

sgmoore, 2024-12-27:
> But 1x1.5 is not possible.

That make sense so you can close this call.

My solution is to have two widgets, a small one on the main home screen and then the larger one another screen of its own and I can swipe over the the large one when needed.

closed 2024-12-27 by pachi81