pachi81/GlucoDataHandler #197: Floating Widget improvements

itrn57, 2024-12-26:
Hi, thank you for the the great app. 
Could you please make a few enchancements to the floating widget functionality?
- Add floating widget to a watch face (similar to Juggluco floating glucose).
- Add floating widget to Always On screen for Galaxy phones.
- Make an option to fix a position of the floating widget.
Thanks a lot for your work.

pachi81, 2024-12-26:
> Hi, thank you for the the great app. Could you please make a few enchancements to the floating widget functionality?
> 
> * Add floating widget to a watch face (similar to Juggluco floating glucose).

Wear OS only support the permission using adb. So the most user can not use it.
I also see no real advantage. So maybe if I have some boredom.

> * Add floating widget to Always On screen for Galaxy phones.

This is no floating widget. For Samsung there is already an app bringing widgets to AOD. 
But I also thinking about this feature, but it has no prio...

> * Make an option to fix a position of the floating widget.

Do you really need it? I always have the floating widget at the same position.
What do you expect? Always the same start position or complete fix position? 

>   Thanks a lot for your work.

Thank you for your feedback. 
Michael

itrn57, 2024-12-26:
This is why I like and prefer Floating Widget to complications:
- Customizable size, I need at least font size 14 for my eyes. 
- Custom placement, I can place it where I like it not to a predefined place of a complication. 
As of fixing a widget position - this is for a pure convenience to prevent it from accident move with a "fat finger".
Thank you.

pachi81, 2025-03-19:
Hi, in the next version, I will add AOD support and locking the floating widget position.
Floating widget for watch is not planned as the permission can not be granted without adb.
There are already some watchfaces with big complications available, so I think there is no need to invest time in a feature, that nearly nobody will/can use...

closed 2025-05-13 by pachi81