pachi81/GlucoDataHandler #198: App shutting down

sgmoore, 2024-12-27:
GlucoDataHandler looks to be shutting down. 

This is almost certainly **not** an issue with GlucoDataHandler as the problems have only started recently but I am wondering if you have any ideas of how to diagnose exactly what is happening and why.

I have GlucoDataHandler set to send values to a watch via WatchDrip+ and the watch has stopped receiving values over a dozen times over the last few days. When that happens the home screen widget displays a value and the time usually says 0 min, but the widget is not actually updating. When I open GlucoDataHandler the displayed value is missing (i.e. showing ---) which looks to me like the app has just started (and if it was not running that would explain why the widget is not getting updated).

I am not getting any warning from google that the app has crashed. 

The other thing that occurs is that my wallpaper resets to the default wallpaper, but I don't think the whole phone is resetting.

I know android can kill apps under certain circumstances, but AFAIK this shouldn't happen as the app if the app battery usage is set to Unrestricted.

Any ideas?

P.S. I was originally running my own build and thought I had messed up, but the same thing happens with the playstore build.

pachi81, 2024-12-28:
Hi, sounds like the app has been stopped in background. If there was any crash, please send me the logs, after crash occurs.
So I can check.
What about notification? Is it active? This may dissappear, if the app is stopped.

sgmoore, 2025-01-13:
This problem disappeared for most of the week, but started happening again at the beginning of the month.
The only thing I can see is that app does seem to be killed by android.

```
1895	01-04 16:10:50.023   375   375 I lowmemorykiller: Kill 'de.michelinside.glucodatahandler' (29299), uid 10253, oom_score_adj 915 to free 67524kB rss, 41308kB swap; reason: device is not responding
3347	01-04 16:11:49.723   375   375 I lowmemorykiller: Kill 'de.michelinside.glucodatahandler' (30898), uid 10253, oom_score_adj 905 to free 67796kB rss, 41288kB swap; reason: device is not responding
4912	01-04 16:13:18.601   375   375 I lowmemorykiller: Kill 'de.michelinside.glucodatahandler' (31112), uid 10253, oom_score_adj 925 to free 71368kB rss, 38164kB swap; reason: device is not responding
7275	01-04 16:15:43.247   375   375 I lowmemorykiller: Kill 'de.michelinside.glucodatahandler' (31246), uid 10253, oom_score_adj 905 to free 69916kB rss, 36776kB swap; reason: device is not responding
7940	01-04 16:16:13.456   375   375 I lowmemorykiller: Kill 'de.michelinside.glucodatahandler' (31500), uid 10253, oom_score_adj 905 to free 70956kB rss, 38124kB swap; reason: device is not responding

```

I presume this means the app is restarting itself and then killed again very shortly.

Restarted the phone and the problem disappeared for another few days and then occurred again. 
One time instead of restarting, I went through loads of running apps which I did not really need and the problem then disappeared, but I assume it will come back again at some time.

The apps using the most memory did not seem to be the same so I am not sure what is causing the problem, except it is pretty obvious that it is not GlucoDataHandler itself.

pachi81, 2025-01-13:
Looks like, the system runs in OOM (out of memory) and kills all less important apps.
Maybe you can increase the importance of GDH for the system, but I think this is a manufacturer related setting.
Which phone do you have?

sgmoore, 2025-01-13:
Phone is a Moto G13 running Android 14 and it is not rooted. (Tried and failed!)

pachi81, 2025-01-13:
I have found these solutions, but it seems to be a common problem with these devices:

1. Disable Battery Optimization for Specific Apps:
Go to Settings > Battery > Battery Optimization.
Find the app(s) affected and set them to "Not Optimized.

2. Disable Background App Management:
Navigate to Settings > Apps > Special App Access > Background App Management.
Ensure the app(s) are not restricted. 

3. Check Adaptive Battery Settings:
Go to Settings > Battery > Adaptive Battery, and consider disabling this feature to see if it improves app behavior.

sgmoore, 2025-01-13:
I have changed 1 and 3.

Don't have a  Background App Management option under Special App Access.

pachi81, 2025-02-03:
Any news on this topic or can I close it?

sgmoore, 2025-02-03:
It is still happening, but a lot less often (maybe once or twice in the last three weeks). 
But I think you can close it as the problem is obviously nothing to with GlucoDataHandler.

closed 2025-02-03 by pachi81