pachi81/GlucoDataHandler #200: Unable to remove "recieving data about complications" notification

Grimphoenix-dotcom, 2025-01-18:
Heyo, not so much a technical issue as a UI one, I want to get rid of the constant "recieving data about complications" notification that will not go away on my pixel watch, unfortunately I can't find any way to do that. the watch also does not offer a "hold the notification for a while and turn it off" option like phones would.
If this is not a feature I feel that it should be, most people want their notifications as clean as possible, and it seems like it should be fairly easy to remove.
sorry if this is the wrong place to say this, and sorry again if this is something that exists and i just missed it. I am not a programmer but do have experience in UI design and consumer psychology.

pachi81, 2025-01-18:
Hi,
this is a "foreground" notification. It can not be removed, as it prevents Android to stop the app in the background. Even on phone it is not recommended to disable it.
On phone you can set it in settings to be empty and swipe it away, do not disable it!
Unfortunately this is not possible in Wear OS.
Michael

Grimphoenix-dotcom, 2025-01-18:
Thank you, I have been playing with it for the last half an hour and realized what I was missing, going to close this issue and repost with my new issue!

closed 2025-01-18 by Grimphoenix-dotcom