pachi81/GlucoDataHandler #201: Graph

Grimphoenix-dotcom, 2025-01-18:
Hiya,
This app is the only one I have found that works with pixel watch 3, so props on that dev!
Though I can not seem to get a graph to appear no matter what I do, selecting what to display on the watch app all i get are variations on displaying the number and the trend.
This may be intended but I hope that it is not, this app is the only one that I have found that works, sorry about the spam.

pachi81, 2025-01-18:
I'm already working on the graph, but after 3 weeks of development it works inside the app, but not for widgets and complications. So either I will find a solution or have to find another graph, which is working.
As I'm doing it in my freetime, such things will take a long time, as I have next to work, children and wife max half an hour per day for development...

For me personal the value, trend and delta is all I need, the graph is the past, which I can see in my libre app...

If you will need a graph, now, you can try Blose.

I will close this issue as duplicate of #154.

closed 2025-01-18 by pachi81

Grimphoenix-dotcom, 2025-01-19:
Thanks for the response, sorry, this is an amazing app and I respect the development and the dev so much, I want to make sure you get the cred you deserve, I reread my comment and sounded like a total karen, so sorry :(

pachi81, 2025-01-19:
Hi, no, it doesn't! Not for me, but I even have no idea, what "karen" means 🤣
Maybe I was a little bit frustrated, because of my problems with graph, so I have to say sorry 🍻

But I also have good news, yesterday I have solved the problem with the bitmap for widget, complication and so on, so I can continue working with this graph library.

Have a nice day!

Michael