pachi81/GlucoDataHandler #204: Change Wear apk name?

Mongoose69, 2025-02-11:
Could you possibly change the name of the GlucoDataHandler Wear apk to something like GDH_Wear.apk?

The filename is a bit long and makes installing it awkward as my phone screen only shows GlucoDataHandler

Not a serious issue but it's annoying as just connecting ADB properly is irritating because I can never remember the correct way of pairing it to my watch 

Thanks.

pachi81, 2025-02-19:
Sorry for late response, but why do not change it yourself before installing?
You can also download from Playstore, which is much easier.

Mongoose69, 2025-02-24:
I did, it's just an annoyance really

pachi81, 2025-02-24:
Why not using playstore? I still not understand the problem...