pachi81/GlucoDataHandler #205: Notification Read Feature Request

manor7777, 2025-02-11:
Xdrip+ has a companion mode that reads notifications from various sources (Dexcom G6, G7, One+, etc.) and transfers the values to other apps like AAPS. Unfortunately, Xdrip+ doesn't use the arrow values (e.g., ↗️) and creates its own, causing issues in apps like AAPS. I currently use GlucoDataHandler to bypass this, but it requires an internet connection to fetch values from Dexcom Share or Nightscout, which isn't always available.

Would it be possible to integrate a feature into GlucoDataHandler that reads notifications directly, allowing for offline use with my Dexcom and AAPS setup? Since Xdrip+ is open source, perhaps some of their code could be modified to achieve this.

If you're able to implement this feature, I'd be happy to buy you a coffee as a token of my appreciation.

Thank you!

manor7777, 2025-02-12:
Hi there @pachi81,

It's not that Xdrip doesn't calculate the trend arrows, it's that it's WRONG. It's because you cannot change what values it considers for which store, for example, if my Dexcom is 5.5↗️ then Xdrip might be 5.5⏫ (Double up arrow). I believe xdrips threshold is lower (say 0.01 per minute instead of 0.1) and it's not changeable. I Believe you can do this :D

To the best or my knowledge, it's because they don't use the same values to calculate what qualifies as an ⬆️ or ⏫ (also ⬇️⏬) I have attached what Dexcom uses to calculate the arrows, at least the extreme ones. You will also want to convert them to Mg/DL but I'm sure you have that already

Attached is the Dexcom notification as requested and the threshold for at least the extreme/rising and falling fast arrows from inside the Dex app

[image]
[image]

manor7777, 2025-02-12:
Sorry I didn't mean to close with comment 🤣

pachi81, 2025-02-12:
Oh, it is configurable... At the moment, I do not plan to make it configurable. But we are checking, if it is possible to parse the arrow from notification. If it is an unicode string it may be possible. Not for icons...

manor7777, 2025-02-12:
It's not really configurable as such, those values are what they use for arrows (that you can put into GDH)

The first option is the default, and means you will get an alert when it's a single arrow, 2nd option will ONLY alert you when it's a double arrow

Since you already have alerts set up and working in GDH that won't be an issue, it's already configurable in GDH!

All you need (I think) are those values to work out the trend arrows yourself on GDH when reading the notification 

If I have understood all that correctly!

pachi81, 2025-02-12:
Ah, you mean, for double up and down it should be related to the alert settings!?
OK, at the moment, these are fixed values related to rate/scope value provided by Juggluco and xDrip. But these values have not a real unit and even hard to compare. For me it looks a little bit like mmol per minute*10.
For my trend calculation I also check the values of the last 25 minutes (at the moment). For rising alert I only check the last delta.
But you are right, maybe I have to re-think the rising and falling alert...

manor7777, 2025-02-12:
Your alerts are fine, I'm just talking about GDH calculating the trend for a local setup since you (I think) can already read the value, not that it's related to the alert settings

I think I was a bit confusing with that's screenshot. That screenshot IS from the alert settings, but it also has values there showing what it uses to make that alert (so what it considers rising fast or falling fast. Aka the arrows ) Your alerts are already great and I don't believe the need changing

0.1-0.2mmol/L or 0.5-1.0mmol every 5 minutes is ⬆️⬇️ and anything above that is ⏫⏬, and guessing the ↗️↘️ something like 0.3-0.4 every 5 mins, but I don't have a value for that directly

I was doing *5 because if it's 0.1mmol/L per minute, *5 would be 5 minutes 

I was just giving what I think are the values so you can work out the trend arrow locally :)

pachi81, 2025-02-13:
Hi,
I also re-think and will let the alarms as it is. 
At the moment, the double arrow will be visible, if the value per minute is 0.3 mmol/l over the last 25 minutes. So I think at the moment the double arrow is not really present, I only got it once at the moment. 
Not sure, if I have to change this behavior, but a double arrow is a real exception...
0.2 mmol per minute is arrow up/down and 0.1 mmol per minute would cause 45° arrow. 
Maybe I can change the double up/down to 0.25, but not really sure.

Regards, Michael

manor7777, 2025-02-13:
Alright, I can't wait to test!

I cant xdrip but because it works the arrows out differently to Dexcom, it messes up when you have a pump, say I'm 7.5↘️ on Dexcom, xdrip will say I'm  7.5 ⬇️⬇️ so it stops delivering insulin when it shouldn't. I use GDH to bypass this so the arrows are correct, but it's not offline (yet)

Look forward to testing your new release :D

pachi81, 2025-02-19:
Hi,
ready to test 😉
You can install in parallel to GDH for testing.
But you have to enable also the Dexcom foreground notification, as the notifications of Dexcom are really strange...

https://drive.google.com/drive/folders/1NIIDsqwnXDdfZjeI9I_-DUQzPacWrSqi

manor7777, 2025-03-01:
Sorry I missed this. GitHub emails are trashola and never arrived half the time. Testing in progress!

pachi81, 2025-03-01:
> Trend arrows are SPOT on, so far (They are a little off axis when using 1 min delta (you use more than 5 arrow directions, excluding doubles), thae 1 min delta should be disabled (as in locked out) in both sources and notification reader areas of Dexcom of any type is the selected app (Mainly Dexcom G7 and Dexcom One+) + See 2nd picture, but not an issue so far when configured properly

Yes, I use 5° steps instead of 45° steps. I think this makes more sense. Especially if you have a trend of 44° all apps are showing steady trend arrow... I think, that is wrong, so I'm using 5° steps. Also double arrows are possible, but in really corner cases...

You can switch to 5 min delta in settings. As I'm using libre, I prefer 1 min delta. That is the reason, I also show this delta value.

The 5 min interval for notification reader is not related to delta, it is related to reading, as the notifications are updated quite often, so it is hard to detect, which one is a new value or only a notification update. For G7 I also recommend to enable Dexcom foreground notification, as it helps to better detect...

> The 15 mins trend value, and the bottom trend value on the app homepage are broken I think, both say 0, but it's development version so expected!

It seems, you understand it wrong, the 15 min delta is the real difference between your last value and the value 15 min ago, so related to your graph, the delta is correct. Only the 1 min delta is calculated, if there is no 1 min interval.
The last trend value is the trend arrow rotation in degree. It is more a debug output for me and it my be removed in final release.

> What a f*cking amazing developer you are, where do I buy you a coffee for your hard and amazing work?!

Checkout the readme for this app here on GitHub 😉

Thank you for your feedback!

pachi81, 2025-03-01:
[image]

Even double arrow is possible...

manor7777, 2025-03-02:
I don't have AndroidAPS installed at the moment due to new phone, the only question I have, is how those arrows will translate to androidaps, at that ONLY supports ⏫⏬⬆️⬇️↗️↘️➡️ and is the whole reason for this update (for me anyway). But I do like that it has more arrows!

Struggling with the delta a little also, Old GDH will tell me +0.2 where New GHD may say +0.02 and I think they should be the same (like old GDH) - may be a bug, or may be just what I'm used to but I'll attach a few pictures anyway - The arrows are also sometimes different (as in completely) which is BAD as AndroidAPS needs accurate arrows. It acts on delivering insulin with these arrows 

[image]

[image]

You can see the differences in those 2 images. And Old GDH is always correct as it's pulling from nightscout

Other than that, it's perfect :)

pachi81, 2025-03-02:
You have to enable 5 minute delta, the only bug is your config 😉
It will translate as normal, 0-40° steady, 45-85° is diagonal up, 90° up...

The old version can only show the arrows, received from source, so not my problem.

manor7777, 2025-03-02:
I believe I have been using 5 minutes delta already, but the arrows are still not what they should be. If you show me where I enable it I will test but this is what I've been using. The previous screenshots were using the below settings

[image]
[image]

manor7777, 2025-03-02:
See the arrows here. the delta is correct this time but the arrows is not. I've been using the above settings since before my initial feedback

[image]

Just NOW the below screenshot shows the delta not right again, but this time the arrow is correct 

[image]

pachi81, 2025-03-02:
You can not compare with xDrip source, as xDrip has its own calibration algorithm.
Just checkout the graph in second

pachi81, 2025-03-02:
Please enable Dexcom foreground notification. And please show the app screenshot, I can not say anything without graph.

manor7777, 2025-03-02:
What do you mean by foreground notification? The one in the screenshot or the foreground channel in 2nd screenshot? It works without the foreground channel as long as the main notification is there (1st screenshot). As you can see the delta should be 0.1 but it's 0.06

[image]

[image]

manor7777, 2025-03-02:
Sorry adding more screenshots including an app one

[image]
[image]

pachi81, 2025-03-02:
Please enable Foreground Channel for Dexcom.
And sorry, but I do not really care at the moment 🤣

[image]

manor7777, 2025-03-02:
Not may people do care if I die, don't worry 😂 I have enabled it, I'll keep testing

Enjoy your holiday dude!

pachi81, 2025-03-02:
The problem with the notification is, that there are multiple notification within 5 minutes, so it is hard to detect, what is a new value, what is only an update of the notification.
So the delta may differ, as the wrong notification is detected...
With foreground channel notification it os working quite better, as this normally only update for new value, so I'm using this as indicator... but I also have seen irregular updates of this notification. I really have no idea, what dexcom is doing with their notifications. It is really weird.
If you really want to have only new values, you need to use either Dexcom Share or BYODA.

I also not sure about the trend calculation, as I do not have any experience with.
I think it is working, but in some situations like having steady values and then an increase, like in your one screenshot, the calculated value is too high. Maybe I have to re-think.

manor7777, 2025-03-03:
Ok, almost final feedback - you will like this :D

I'm awaiting the results for AndroidAPS integration (with the arrows etc), that will take a few days!

The new GDH seems to, more often than not, be MORE accurate than the OLD GDH for the delta, after giving it more time (the +-0.1). The arrows I still believe should have a 45° mode - maybe an option in settings? I like the extra arrow directions but I don't think AndroidAPS does, but I will have final results in a few days. I will post a final feedback in a few days but mostly great so far

Thanks again!

pachi81, 2025-03-05:
If the alarms occur again, please send the logs, but there seems to be ignored, only the error message appear...

Related to the irregular readings, have you checked the graph in GDH? Are there only new values every 5 minutes or are there more? If there are more, please provide logs and app screenshot with graph, but be aware, that the logs only covers 15-20 minutes depends on the phone.

One more thought, you can try to specify the interval in settings - transfer values, so GDH will pnly send new values every 5 minutes, even if there are more. But if the readings from notification comes to late it is much harder. Therefor I need the lofs to analyse!
Logs can be saved in Import/Export settings. But not the first entry for settings!

pachi81, 2025-03-10:
Hi @manor7777 any news on this topic? Can you provide logs and graph screenshots for the problem? Otherwise there is nothing I can do...

manor7777, 2025-03-17:
Hi, so sorry been ill the past week, still a bit ill.

The only major issue is when it won't read the text LOW or HIGH (it's expecting a number) - but I'm not putting my health at risk on purpose to get that going again, see the high one here https://github.com/pachi81/GlucoDataHandler/issues/205#issuecomment-2700433245

The timing issue I couldn't resolve, it's not really anything I think you can do since there is a lag with the process of reading notifications 

Other than that, I'm super happy for a release, waiting so I can get it back in my watch again!

manor7777, 2025-03-19:
Ooh to add to this, can you allow it to pull COB (carbs on board) from the AndroidAPS (AAPS) notification? You already support COB just not reading it from notification 

[image]

pachi81, 2025-03-19:
@manor7777 This feature is mainly for source, do not send any broadcast or anything else. 
And as there is no known app providing COB at the moment, I do not support COB reading from notification.
But AAPS provides the COB value, if I remember correct in the status-line inside the broadcast. Can you check to enable it.

The other problem is, at the moment I only support COB and IOB from one source with same timestamp. For notification support, you may have one notification for IOB, one for COB with different timestamps. This is hard to handle in my current code. So I will only add this feature it is really needed... 
In my opinion, AAPS should offer this in broadcast and it should be changed in the code by AAPS, if it is not already available in status-line...

pachi81, 2025-03-19:
Check AAPS code and it should be part of the normal broadcast for "Samsung Tizen" or "Data Broadcaster". If not, please check your AAPS configuration...
So in my opinion, it makes no sense to request data from notification, if there available as broadcast...

manor7777, 2025-03-19:
Ah you are correct, I have a new phone since I last used AAPS so I hadn't enabled Tizen. That works perfectly

Other than GDH not reading HIGH and LOW values, I'm super happy, and thankful you got this working!

I will buy you a coffee when I next get paid, and can't wait for release so I can get alerts on my watch again 🤣

svmi1, 2025-04-16:
Hi, I installed from Playstore (v1.3.1) but the notification reader feature is not available in Settings -> Sources.
How can I activate the notifaction reader?

pachi81, 2025-05-02:
Hi @tmarini1,
it is already possible in the current playstore version using Dexcom Share.
In version 2.0 I added notification reader for getting data from Dexcom notification. 
But the notifications of Dexcom are really weird, as they have multiple updates, not only for new values...

tmarini1, 2025-05-02:
I was told version 2 beta had the ability to get data direct from standard app without having to be online?

pachi81, 2025-05-02:
> I was told version 2 beta had the ability to get data direct from standard app without having to be online?

Yes, while reading notification as xDrip+ already is doing.

tmarini1, 2025-05-02:
How do I go about getting this "offline" version? I'll try it out.

pachi81, 2025-05-02:
Just wait around a week or assign yourself to the beta version via playstore.

tmarini1, 2025-05-03:
Hi there. Now have beta 2.0. I think I misunderstood. This version can or cannot receive glucose values from the official dexcom G7 app downloaded from the play store?

pachi81, 2025-05-03:
It can read the notification. Or using Dexcom Share. There is no other interface provided by Dexcom app.

closed 2025-05-13 by pachi81