pachi81/GlucoDataHandler #207: how to build?

IzzySoft, 2025-02-15:
At IzzyOnDroid we support [Reproducible Builds](https://reproducible-builds.org/) (see: [Reproducible Builds, special client support and more at IzzyOnDroid](https://android.izzysoft.de/articles/named/iod-rbs-mirrors-clients)). Trying for yours, I was unable to even build it. Running `./gradlew assembleRelease` only yielded

```
> Configure project :auto
WARNING: The option setting 'android.defaults.buildfeatures.buildconfig=true' is deprecated.
The current default is 'false'.
It will be removed in version 9.0 of the Android Gradle plugin.
To keep using this feature, add the following to your module-level build.gradle files:
    android.buildFeatures.buildConfig = true
or from Android Studio, click: `Refactor` > `Migrate BuildConfig to Gradle Build Files`.
Configuration cache entry stored.
FAILURE: Build failed with an exception.

* Where:
Build file '/build/repo/common/build.gradle' line: 17

* What went wrong:
A problem occurred evaluating project ':common'.
> Parameter specified as non-null is null: method com.android.build.gradle.internal.dsl.BaseFlavor.buildConfigField, parameter value
```

We'd appreciate if you could help making your build reproducible. We've also prepared some [hints on reproducible builds](https://gitlab.com/IzzyOnDroid/repo/-/wikis/Reproducible-Builds/RB-Hints-for-Developers) for that.

Looking forward to your reply!

pachi81, 2025-02-15:
Hi!
I never had problems building the project in Android Studio, but I will check.

Regards, Michael

IzzySoft, 2025-02-15:
Thanks! We use CLI to build (in ephemeral Podman containers), so GUI is not possible. And btw, the error seems to mean the build process wanted to read something from a global setting, which obviously was not defined when I built:

    buildConfigField("String", "DEXCOM_APPLICATION_ID", rootProject.DEXCOM_APPLICATION_ID)

A [quick search](https://github.com/search?q=repo%3Apachi81%2FGlucoDataHandler%20DEXCOM_APPLICATION_ID&type=code) reveals that should have happened in `[common/src/main/java/de/michelinside/glucodatahandler/common/tasks/DexcomShareSourceTask.kt](https://github.com/pachi81/GlucoDataHandler/blob/8485b0091c5b87ffc9fcf8dd242bdae4efa21ad5/common/src/main/java/de/michelinside/glucodatahandler/common/tasks/DexcomShareSourceTask.kt#L94)` (also in line 225 there). No idea why that did not happen. I'm no Android dev, so I'm a bit lost where to look further.

pachi81, 2025-02-15:
Ah, now I know the problem.
This is a private property, which is not part of GitHub. For building it, you to define these private properties. 
As it is like a password, I do not want to share. Other projects, like xDrip make it public, but I do not want it...

IzzySoft, 2025-02-15:
I see. Without telling me what it is, where would I need to set it so the app does build? I guess it would build with a different `APPLICATION_ID`? Or do you rather want us to skip RB entirely (which of course would also be an option; RB is just an optional but recommended feature here at IzzyOnDroid)?

pachi81, 2025-02-15:
With a different ID it would build, but maybe not work. At the moment I'm not at home, so I will write you tomorrow.

IzzySoft, 2025-02-15:
To be fair, let me be open as well: with a different ID it would build, but not be RB. But: the diff would tell me the real ID, so I potentially can make it RB then. Of course I'd not publish the ID in plain text. What we do there e.g. with other API keys and the likes is write them to a file, base64-encode that, and put the string into the build recipe. That's no encryption, to be fair, but at least "hides" it from simple scraping.

OK, my cards are on the table now. If that approach is OK with you, I gladly test it. If you say it would cause you headaches, we can simply drop the topic altogether, no bad feelings.

pachi81, 2025-02-15:
No problem, as I still not use my own one. I use the one from xDrip:
https://github.com/NightscoutFoundation/xDrip/blob/728c71025ee44dd185aedb6a528f81582deae266/app/src/main/java/com/eveningoutpost/dexdrip/cgm/sharefollow/ShareConstants.java#L10

So it is no secret 😉
But I thought about creating my own one. But you are right, maybe I have to encrypt it...

IzzySoft, 2025-02-16:
So how do I set that for the build to succeed? I mean, using `sed` to replace the constant by its value is ugly. Is there some `local.properties` or so where you put it for the build?

pachi81, 2025-02-16:
Yes, it is part of the local.proerties file in root directory.
There is also a setting for "releasePath", which is needed for post build step to copy the apk file to another directory.
I´m using it to copy it to my NAS storage to have access, too. But you can also copy to temp folder...
And the "sdk.dir" is defined.

IzzySoft, 2025-02-18:
So I put `DEXCOM_APPLICATION_ID=TheIdHere` into `local.properties`, and that's it? Or are there other steps needed?

pachi81, 2025-02-19:
Sorry, was busy...
You also have to set `releasePath` to a valid path, for example temp...

IzzySoft, 2025-02-20:
Hm, tha APK at release 1.3.1 was not built from the commit the tag points to, was it? The embedded VCS info says it was built from dad8d528ea47d0ae8e56778c5d5a963b881a9061, while the tag points to 8485b0091c5b87ffc9fcf8dd242bdae4efa21ad5

OK, the first one of course was not RB. Built from the embedded commit hash we get closer. Funny dex diff:

```diff
   VISIBILITY_SYSTEM Ldalvik/annotation/Signature; value={ "Ljava/lang/Object;" "Landroid/os/Parcelable$Creator<" "Lcom/eveningoutpost/dexdrip/services/broadcastservice/models/Settings;" ">;" }
-  VISIBILITY_SYSTEM Ldalvik/annotation/MethodParameters; accessFlags={ 4096 } names={ null }
-  VISIBILITY_SYSTEM Ldalvik/annotation/MethodParameters; accessFlags={ 4096 } names={ null }

```

Plus some Windows line endings. OK, the latter are easy to fix. The funny diff probably means you've built using JDK 21, so let's try that as well (by default, we're using OpenJDK 17 here). Looks like my guess was correct there:

```json
  {
    "upstream_signed_apk_sha256": "4d1673bf7f75784da6d861bf399f404b420aac5f881bdb0cc85e64b6c28af008",
    "built_unsigned_apk_sha256": "ddf4e44dd8e9b75cf25e3a3f14158970420109aef6874e56127546d237a7364a",
    "signature_copied_apk_sha256": "4d1673bf7f75784da6d861bf399f404b420aac5f881bdb0cc85e64b6c28af008"
  }
```

Congrats 🥳 That means we succeeded with a reproducible build there! And here comes [your welcome toot](https://floss.social/@IzzyOnDroid/114033513344506732):

[image]

closed 2025-02-20 by IzzySoft