pachi81/GlucoDataHandler #213: [Feature Request]: Trend calculation per minute

sgmoore, 2025-03-27:
I'm running the beta but I don't have **Trend calculation** enabled as the description says it will calculate the trend based based on five minutes.

Was wondering if it would be possible to have an option to calculate this on a per minute basis? 

My source is Juggluco and I have seen the trend arrow point upwards and to the right when **none** of the 1 minute, 5 minute or 15 minute deltas (as shown on the beta version) were positive.

The times when a trend arrow would be most helpful to me are :

- After treating a low or almost low, but at the moment I get down arrows for a long time despite knowing that my figures are increasing.

- When out for a walk to try to bring my figures down when sometimes my figures start falling fast. 

At the moment I am basically ignoring the trend arrow and looking at the actual delta but this is much smaller and harder to read than the arrow on the widget, the app and on my smartwatch.

pachi81, 2025-03-27:
I think you have re-read the info ;-) 
There is no trend calculation based on five minutes. The trend calculation is based on the last values/deltas and the trend arrow rotates in **5°** interval.
It will also do this for Juggluco, as Juggluco offer such granular trend value.
And even if the last delta is positiv, the trend can be negativ... This is not an issue.

sgmoore, 2025-03-27:
> I think you have to re-read the info ;-)

Oops. Didn't have my glasses on and read 5° as 5' 

Thought this might be simple, but maybe not, so I'll close this request.

closed 2025-03-27 by sgmoore