pachi81/GlucoDataHandler #214: Possible freezing issue with current Playstore beta

Arakon, 2025-04-16:
Unsure if this may already be fixed with one of the newer commits, so I figured I'd report it to be on the safe side.

After several days of runtime with no reboot/shutdown of the phone and watch, the complications on the watch will freeze. The time since last value, value, trend, delta and graph (in 3 different complications) no longer update, but also don't show a signal loss or anything. Other complications appear to still work normally.
Force restarting (Force stop in app settings) and manually starting GDH again on the phone will usually result in an update to the values, but will soon freeze again. Restarting both phone and watch solves it until several days pass again.

The widget on the phone continues to update fine for the most part, so the data still gets to GDH.
Sometimes, the "Time since last value" on the widget gets stuck at 0 mins, though, but that may be unrelated.

Hardware:
- Google Pixel 8 on current stable Android 15
- Samsung Galaxy Watch 5 on current stable WearOS 5
- Dexcom G7, values being transferred from xDrip+ to GDH on the phone

pachi81, 2025-04-16:
Hi, 

thank you for reporting! At the moment there seems to be a memory issue causing OutOfMemory excptions after some days. Sounds like it is related to this issue. I have done some changes and hope, that it will fix this problem, but I'm not sure about.
If the problem is part of the graph library, I have a bigger problem...
In some days, there will be a new beta version available in playstore. So please try it out.

Regards, Michael

Arakon, 2025-04-16:
I can not be 100% sure, but I was running a self compiled older version before (compile date is 23rd of march, so probably a30692e0bb86dedac691a16eecc5f88a7ef03950) and I didn't notice the issue. I can't say for sure if I just didn't have it running for long enough or if the issue actually didn't occur.

I'll update to the new beta once it drops and let you know if it occurs again.

pachi81, 2025-04-22:
Yes, the current beta still contain the problem. But I tried another solution and it seems to be fixed, now. So the next beta will contain a fix. Not sure, when it will be available, as I still doing the last changes and improvements...

pachi81, 2025-04-25:
@Arakon there is a new version available for beta testers. This includes a fix for memory issue. Please update and check, if the problem is solved...
Thank you!

Arakon, 2025-04-25:
Yup, already updated. Will report back in some days to see if it occured again.

Arakon, 2025-04-27:
Issue returned after 2 or 3 days, but different.  Widget still updates fine, but the watch claims a lost signal (number and trend is crossed out and changes color). It then says that it has been 7 minutes since the last value. Several minutes later, it will suddenly update, but jump straight to i.e. 4 minutes since last value. A few secs later, widget and xdrip+ update again, but watch will once again show signal loss.
After continuing like this for an hour or so, it stopped updating altogether on the watch only, just shows a value and graph that never changes. When tapping the graph, it opens a screen with lines and labels, but it remains blank for a while. When returning to watchface, the graph and number will have updated, but with gaps for the entire time since the last update.
Rebooting the watch seems to have fixed it, so it appears to be an issue on the watch app only.

[image]

pachi81, 2025-04-27:
Hi, there is also a problem from Samsung in Wear OS 5. It is not only related to my app.
Please save logs of phone and watch the next time. Otherwise I can not check anything.

Arakon, 2025-04-27:
How do I get the logs next time it happens?

pachi81, 2025-04-27:
Phone app - settings - Export/Import - Save Logs...

Arakon, 2025-04-29:
Managed to get a log on the phone side. However, the log of the watch is just a 0 byte file. It's as if the watch app just didn't respond to the log request.

Also, after about an hour, it seemed to recover and start updating on the watch again.

[GDH_PHONE_APP_20250429_164426.txt](https://github.com/user-attachments/files/19964126/GDH_PHONE_APP_20250429_164426.txt)

pachi81, 2025-04-29:
At the moment there are really strange problems for basis features, which I have not changed for a year. Don't know what is going wrong at the moment.

Arakon, 2025-05-02:
Updated to today's beta, now the chart is missing completely on the watch.

[GDH_WEAR_APP_20250502_195724.txt](https://github.com/user-attachments/files/20016611/GDH_WEAR_APP_20250502_195724.txt)
[GDH_PHONE_APP_20250502_195722.txt](https://github.com/user-attachments/files/20016612/GDH_PHONE_APP_20250502_195722.txt)

pachi81, 2025-05-02:
Are you using Juggluco?
I already have found the problem. You have to re-install on watch.

pachi81, 2025-05-02:
Ok, the problem is related to duplicate data in the database. I have seen for Juggluco. Normally these values will be removed by graph creation.
That's why I have changed the database to prevent such duplicates.
Unfortunately, that if older data exists, the migration fails because of these old duplicates...
I already created a fix, but it will take some time until it is available.

pachi81, 2025-05-04:
Please check, if battery optimization is disabled for watch app

Arakon, 2025-05-05:
It's disabled. It also only occurs for an hour every 2-3 days, at least as far as I have been able to notice.

pachi81, 2025-05-05:
I also recognized such a problem. But it is not only related to GDH. Also other foreground apps get killed by system after some days. I think, this is a Samsung bug. I have seen some logs, and the app gets killed by system and even aftee the restart was killed again immediately. After around 5 minutes the app was able to start again.
I already tried to improve this behavior in version 1.3.1, but it does not really helped. 
I think, that it can not been resolved by myself.
That is the reason I shutdown my watch in the evening as I do not need it at night...